Koenkk/zigbee2mqtt #19124: IKEA Blinds in group not responding to MQTT Messages.

chinswain, 2023-09-30:
### What happened?

I have a group of IKEA blinds but they don't respond to the commands published to MQTT. Individual blinds work as expected.

Zigbee2MQTT is running in a docker container (on HOST)

Another group containing bulbs works as expected.

### What did you expect to happen?

Both blinds in group to respond to commands.

### How to reproduce it (minimal and precise)

Add multiple IKEA blinds to a group

Send open or close commands:

 'zigbee2mqtt/Bedroom Blinds/set' with data 'CLOSE'

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 P

### Debug log

Debug 2023-09-30 15:39:43Received MQTT message on 'zigbee2mqtt/Bedroom Blinds/set' with data 'CLOSE'
Debug 2023-09-30 15:39:43Publishing 'set' 'state' to 'Bedroom Blinds'

rtclauss, 2023-10-12:
Same on 1.33.1

### Adapter firmware version
20230901

### Adapter
zzh

In the Z2M debug logs I see:
```
debug 2023-10-11 20:02:49: Received MQTT message on 'zigbee2mqtt/Owner Suite/Blinds/set' with data 'CLOSE'
debug 2023-10-11 20:02:49: Received MQTT message on 'zigbee2mqtt/Owner Suite/Blinds/set' with data 'CLOSE'
debug 2023-10-11 20:02:49: Publishing 'set' 'state' to 'Owner Suite/Blinds'
```

In MQTT the topic looks like this when I press "Close" on open blinds:
[image]

If I close both blinds individually I see this on the MQTT Topic:
[image]

FriedCheese2006, 2023-10-15:
Coincidentally, I noticed this a few months back but moved on. Started digging into it this morning and found this issue already open. So, sam problem here.

dwradcliffe, 2023-12-11:
Same, happy to help debug but not sure what to look for.

rtclauss, 2024-06-10:
Still an issue

timhourigan, 2024-09-28:
Potentially related to, https://github.com/Koenkk/zigbee2mqtt/issues/5378#issuecomment-750377703, which says it isn't possible, at least for Fyrtur blinds.

closed 2025-01-12 by github-actions[bot]