Koenkk/zigbee2mqtt #19129: New Ubisys Firmare can not be installed

Danit2, 2023-10-01:
### What happened?

When i want to update the new firmware i recive the following error: 
```
Zigbee2MQTT:error 2023-10-01 06:48:17: Update of 'Sonnenstore' failed (Timeout: device did not request any image blocks)
Zigbee2MQTT:debug 2023-10-01 06:48:17: Error: Timeout: device did not request any image blocks
    at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:371:28
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
```

Is this a problem of the device or from Z2M?

### What did you expect to happen?

I can update the firmware. 

### How to reproduce it (minimal and precise)

Try to update. 

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20221226

### Adapter

Sonoff-P

sjorge, 2023-10-01:
I've seen this one or twice if the device had issues communicating with the coordinator, it will basically timeout fetching blocks. Moving the device closer to the coordinator might help. (Although when they are install in wall it can be difficult to do so)

jzecca, 2023-10-02:
I'm having the very same issue with my 2 Ubisys D1-R, both less than 2 meters away from my Sonoff-P dongle (LQI > 170), using Z2M 1.33.0-1.

sjorge, 2023-10-02:
Is the Sonoff-P the TI or EmberZNet variant?

SponiHH, 2023-10-02:
Me too. With env:
- HA Yellow 
- Ubisys h1, s1, s2, s1-r

The updates to the previous fw 2.21 from June was fine. Without any problems. The s1-r is only 20 cm from yellow.

But there was a hint from Ubisys fw 2.21 doc. They h‘ve „Added safeguards against some misbehaving third party OTA servers“ . Could this be the issue? From which fw version did you start the update?

sjorge, 2023-10-02:
I seem to be able to replicate it with my C4, will try to do some more digging this weekend. While sniffing the traffic.

sjorge, 2023-10-02:
At a quick glance it does seem the device never enters the OTA mode and is bot requesting new blocks.

It might indeed be related to the changes from the previous update.

I wonder what we might be doing different from what they expect.

jzecca, 2023-10-02:
> Is the Sonoff-P the TI or EmberZNet variant?

The "P" is the TI CC2652P variant.

svenso, 2023-10-02:
I am having the same issue with all 17 devices. Interestingly, I had to exchange one due to a defect this weekend. Updating from 1.9.2 to 2.3.0 worked fine.
But from 2.2.1 it won't update.

sjorge, 2023-10-02:
> I am having the same issue with all 17 devices. Interestingly, I had to exchange one due to a defect this weekend. Updating from 1.9.2 to 2.3.0 worked fine.
> 
> But from 2.2.1 it won't update. 

Yep it looks like none on 2.2.1 are requesting blocks once z2m initiates the OTA. And older device i had around for testing worked fine when i tried when i got originally pinged.

That was on my dev network, not realizing there was a new fw. All Ubisys devices on my prod netwerk are on 2.2.1 and none (only tried one of each type) seem to want to update. (I tried H1, S1, S2, C4, and J1)

I fear that the devices in 2.3.0 will probably also refuse a future update as they should have the same fix as devices on 2.2.1.

There are a few intermediate firmwares needed for a full upgrade chain, but 2.x should all work. So in the case if a device < 2.x it will just go to the latest 2.x is can get.

I'll try and sniff traffic this weekend to see what z2m sends to the device and if it responds at all or not. Based in herdsman logging, it doesn't respond at all.

matok, 2023-10-02:
I just came across exactly the same issue with LD6 for which I'm trying to implement a converter.
z2m 1.33.0
Sonoff-P dongle with 20221226 fw
LD6 version 1.1.4

sjorge, 2023-10-03:
@koenkk it seems when the device does a OTA check we respond either no image available.

It seems the Ubisys device then won't requests blocks when we the coordinator try to start one later.

I wonder it will only so it if we send them after the device itself requests it. Would that make sense?

Koenkk, 2023-10-03:
how often does the device check?

sjorge, 2023-10-03:
> how often does the device check?

~ every 10 minutes

```
debug 2023-10-03 18:01:06: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:02:42: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:04:17: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:05:59: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:07:40: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:09:15: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:10:54: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:12:27: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:14:09: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:15:50: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:17:23: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:18:59: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:20:40: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:22:16: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:23:57: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:25:31: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:27:09: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:28:42: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:30:18: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:31:53: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:33:32: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:35:05: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:36:46: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:38:27: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:40:02: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:41:40: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:43:18: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:44:50: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:46:28: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:48:08: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:49:46: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:51:28: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:53:03: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:54:36: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:56:11: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:57:46: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 18:59:21: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:00:57: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:02:39: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:04:18: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:05:54: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:07:30: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:09:07: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:10:42: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:12:24: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:14:03: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:15:44: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:17:19: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:18:58: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:20:35: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:22:08: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:23:47: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:25:25: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:27:01: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:28:34: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:30:10: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:31:52: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:33:25: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:35:01: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:36:42: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:38:22: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:39:58: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:41:38: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:43:15: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:44:52: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:46:27: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:48:05: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:49:39: Device 'switch/bedroom/light' requested OTA
debug 2023-10-03 19:51:13: Device 'switch/bedroom/light' requested OTA
```

Koenkk, 2023-10-03:
@sjorge does it work after increasing the timeout to e.g. `150000`? https://github.com/Koenkk/zigbee-herdsman-converters/blob/2edb88a076eba4448861a60013ab8174f7220764/src/lib/ota/common.ts#L321

sjorge, 2023-10-03:
The timeout already seems to be 150000 ?

Basically what seems to happen is, the device requested OTA, we so we have no image.
If I hit update for the device it timesout like originally described. I also tried timing me trigging the OTA around the device OTA checkin, without much success.

Koenkk, 2023-10-03:
>  I also tried timing me trigging the OTA around the device OTA checkin, without much success.

Do you see a `Responded to OTA request of '${data.device.name}' with 'NO_IMAGE_AVAILABLE'` when you do this (you shouldn't)

sjorge, 2023-10-03:
I upped it to `1500000` so 1 extra zero:
```
debug 2023-10-03 20:16:32: Received MQTT message on 'zigbee2mqtt/cover/office/window/availability' with data '{"state":"online"}'
debug 2023-10-03 20:16:32: Active device 'cover/office/window' was last seen '1.87' minutes ago.
info  2023-10-03 20:16:32: MQTT publish: topic 'zigbee2mqtt/cover/office/window/availability', payload '{"state":"online"}'
debug 2023-10-03 20:17:50: Received Zigbee message from 'cover/office/window', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31540,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:17:50: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:50.397Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:50: Device 'cover/office/window' requested OTA
debug 2023-10-03 20:17:50: Check if update available for '0x001fee000000533f' (J1 (5502))
debug 2023-10-03 20:17:50: Is new image available for '0x001fee000000533f', current '{"fieldControl":1,"manufacturerCode":4338,"imageType":31540,"fileVersion":35718162}', latest meta '{"hardwareVersionMin":0,"hardwareVersionMax":7,"fileVersion":36701218,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B34-0000-0007-02300422-spo-fmsh.ota.zigbee"}'
debug 2023-10-03 20:17:50: Update available for '0x001fee000000533f': YES
info  2023-10-03 20:17:50: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:50.397Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
info  2023-10-03 20:17:50: Update available for 'cover/office/window'
debug 2023-10-03 20:17:50: Responded to OTA request of 'cover/office/window' with 'NO_IMAGE_AVAILABLE'
info  2023-10-03 20:17:50: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:50.647Z","linkquality":32,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:54: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"cover/office/window","transaction":"w5w9u-15"}'
info  2023-10-03 20:17:54: Updating 'cover/office/window' to latest firmware
debug 2023-10-03 20:17:55: Received Zigbee message from 'cover/office/window', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200319-DE-FB1","swBuildId":"2.2.1"}' from endpoint 1 with groupID 0
info  2023-10-03 20:17:55: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:55.027Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:55: Updating to latest '0x001fee000000533f' (J1 (5502))
debug 2023-10-03 20:17:55: Received Zigbee message from 'cover/office/window', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31540,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:17:55: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:55.580Z","linkquality":32,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:55: getNewImage for '0x001fee000000533f', meta {"hardwareVersionMin":0,"hardwareVersionMax":7,"fileVersion":36701218,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B34-0000-0007-02300422-spo-fmsh.ota.zigbee"}
debug 2023-10-03 20:17:55: getNewImage for '0x001fee000000533f', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":60,"otaHeaderFieldControl":4,"manufacturerCode":4338,"imageType":31540,"fileVersion":36701218,"zigbeeStackVersion":2,"otaHeaderString":"ubisys J1 2.3.0\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":151294,"minimumHardwareVersion":0,"maximumHardwareVersion":7}
debug 2023-10-03 20:17:55: Got new image for '0x001fee000000533f'
debug 2023-10-03 20:17:56: Received Zigbee message from 'cover/office/window', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31540,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:17:56: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:56.805Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
info  2023-10-03 20:17:56: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:17:56.978Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
info  2023-10-03 20:17:56: Update of 'cover/office/window' at 0.00%
[... cut ...]
info  2023-10-03 20:21:17: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:21:17.474Z","linkquality":36,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"progress":0,"state":"updating"}}'
info  2023-10-03 20:21:17: MQTT publish: topic 'zigbee2mqtt/cover/office/window', payload '{"energy":0.06,"last_seen":"2023-10-03T18:21:17.594Z","linkquality":32,"position":100,"power":0,"state":"OPEN","update":{"installed_version":35718162,"latest_version":36701218,"progress":0,"state":"updating"}}'
```

Seems to just be stuck doing nothing.

sjorge, 2023-10-03:
Interesting, I got with the massive time out is working

```
debug 2023-10-03 20:16:32: Received MQTT message on 'zigbee2mqtt/switch/kitchen/light/availability' with data '{"state":"online"}'
debug 2023-10-03 20:16:32: Active device 'switch/kitchen/light' was last seen '1.49' minutes ago.
info  2023-10-03 20:16:32: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light/availability', payload '{"state":"online"}'
debug 2023-10-03 20:16:43: Received Zigbee message from 'switch/kitchen/light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31539,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:16:43: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:16:43.578Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:16:43: Device 'switch/kitchen/light' requested OTA
debug 2023-10-03 20:16:43: Check if update available for '0x001fee000000594d' (S2 (5502))
debug 2023-10-03 20:16:43: Is new image available for '0x001fee000000594d', current '{"fieldControl":1,"manufacturerCode":4338,"imageType":31539,"fileVersion":35718162}', latest meta '{"hardwareVersionMin":0,"hardwareVersionMax":6,"fileVersion":36701218,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B33-0000-0006-02300422-spo-fms2.ota.zigbee"}'
debug 2023-10-03 20:16:43: Update available for '0x001fee000000594d': YES
info  2023-10-03 20:16:43: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:16:43.578Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
info  2023-10-03 20:16:43: Update available for 'switch/kitchen/light'
info  2023-10-03 20:16:43: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:16:43.928Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:16:43: Responded to OTA request of 'switch/kitchen/light' with 'NO_IMAGE_AVAILABLE'
debug 2023-10-03 20:17:42: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"switch/kitchen/light","transaction":"w5w9u-12"}'
info  2023-10-03 20:17:42: Updating 'switch/kitchen/light' to latest firmware
debug 2023-10-03 20:17:42: Received Zigbee message from 'switch/kitchen/light', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20191128-DE-FB1","swBuildId":"2.2.1"}' from endpoint 1 with groupID 0
info  2023-10-03 20:17:42: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:17:42.197Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:42: Updating to latest '0x001fee000000594d' (S2 (5502))
debug 2023-10-03 20:17:42: Received Zigbee message from 'switch/kitchen/light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31539,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:17:42: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:17:42.487Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
debug 2023-10-03 20:17:42: getNewImage for '0x001fee000000594d', meta {"hardwareVersionMin":0,"hardwareVersionMax":6,"fileVersion":36701218,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B33-0000-0006-02300422-spo-fms2.ota.zigbee"}
debug 2023-10-03 20:17:42: getNewImage for '0x001fee000000594d', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":60,"otaHeaderFieldControl":4,"manufacturerCode":4338,"imageType":31539,"fileVersion":36701218,"zigbeeStackVersion":2,"otaHeaderString":"ubisys S2 2.3.0\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":148990,"minimumHardwareVersion":0,"maximumHardwareVersion":6}
debug 2023-10-03 20:17:42: Got new image for '0x001fee000000594d'
debug 2023-10-03 20:17:43: Received Zigbee message from 'switch/kitchen/light', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31539,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-03 20:17:43: MQTT publish: topic 'zigbee2mqtt/switch/kitchen/light', payload '{"energy":1.72,"last_seen":"2023-10-03T18:17:43.337Z","linkquality":43,"power":8,"power_on_behavior_l1":"on","power_on_behavior_l2":"on","state_l1":"ON","state_l2":"ON","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"}}'
[... cut ...]
The cover one, is stil stuck. I also got a 3rd one that didn't make any progess at all that just timed out.
I wonder if we can do something like:
- add a flag to indicate we should queue the OTA instead of try and trigger it on a per device basis
- when triggering an update with the flag set, we queue up the OTA instead
- send the image once the device requests its if it's queued ?

I also noticed a lot of info publishes that don't seem to have a trigger associated with it. Without grep the log is too verbose to be readable unfortuantely.

sjorge, 2023-10-03:
> I am not sure if it helps.
> I sniffed the traffic this evening. After restarting Zigbee2Mqtt, the first time, a ZCL OTA Query Next Image Response is sent to the device:
> Request:
>
> Response:
>
> But then, the ubisys immediately responds with a default answer:
>
> For me (without any zigbee knowledge), this could be the reason, why it won't start pulling the image blocks.
> Meaning we have some sort of specialities in our response which makes the ubisys stack unhappy.
> Maybe anyone has a ubisys bridge to sniff their traffic?

That one switch did manage to update for me, no luck with any of the others though 😞~ sjorge

matok, 2023-10-03:
Not sure if this might be helpful, but this happens right after I turn the LD6 on:

```
debug 2023-10-03 21:35:00: Received Zigbee message from 'kitchen-workdesk-lights-12', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":18088966,"imageType":31532,"manufacturerCode":4338}' from endpoint 232 with groupID 0
debug 2023-10-03 21:35:00: Device 'kitchen-workdesk-lights-12' requested OTA
debug 2023-10-03 21:35:00: Check if update available for '0x001fee00000094e1' (LD6)
debug 2023-10-03 21:35:01: OTA ubisys: got firmware page, status: 200, data.length: 14318
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B01-0000-0006-01110206-spo-fmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B01-0000-0006-0192020D-spo-fmd.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B02-0000-0001-010F0206-spo-fms.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B02-0000-0001-0192020D-spo-fms.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B02-0002-0007-010F0206-spo-fms.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B02-0002-0007-0192020D-spo-fms.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B03-0000-0006-010E0206-spo-fms2.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B03-0000-0006-0191020D-spo-fms2.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B04-0000-0007-01160206-spo-fmsh.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B04-0000-0007-0191020D-spo-fmsh.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B05-0000-0004-010F0206-spo-rms.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B05-0000-0004-0191020D-spo-rms.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B06-0000-0004-010E0206-spo-rms2.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B06-0000-0004-0191020D-spo-rms2.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B07-0000-0004-01160206-spo-rmsh.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B07-0000-0004-0191020D-spo-rmsh.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B08-0000-0004-01110206-spo-rmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B08-0000-0004-0192020D-spo-rmd.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B09-0000-0004-01090206-spo-fmi4.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B09-0000-0004-0192020D-spo-fmi4.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0A-0000-0005-010A0206-m7b-r0.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0A-0000-0005-0193020D-m7b-r0.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0B-0000-0001-01000206-m7b-h10.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0B-0000-0001-01900210-m7b-h10.ota1.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0C-0000-0000-01000206-m7b-wd1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0D-0000-0001-01140240-m7b-h1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0D-0000-0001-011C0300-m7b-h1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0D-0000-0001-01200403-m7b-h1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B0D-0000-0001-01400422-m7b-h1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B11-0000-0001-00920210-m7b-ld6.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B11-0000-0001-00940240-m7b-q95.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B21-0000-0006-0193020E-spo-fmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B21-0000-0006-0194020E-spo-fmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B22-0000-0001-0193020D-spo-fms-rev0-1.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B22-0002-0007-0193020D-spo-fms.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B23-0000-0006-0192020D-spo-fms2.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B24-0000-0007-0192020D-spo-fmsh.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B25-0000-0004-0192020D-spo-rms.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B26-0000-0004-0192020D-spo-rms2.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B27-0000-0004-0192020D-spo-rmsh.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B28-0000-0004-0194020E-spo-rmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B28-0000-0004-0195020E-spo-rmd.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B29-0000-0004-01930221-spo-fmi4.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B29-0000-0004-01940221-spo-fmi4.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B2A-0000-0005-02000230-m7b-r0.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B2A-0000-0005-02010230-m7b-r0.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B2B-0000-0001-01920210-m7b-h10.ota.zigbee
debug 2023-10-03 21:35:01: OTA ubisys: image found: 10F2-7B2C-0000-0001-01010300-ld6.ota.zigbee
[... cut ...]
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
```

Without knowing anything about it, the `"update_available": null` looks a bit suspicious to me as the `"update"` object contains `"state":"available"`.

`payload '{"color_mode":"xy","linkquality":0,"power_on_behavior":null,"state":null,"state_l1":"OFF","update":{"installed_version":18088966,"latest_version":19924002,"state":"available"},"update_available":null}`

m0wlheld, 2023-10-04:
Same issue here with 

* Ubisys S1 @ 20200518-DE-FB1 | 2.2.1, update available [2.3.0](https://www.ubisys.de/wp-content/uploads/10F2-7B32-0002-0007-02300422-spo-fms.ota.zigbee)
* Ubisys H1 @ 20230216-DE-FB1 | 1.3.1, update available is [1.4.0](https://www.ubisys.de/wp-content/uploads/10F2-7B0D-0000-0001-01400422-m7b-h1.ota.zigbee)

Z2M is at 1.33.1

m0wlheld, 2023-10-04:
Also, check #19161 : Could this be a generic issue?

Koenkk, 2023-10-04:
> add a flag to indicate we should queue the OTA instead of try and trigger it on a per device basis

@sjorge this is basically what we do when the timeout is big enough (since the device requests an update any minute it's not a problem). I don't understand why this doesn't already work:

- When you request an update and the timeout is not reached yet, [`NO_IMAGE_AVAILABLE`](https://github.com/Koenkk/zigbee2mqtt/blob/7e63039144ec2adcce3cb076c9ea68d633d1596a/lib/extension/otaUpdate.ts#L123) shouldn't be send as update is [still in progress](https://github.com/Koenkk/zigbee2mqtt/blob/7e63039144ec2adcce3cb076c9ea68d633d1596a/lib/extension/otaUpdate.ts#L79).
- That indicates that it times out too quickly, since the OTA update is waiting for a [`queryNextImageRequest`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/19e8dc35017ecd4ee48fc3a2c14dbd918826526b/src/lib/ota/common.ts#L195)

Could you add some `console.logs` to the code to figure out where it goes wrong?

@svenso thanks for the sniff, but to compare the behaviour a sniff with the original hub is needed. However I expect that that hub just waits for the `queryNextImageRequest` and then sends the update, something that should also be implemented in z2m.

svenso, 2023-10-04:
@Koenkk 
I was able to update one device with deconz on the same network and sniffed it.
It looks like as it sent just an image notify, and a query next image response (which did not get an "Unsupported Cluster Command" back) and then started sending the update:

[image]

Here is the working Query Next Image Response Frame:
[image]
The only difference I see, is the Frame Control Field on the application support layer (which has Ack Required = 1). But I can hardly believe that this can be the reason.

I am able to share the whole dump privately, if it helps / is required.

Koenkk, 2023-10-05:
@svenso please send the z2m and deconz one to me + network keys on telegram (@koenkk)

Koenkk, 2023-10-07:
@svenso found the issue, we responded with the wrong transaction sequence number. This probably also explain why device kept requesting OTA even when we responded with a `NO_IMAGE_AVAILABLE`.

Integrated all the fixes!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-07 by Koenkk

sjorge, 2023-10-07:
I was just about to add some console logs, but since its already fixed i just verified it is indeed fixed.

Working now! Strange it ever worked without.

m0wlheld, 2023-10-16:
Please re-open with reference to #19248 

None of my Ubisys Devices (S1, H1) can be updated using latest-dev, Build: [81a8df9](https://github.com/Koenkk/zigbee2mqtt/commit/81a8df9)

```
info 2023-10-16 11:10:40Updating '0x001fee0000006cd5' to latest firmware
info 2023-10-16 11:10:41MQTT publish: topic 'zigbee2mqtt/0x001fee0000006cd5', payload '{"energy":1.86,"linkquality":43,"power":0,"power_on_behavior":"off","state":"OFF","update":{"installed_version":35718162,"latest_version":36701218,"state":"available"},"update_available":null}'
info 2023-10-16 11:10:41MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x001fee0000006cd5"},"error":"Update of '0x001fee0000006cd5' failed (maxContentLength size of -1 exceeded)","status":"error","transaction":"eupuj-23"}'
error 2023-10-16 11:10:41Update of '0x001fee0000006cd5' failed (maxContentLength size of -1 exceeded)
```