Koenkk/zigbee2mqtt #19130: [New device support]:  Please set right image for MOES bulb ZigBee GU10 RGB+CCT

josemaX, 2023-10-01:
### Link

https://es.aliexpress.com/item/1005004332509922.html

### Database entry

{"id":71,"type":"Router","ieeeAddr":"0xa4c13854453faccb","nwkAddr":11629,"manufId":4417,"manufName":"_TZ3210_rcggc0ys","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_rcggc0ys","powerSource":1,"zclVersion":3,"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":0,"currentY":0,"currentHue":0,"currentSaturation":254,"colorTemperature":153}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1696154733519,"defaultSendRequestWhen":"immediate"}

### Comments

This device image is shown as RGB led driver, please change the image to some kind of bulb.

The bulb is working and showing all attributes, the image used is the issue.

Thanks

### Supported color modes

RGB+CCT

### Color temperature range

Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2023-10-03:
Fix will be included in todays release!

closed 2023-10-03 by Koenkk

josemaX, 2023-10-03:
I've updated just a while to 1.33.1 and now these bulbs appears with the "default/standard" Zigbee Z red logo.
[image]

Thanks!

Koenkk, 2023-10-03:
Should be fixed in a few hours from now.

josemaX, 2023-10-03:
> Should be fixed in a few hours from now.

Yes, now the image is right

Thanks again!