Koenkk/zigbee2mqtt #19131: Entity  is unknown

tomiss288, 2023-10-01:
### What happened?

After removing device (broken thermostat) with option force delete I still revived information "Entity '0xa4c138a7322068fe' is unknown". I checked configuration files in director zigbee2mqtt and there's no any address '0xa4c138a7322068fe in files: configuration.yaml, database.db, state.json.
I found in only one file coordinator_backup.json info about device 0xa4c138a7322068fe, but I think that's backup and has no effect.

### What did you expect to happen?

How to remove old device permanently ?

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20221226

### Adapter

zStack3x0

francisp2, 2023-10-01:
That is the backup of the NVM of your stick. Stop Zigbee2MQTT, Remove it from coordinator_backup.json, restart Zigbee2MQTT

tomiss288, 2023-10-01:
> That is the backup of the NVM of your stick. Stop Zigbee2MQTT, Remove it from coordinator_backup.json, restart Zigbee2MQTT

It doesn't work, still there's error message "Entity '0xa4c138a7322068fe' is unknown"

closed 2023-11-08 by github-actions[bot]