Koenkk/zigbee2mqtt #19134: Memory leak or device limit?

rossowl, 2023-10-01:
### What happened?

I have 172 devices, 109 routers and 63 endpoint. 169 devices, no problem. 172 devices, z2m randomly restarts without success. I must unplug usb stick and plug in again. Then z2m works for a few days. And then again. I have zzh stick connected to Intel Nuc with 8GB mem, z2m as hassio addon and zzh connected by usb2 cable. Thanks for response.

### What did you expect to happen?

Randomly restarts without success.

### How to reproduce it (minimal and precise)

Network stops and doesnt work until I uplug zzh stick a then plug in.

### Zigbee2MQTT version

latest hassio addon

### Adapter firmware version

latest

### Adapter

zzh

rossowl, 2023-10-01:
When I unpair 3 routers, everything works without problems. Tested with sonoff a few days ago. Same problem. 170 devices and z2m or herdsman crash.

Rotzbua, 2023-10-01:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#overiew

szymucha94, 2023-10-01:
That's a large network. Have you considered splitting it between two coordinators on different channels?

rossowl, 2023-10-02:
Thank you for response. Okay, I will install second coordinator to RPi4 with docker and try Z2M Proxy in HA.

closed 2023-10-03 by rossowl