Koenkk/zigbee2mqtt #19138: Danalock Pin codes not working

dhadanalock, 2023-10-02:
### What happened?

When trying to save a new pin code to the Danapad it writes "Publish "set' pin_code to 'Lockname' failed: 'Error: user must be numbers'" It happens when I change the User type. And what's worse - it also deletes the first pin code already stored on the lock.

### What did you expect to happen?

That it would accept the number that I write in "User" and save, change, delete the pin code position.

### How to reproduce it (minimal and precise)

Enroll a Danalock Zigbee - try saving a pin
Write '5' in 'User"
Then chose 'Unrestricted' in 'User type'.
Then the error will happen.

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20220219

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus - Model id: ea60

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12782661/log.txt)

BrianKarlsen, 2023-10-28:
@dhadanalock 

Since updating my zigbee2mqtt docker to latest version (1.33.1) i cant control my lock anymore.
State reporting (locked/unlocked) is fine
Do you have the same problem? 

Getting this in the logs:
[image]

Koenkk, 2023-10-28:
@dhadanalock I guess this happens when setting it from the z2m frontend, can you check with the latest dev branch? It has some fixes regarding the composite types under the exposes tab (of which the pincode is one)

BrianKarlsen, 2023-10-28:
@Koenkk Do you have any idea why I cant control the lock after 1.33.1?
I tried to delete the device from z2m and now I can't even pair it again..

[image]

dhadanalock, 2023-10-28:
Hi Brian.
Did you try giving the lock 9 clicks in the pin hole.
This tells the lock to clear all information about the enrollment on the hub.

Could be that this would help you ebroll it again.

I have updated my HA just now also and have no issues with my 2 locks at home or the test lock that I have in my home office.

/Dan, Danalock

[image]<https://user-images.githubusercontent.com/24981461/278804069-f1d9c664-aa21-45e8-a778-29a3944a15fe.png>

BrianKarlsen, 2023-10-28:
Hi Dan.

I tried that before starting the pairing proces, but without luck.
It flashes green around 10 times but never goes to white.
No reaktion in Z2M...

dhadanalock, 2023-10-28:
Hi Brian.

Do a full reset, 9 clicks (Again to clear any data from the latest attempt), then10 clicks. Pull a battery to reboot. Try enrollment again.
Remember to recalibrate, set any settings, and load pin codes again - it is all cleared with the 10 clicks.

Could you scan the qr code to see the production date, that will let me know if it the 1.3 Zigbee hardware or the 3.0 Zigbee hardware.

Br. Dan

BrianKarlsen, 2023-10-28:
Did not help. still 10 green and no white. No reaction in Z2M..
Date: 05.20.2021

dhadanalock, 2023-10-28:
Have you updated the lock according to https://danalock.com/zigbee

BrianKarlsen, 2023-10-28:
I've had mail contact with you in danish back in april :)
I did a update then... Shall i do it again?

(we can continue in danish again)

dhadanalock, 2023-10-28:
Let's keep it in English if others experience the same and we figure out why this is happening.

I am trying right now to see if I can force mine to do the same.

Normally one click (inclusion mode) will make it flash one time (if the lock already thinks it is enrolled on the hub) or 4-5 times (when not enrolled).

It shouldn't flash 10 times.

Can't seem to force mine to fail like that.

I am actually not sure what it will do if the enrollment isn't doing what we except. Maybe if you have another Zigbee hub to test on we could figure that part out. If the lock enrolls on another hub we can start looking at the hub.

I assume you have tried restarting the hub right?

/Dan

BrianKarlsen, 2023-10-28:
Restarted the hub for the 4. time
Ran the cycle again an suddently it only flashed 5 times and BOOM!Back i Z2M and I can control it again :)

Really weird... 
Thanks

dhadanalock, 2023-10-28:
Perfect!

But yes, weird.

But glad you got it back in again.
And pin codes sounds like from Koen K to be resolved when dev fixes gets out live.

I will be testing the den branch at a later time.
Don't have the time this weekend.

/Dan

BrianKarlsen, 2023-10-28:
I dont use pincodes :)

I updated Z2M because I installed new Sonoff radiator thermostats and they needed it to work.

BrianKarlsen, 2023-10-31:
Hi again. Unfortuatly I lost the connection again yesterday. Cant control the lock from Z2M.
Redid the process but the lock is back to flashing 10 times and not pairing.

dhadanalock, 2023-10-31:
Hi Brian.
Could you write me an email on ***@***.*** and we could see if we can get your lock back here for some testing on our home assistant setup.

My locks both at home and in the office are still online.

I believe that sending the lock to us for testing is easier than starting over or perhaps bug searching on HA and Z2M.

It could be the lock, but I am most leaving towards an issue that happened when updating the platform. But I can test the lock for you.

Breast regards
Dan, Danalock

closed 2024-05-03 by dhadanalock