Koenkk/zigbee2mqtt #19141: [New device support]:

dankarization, 2023-10-02:
### Link

https://miboxer.com/product/ac-triac-dimmer-zigbee-3-0-2-4g-push

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0xa4c138b6a5f96053","nwkAddr":47388,"manufId":4417,"manufName":"_TZB210_g01ie5wu","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"65503":"�έ,\u0012","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZB210_g01ie5wu","zclVersion":3,"appVersion":64,"modelId":"TS0501B","powerSource":1}},"lightingColorCtrl":{"attributes":{"57600":[1000,0],"tuyaRgbMode":0}},"genLevelCtrl":{"attributes":{"61440":1000,"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1696260591577,"defaultSendRequestWhen":"immediate"}

### Comments

I got all data points, but did not understand documentation about 'exposes' part and tuyaDatapoints.
Especially, brightness needs 'scale0_254to0_1000' what I was able to found in other converters, but 'min_brightness' exposes values from 0 to 100, and I did not see 'scale0_254to0_100' or some analog to convert this parameter from 0-100 to 0-254
Even on-off commands does not work at all 😩

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{
        // The model ID from: Device with modelID 'TS0601' is not supported
        // You may need to add \u0000 at the end of the name in some cases
        modelID: 'TS0501B',
        // The manufacturer name from: Device with modelID 'TS0601' is not supported.
        manufacturerName: '_TZB210_g01ie5wu',
    }, ],
    model: 'TRI-C1ZR',
    vendor: 'MiBoxer',
    description: 'AC Triac Dimmer (Zigbee 3.0 + 2.4G + Push)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.light_brightness().withMinBrightness(),
        e.power_on_behavior().withAccess(ea.ALL),
        e.countdown(),
        e.breaker_mode()
    ],
    meta: {
        // {"1":"ON/OFF","2":"Mode","3":"Brightness","4":"Minimum Brightness","7":"Countdown","9":"Adjust","34":"Breaker Mode"}

        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],
            [3, 'brightness', tuya.valueConverter.scale0_254to0_1000],
            [4, 'min_brightness', tuya.valueConverter.raw],
            [7, 'countdown', tuya.valueConverter.raw],
            [34, 'breaker_mode', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

### Supported color modes

1 color dimmable lights

Flo70469, 2023-10-19:
Did  you got some news? i am interested too.

dankarization, 2023-10-20:
No news, I can't use data I got from tuya developer portal the right way for the converter :(

Koenkk, 2023-10-20:
Doesn't this device work out of the box? Should be detected as `TS0501B`

dankarization, 2023-10-20:
By default converter there's no option to toggle breaker mode on or off (in this mode it works as a relay, and by default it was enabled for me) plus no minimal brightness setting and no countdown.

Koenkk, 2023-10-20:
Try with:

```js
    fromZigbee: [...tuya.extend.light_onoff_brightness().fromZigbee, tuya.fz.datapoints],
    toZigbee: [...tuya.extend.light_onoff_brightness().toZigbee, tuya.tz.datapoints],
```

You can remove the datapoints `state` and `brightness` from `tuyaDatapoints`.

mlichvar, 2024-02-22:
Was anyone able to set the minimum brightness with z2m? The documentation says it can be set with a remote, but I don't have that.

When I try it with this definition:
````
const definition = {
    zigbeeModel: ['TS0501B'],
    model: 'TS0501B',
    vendor: '_TZB210_g01ie5wu',
    description: 'Automatically generated definition',
    extend: [tuyaLight({minBrightness: true})],
    meta: {},
};
````

I get this error:
````
genLevelCtrl({"64512":{"value":511,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

mlichvar, 2024-03-02:
With the Miboxer remote I can set the minimum brightness down to 3%. It would be nice if it could be set lower with Z2M. Here are some messages I get when I'm changing it with the remote.

````
zigbee/bridge/logging {"level":"debug","message":"No converter available for 'TS0501B' with cluster 'lightingColorCtrl' and type 'raw' and data '{\"data\":[8,47,10,2,241,72,12,1,0,0,0,0,0,12,0,12,0,3,0],\"type\":\"Buffer\"}'"}
zigbee/bridge/logging {"level":"debug","message":"No converter available for 'TS0501B' with cluster 'lightingColorCtrl' and type 'raw' and data '{\"data\":[8,54,10,2,241,72,12,1,0,0,0,0,0,12,0,12,0,5,0],\"type\":\"Buffer\"}'"}
zigbee/bridge/logging {"level":"debug","message":"No converter available for 'TS0501B' with cluster 'lightingColorCtrl' and type 'raw' and data '{\"data\":[8,61,10,2,241,72,12,1,0,0,0,0,0,12,0,12,0,7,0],\"type\":\"Buffer\"}'"}
````

The second value from the end is the minimum brightness. I saw a similar payload reported at https://github.com/Koenkk/zigbee-herdsman-converters/issues/6858, which has a value of 0 at that place. Any suggestions on how could it be set?

SkyDroVV, 2024-03-11:
Model TRI-C1ZR manufactured by MiBoxer
Zigbee ID: TS0501B | _TZB210_g01ie5wu

Is the device now supported by ZHA?

Subia75, 2024-03-14:
HI,
I was thinking of getting this dimmer, but I didn't understand if it is detected and can be used by zigbee2mqq.

thanks to anyone who can answer me.

mlichvar, 2024-03-14:
It works with Z2M out of the box. Switching on/off works, changing brightness works, it can even be switched to the relay mode except it's mapped to the "Do not disturb" setting. The only missing piece I think is setting of the minimum brightness. You would need the Miboxer remote to do that and the minimum seems to be 3%. As it arrived from the store, the minimum was set higher.

The Miboxer dimmer was one of the two I could source locally that has trailing edge dimming. Works very nicely with Philips WarmGlow bulbs. No buzzing or flickering. The other one was a dimLED dimmer, but it dimmed too slowly and its power consumption was high (about 1.3W).

Subia75, 2024-03-14:
Thanks for the quick response.
I ask a specific question.
Since you're using it, I can't find how much power it supports for a LED bulb and maybe it's written in the instructions.
I should connect a 220W 220V LED strip to it.

mlichvar, 2024-03-14:
The manual says 300W at 220V. I have not tried that. My load is only about 32W.

Subia75, 2024-03-14:
I ask because in most dimmers I find 200W for LEDs and 300W for halogen lights but in this one only 300w and I wouldn't want to burn it

mlichvar, 2024-06-14:
I was finally able to capture the command for setting the minimum brightness from a Tuya gateway. It can be set down to 0%, which is very nice. A PR is submitted in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7638. It works for me, but it will probably need more work.

mikedxb82, 2024-09-03:
Hello everyone, I've got 30 of these in my house alongside another 80 other zigbee dimmers from other brands.

This model is the only one that does not update the state to HA when the dimmer is activated with the physical switch button.

Is there any way to get the status of the dimmer updating HA if it's manually physically activated?

Thanks a lot in advance!

mikedxb82, 2024-12-01:
Hi i don't know if any maintener is looking at the thread.

These miboxer dimmers don't change the status, when toggled with the physical switch.

I'm happy to do try to look at it, but I need a bit of guidance to look where to find the events in the logs.

TheBazeman, 2024-12-21:
When you go to reporting tab of the device and put it like this the physical button reports the state to HA :-)

[image]

nazeem-soeltan, 2025-02-06:
> When you go to reporting tab of the device and put it like this the physical button reports the state and brightness to HA :-)
> 
> [image]

The screenshot, from which application is it?
Is it possible to set these values in/via Home Assistant

TheBazeman, 2025-02-07:
> > When you go to reporting tab of the device and put it like this the physical button reports the state and brightness to HA :-)
> > [image]
> 
> The screenshot, from which application is it? Is it possible to set these values in/via Home Assistant (ZHA)

These settings are possible with HA/Zigbee2MQTT. Not using ZHA so cant help you with that

diasrobson, 2025-03-26:
Hi then I just got this dimmer and I’ve put it in place but when pairing it to the my z2m it shows up as a TS0501B this gets it somewhat working but it seems to be missing some critical options has anyone had any joy getting it working as intended 

[image]
[image]
[image]
[image]

dankarization, 2025-05-27:
still relevant

closed 2025-08-04 by github-actions[bot]