Koenkk/zigbee2mqtt #19142: [Wrong device]: Miboxer and MiBoxer

dankarization, 2023-10-02:
### Link

-

### Model

-

### Description

-

### Vendor

MiBoxer

### Picture (link)

-

### Database entry

-

### Notes

These 4 devices:
https://www.zigbee2mqtt.io/devices/FUT039Z.html
https://www.zigbee2mqtt.io/devices/FUT035Z.html
https://www.zigbee2mqtt.io/devices/FUT036Z.html
https://www.zigbee2mqtt.io/devices/FUT038Z.html
are listed under 'Miboxer' vendor, but there are also several devices under 'MiBoxer' one. Both of them are the same company, but due to this typo they are listed as different ones in z2m. Just changing 1 letter will make catalogue a little bit better :)

dankarization, 2023-10-02:
Same with https://www.zigbee2mqtt.io/devices/ZXZIR-02.html. Listed under "AUBESS", when there are already a lot of devices under "Aubess"

dankarization, 2023-10-02:
https://www.zigbee2mqtt.io/devices/ZG_LED_DRIVER42CC.html "ENVILAR" instead of "[Envilar](https://www.zigbee2mqtt.io/supported-devices/#v=Envilar)"

dankarization, 2023-10-02:
https://www.zigbee2mqtt.io/devices/5120.2210.html [iluminize](https://www.zigbee2mqtt.io/supported-devices/#v=iluminize) instead of [Iluminize](https://www.zigbee2mqtt.io/supported-devices/#v=Iluminize)

dankarization, 2023-10-02:
https://www.zigbee2mqtt.io/devices/ZM-H7.html 
[ZemiSmart](https://www.zigbee2mqtt.io/supported-devices/#v=ZemiSmart) instead of [Zemismart](https://www.zigbee2mqtt.io/supported-devices/#v=Zemismart)

That's it. Cannot found any more

Koenkk, 2023-10-03:
Could you make a PR to https://github.com/koenkk/zigbee-herdsman-converters to fix this?

dankarization, 2023-10-03:
I am not very good at it. Can you link me to documentation howto?

dankarization, 2023-10-04:
Done

closed 2023-10-04 by dankarization