Koenkk/zigbee2mqtt #19147: [New device support]: Zigbee Computer Power - model 'TS0601' and manufacturer name '_TZE204_6fk3gewc'

LuizizM, 2023-10-03:
### Link

https://vi.aliexpress.com/item/1005002776442806.html?spm=a2g0o.order_list.order_list_main.11.6daccaa4F9lIDH&gatewayAdapt=glo2vnm

### Database entry

Device type Router Zigbee Model TS0601 Zigbee Manufacturer _TZE204_6fk3gewc Support status Not supported  IEEE Address 0xa4c1382e33144c13 Network address 0x56E1

### Comments

i tried to add this Zigbee adapter but im new and dont know how to get the data from the device and need help with it , is a power switch that allows you to turn on you PC remotely using zigbee, has child lock that dont allow it to be turned on/off from the computer switch only allowing to be turned remotely, also has a feature to bind a 433mhz remote to turn on and off the computer by pressing the remote as well.

Fenixekb, 2023-10-07:
Yes, I really need this converter too

LuizizM, 2023-10-31:
up

phoenixckua, 2023-11-18:
I need too

Evililim, 2023-11-23:
I have one too :)

In TUYA IoT Platform I have extracted this information :
> {"1":"Switch","101":"ModeReset","102":"RF remote control","103":"RF study","104":"buzzer","105":"Relay Status","106":"Child Lock","209":"Cycle Timing","210":"Random Timing"}

- Switch : to switch the computer on or off
- RF remote control : to enable or disable RF remote control use
- RF study : to pair a RF 433 remote, such as the one supplied with
- buzzer : to activate or deactivate buzzer sounds on relay switching
- Child Lock : to disable or enable the physical power buton of the computer

What other info can i provide to help ?

alex-shl, 2023-11-28:
i need too

Bildas, 2023-12-06:
Please, i need it too

Bildas, 2023-12-06:
Hi, please those who understand it, if you could take a look at it and write at least a converter for zigbee2mqtt, it would certainly make us all very happy :-) And thanks in advance

Fenixekb, 2023-12-13:
> "Hey guys, I finally did it myself. Main functions such as turning on the PC, child lock, on/off beeper are implemented. Finish the rest, it seems useless to me :-) PS: you know everything is for the zigbee2mqtt converter" [6FK3GEWC.txt](https://github.com/Koenkk/zigbee2mqtt/files/13637200/6FK3GEWC.txt)

Thank you very much!

BenJamesAndo, 2024-01-10:
Thank you @Bildas for the code.
I've expanded upon it to allow for pairing and using the RF remote that ships with the device. Needs a bit more testing but I was able to pair the remote.
[6FK3GEWC.txt](https://github.com/Koenkk/zigbee2mqtt/files/13882424/6FK3GEWC.txt)
For those unaware, rename it to `6FK3GEWC.js` and then upload it to be in the same folder as the `configuration.yaml` file for zigbee2mqtt. Then restart zigbee2mqtt and it should work.

If I get time I'll see if I can create a pull request so we can get official support for this.

BenJamesAndo, 2024-01-16:
Hey everyone, the PR I made is coming together. Hoping to finalise it soon. But I need a bit of testing done. I'm not sure what Mode Reset or Relay Status does. Also setting RF remote control to off doesn't disable the RF remote from still working. Below is an updated external converter that you can use to test. Perhaps I wired up my device wrong but if in fact these functions don't do anything then I'll remove them. I didn't add in Cycle Timing or Random Timing as that would probably involve putting in a number value and I'm not sure how to do that. Similar outcomes could likely be achieved with an automation anyway.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tzDatapoints = {
    ...tuya.tz.datapoints,
        key: ['switch', 'mode_reset', 'buzzer', 'child_lock', 'rf_remote_control', 'rf_study', 'relay_status']
}

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_6fk3gewc',
        },
    ],
	model: 'PCI E',
    vendor: 'WETEN',
    description: 'PC SWITCH',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    exposes: [
        e.binary('switch', ea.STATE_SET, 'ON', 'OFF').withDescription('PC Power'),		
        e.binary('mode_reset', ea.STATE_SET, 'ON', 'OFF').withDescription('Perhaps changes between Shutdown or Restart, does not seem to actually do anything'),
        e.binary('relay_status', ea.STATE_SET, 'ON', 'OFF').withDescription('Not sure what this does'),
        e.binary('buzzer', ea.STATE_SET, 'ON', 'OFF').withDescription('Buzzer on means no buzzer noise'),
	e.binary('child_lock', ea.STATE_SET, 'LOCK', 'UNLOCK').withDescription('Child safety lock'),
	e.binary('rf_remote_control', ea.STATE_SET, 'ON', 'OFF').withDescription('To enable or disable the use of RF remote control, does not seem to actually work'),
	e.binary('rf_study', ea.STATE_SET, 'ON', 'OFF').withDescription('To pair a RF 433 remote, such as the one supplied')
        ],
    meta: {
        tuyaDatapoints: [
            [1, 'switch', tuya.valueConverter.onOff],
            [101, 'mode_reset', tuya.valueConverter.onOff],
            [105, 'relay_status', tuya.valueConverter.onOff],
            [104, 'buzzer', tuya.valueConverter.onOff],
            [106, 'child_lock', tuya.valueConverter.lockUnlock],
            [102, 'rf_remote_control', tuya.valueConverter.onOff],
            [103, 'rf_study', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

kshexe, 2024-02-06:
How do I get into zigbee connection mode?

BenJamesAndo, 2024-02-06:
> How do I get into zigbee connection mode?

Long press the reset button for 5s until the LED indicator flashes three times, which means the device has entered pairing mode. If buzzer feedback is on then you'll also hear 3 short beeps.
The reset button can be found on the PCIe card.

I'll add the pairing instructions to the device page.

This device has now been officially added so there's no need to use the external converter file and this issue can be closed.

kshexe, 2024-03-02:
This is my device, but it doesn't stick to z2m after pairing mode.

Tuya PRO Long Zigbee. The seller said they don't support z2m.

ZenoBell, 2024-03-03:
There are only two  devices that are Zigbee compatibles and there are the same. The difference is one with a long plate for normal pc cases and the other with a short plate for mini atx builds. Seller i guess has outdated info and doesnt still know that i works with z2mqtt.  I still although dont see it on the supported devices page.

BenJamesAndo, 2024-03-03:
> This is my device, but it doesn't stick to z2m after pairing mode.

What do you mean it doesn't stick to z2m? Who is the seller? I did inform one of the sellers on Aliexpress that this device was now supported.

> I still although dont see it on the supported devices page.

This is the supported devices page: https://www.zigbee2mqtt.io/devices/PCI_E.html

kshexe, 2024-03-04:
Tuya PRO Long Zigbee model pairing is not possible. Is there a way to check?

closed 2024-10-02 by github-actions[bot]