Koenkk/zigbee2mqtt #19151: 064882 Legrand supports does not work on 001f firmware

ronangaillard, 2023-10-03:
### What happened?

I have four 064882 Legrand devices which work really well on 0017 firmware version (Cable outlet mode works).

I have one that is using the 001f firmware. Setting the output on/off (State) works well but setting Cable outlet mode does not work (nothing happens on the heater side).

### What did you expect to happen?

To be able to control the Cable outlet mode on the 001f firmware

### How to reproduce it (minimal and precise)

Update a 064882 Legrand to the 001f firmware with OTA, and try to change the Cable outlet mode 

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20210708

### Adapter

slae.sh CC2652RB

### Debug log

```
Zigbee2MQTT:debug 2023-10-03 02:21:07: Received MQTT message on 'zigbee2mqtt/0x000474000024bc14/set' with data '{"cable_outlet_mode":"comfort"}'
Zigbee2MQTT:debug 2023-10-03 02:21:07: Publishing 'set' 'cable_outlet_mode' to '0x000474000024bc14'
Zigbee2MQTT:info  2023-10-03 02:21:07: MQTT publish: topic 'zigbee2mqtt/0x000474000024bc14', payload '{"cable_outlet_mode":"comfort","device_mode":"pilot_on","linkquality":36,"power":0,"power_apparent":0,"state":"ON","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:debug 2023-10-03 02:21:10: Received MQTT message on 'zigbee2mqtt/0x000474000024bc14/set' with data '{"cable_outlet_mode":"eco"}'
Zigbee2MQTT:debug 2023-10-03 02:21:10: Publishing 'set' 'cable_outlet_mode' to '0x000474000024bc14'
Zigbee2MQTT:info  2023-10-03 02:21:10: MQTT publish: topic 'zigbee2mqtt/0x000474000024bc14', payload '{"cable_outlet_mode":"eco","device_mode":"pilot_on","linkquality":36,"power":0,"power_apparent":0,"state":"ON","update":{"state":"available"},"update_available":true}'
```

Tonio16, 2023-10-04:
Hello

Try with the 1.33.1. There is a fix regarding legrand OTA update. Legrand devices update should be started after you have disconnected and reconnected the power.
Vous devez passer en version 1.33.1 de Zigbee2mqtt car cette nouvelle version résout des soucis OTA pour Legrand.
De plus vous devez couper le courant puis le remettre avant de commencer la mise à jour. Cela active un mode mise à jour pour les dispositifs legrand. Par contre, je ne saurais dire combien de temps le dispositif reste réceptif aux mises à jour.

Antoine

ronangaillard, 2023-10-10:
Sorry for the delay.

Indeed it now works with the 1.33.1 and a new update.

I'm closing the issue.

closed 2023-10-10 by ronangaillard