Koenkk/zigbee2mqtt #19153: Error while starting zigbee-herdsman

skynetjk, 2023-10-03:
### What happened?

Hello

I have a problem, in sunday zigbee2mqtt stopped working,
the following message appears:

'''
Zigbee2MQTT:error 2023-10-01 21:31:34: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-10-01 21:31:34: Failed to start zigbee
Zigbee2MQTT:error 2023-10-01 21:31:34: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-10-01 21:31:34: Exiting...
Zigbee2MQTT:error 2023-10-01 21:31:35: Error: target adapter aps link key data table size insufficient (size=3)
'''

I have SLZB-06 via IP.
Below is my config:

'''
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt_lan
serial:
  port: tcp://192.168.91.204:6638
  baudrate: 115200
'''

I restarted HA, SLZB-06, reinstalled zigbee2mqtt and uploaded the firmware to SLZB-06 and nothing helps.

Can anyone help?

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

0.9.9

### Adapter

SLZB-06

### Debug log

Zigbee2MQTT:error 2023-10-01 21:31:34: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-10-01 21:31:34: Failed to start zigbee
Zigbee2MQTT:error 2023-10-01 21:31:34: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-10-01 21:31:34: Exiting...
Zigbee2MQTT:error 2023-10-01 21:31:35: Error: target adapter aps link key data table size insufficient (size=3)
    at AdapterBackup.restoreBackup (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-backup.ts:328:31)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at ZnpAdapterManager.beginRestore (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:299:9)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:80:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)

Arn0uDz, 2023-10-04:
I also had a lot of trouble getting the SLZB-06 too work. This is my config:

socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false

mqtt:

serial:
  port: tcp://192.168.1.138:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
advanced:
  transmit_power: 20

closed 2023-11-12 by github-actions[bot]

cinadr, 2024-09-24:
Is this issue ever resolved? Was there any workaround?

daschmidt1994, 2025-02-23:
same problem here

Bodya4Git, 2025-03-24:
Did anyone found any fix for this issue?

drgreenbaum, 2025-03-26:
This happened yesterday for me. Haven’t been able to solve it yet. Anyone?