Koenkk/zigbee2mqtt #19157: [Wrong device]: Moes BHT-006GBZB got recognized as Moes BHT-002/BHT-006

Krakish6, 2023-10-03:
### Link

https://www.aliexpress.com/item/1005005230261106.html

### Model

Moes BHT-006GBZB

### Description

Smart heating thermostat

### Vendor

Moes

### Picture (link)

https://www.zigbee2mqtt.io/devices/BHT-006GBZB.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c138762698a1ae","nwkAddr":36659,"manufId":4417,"manufName":"_TZE204_aoclfnxz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"c.�,g","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_aoclfnxz","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{"scenes":{"0_0":{"name":"Scene 0","state":{}}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1696354615464,"defaultSendRequestWhen":"immediate"}#

### Notes

After first pairing it was correctly detected as Moes BHT-006GBZB with correct picture but after removing and adding the device again its now detected as Moes BHT-002/BHT-006

Koenkk, 2023-10-04:
See https://github.com/Koenkk/zigbee2mqtt/issues/19091 for the reason.

closed 2023-10-04 by Koenkk

indigo2gm, 2025-01-25:
The problem has not been solved, to date it happens to me .