Koenkk/zigbee2mqtt #19161: Device update status is not displayed in Home Assistant

Laxilef, 2023-10-03:
### What happened?

After starting the OTA update, this is not displayed in Home Assistant.

Fix. Add after https://github.com/Koenkk/zigbee2mqtt/blob/7e63039144ec2adcce3cb076c9ea68d633d1596a/lib/extension/homeassistant.ts#L1186
```
                    json_attributes_topic: `${settings.get().mqtt.base_topic}/${entity.name}`, // state topic
                    json_attributes_template: `{"in_progress": "{{ value_json['update']['state'] == 'updating' }}"}`,
```

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230507

### Adapter

CC2652

Koenkk, 2023-10-04:
Could you make a PR for this?

Laxilef, 2023-10-05:
> Could you make a PR for this?

I'm not sure I'll do everything right. It will be better if this is done by someone who is familiar with the z2m codebase. Thx.

Koenkk, 2023-10-06:
It's not complicated, you can press the Edit icon here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.ts

closed 2023-10-08 by Koenkk