Koenkk/zigbee2mqtt #19168: Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

madner81, 2023-10-04:
### What happened?

Hi all,
I've a question about a log entry: I've just completed the setup of the addon that uses a SONOFF E Dongle. It seems that everything is woking fine (I've paired just one device at the moment), but every time I start the the zigbee2mqtt, in the log tab of the addon page in Home Assistant, I have an assertion failed entry.

Looking at the log int the /config/zigbee2mqtt/log/ folder, the Assertion Failed entry in not present.

Do you know which is the meaning of this entry and if it represents a potential problem? Considering that I've just paired just one device, I'm still in time to switch to ZHA.

Any help will be really appreciated.

Have a nice day.

### What did you expect to happen?

Have a log with no errors

### How to reproduce it (minimal and precise)

Just start

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus V2 model "ZBDongle-E"

### Debug log

Zigbee2MQTT:info 2023-10-04 11:41:23: Starting Zigbee2MQTT version 1.33.1 (commit #unknown)
Zigbee2MQTT:info 2023-10-04 11:41:23: Starting zigbee-herdsman (0.19.0)
Zigbee2MQTT:debug 2023-10-04 11:41:23: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[XX],"extendedPanID":[XXX,XXX,XX,XXX,X,XXX,XXX,XXX],"networkKey":"HIDDEN","panID":XXXX},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_XXXXXXXXXXXXXX-XXXX"}}'
**Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]**
Zigbee2MQTT:info 2023-10-04 11:41:27: zigbee-herdsman started (resumed)

merlinpimpim, 2023-10-15:
Same "problem" with a Skyconnect Dongle !

Tarik2142, 2023-10-24:
[image]
Version 1.30.3 starts

madner81, 2023-10-24:
1.33.1 starts too, I wrote that everything works fine with the only exception of the assertion failed entry in the log

merlinpimpim, 2023-10-24:
Sometime i have the exception of the assertion and z2m won't start. Then I have to launch z2m in debug mode (1 time).

Palthe, 2023-11-04:
I also appear to have the same error with the kyconnect Dongle : 

[22:15:39] INFO: Preparing to start...
[22:15:39] INFO: Socat not enabled
[22:15:40] INFO: Starting Zigbee2MQTT...
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]

madner81, 2023-11-05:
UPDATE: now it's randomly happening while the addon is working too, not only in the start phase

stewe93, 2023-12-04:
Maybe a similar issue here: 
I'm using unraid, updated the image and the container fails to start and stuck at this phase: 
```
Using '/app/data' as data directory
Zigbee2MQTT:info  2023-12-04 12:37:03: Logging to console and directory: '/app/data/log/2023-12-04.12-37-03' filename: log.txt
Zigbee2MQTT:info  2023-12-04 12:37:03: Starting Zigbee2MQTT version 1.34.0-dev (commit #5e3bd75)
Zigbee2MQTT:info  2023-12-04 12:37:03: Starting zigbee-herdsman (0.25.2)
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: 55
```
It should restart automatically but it doesn't. 

After this, if I restart the container manually, its fine and everything works as expected.

I'm using the Zigbee Dongle-E with multipan fw, if I remember correctly 4.3.1 fw version, configuration.yaml relevant part: 
```
serial:
  port: tcp://localhost:20108
  adapter: ezsp
```

cvladan, 2023-12-08:
Same problem. Is there a way to downgrade addon?

stevenh, 2023-12-14:
Enabling debug, the error appears to occur when configuring the buffer count to `255` and the status code is `55`.

```text
  zigbee-herdsman:adapter:ezsp:ezsp Set EzspConfigId.CONFIG_PACKET_BUFFER_COUNT = 255 +0ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> setConfigurationValue: {"configId":1,"value":255} +1ms
  zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":true,"configId":1,"value":255} +0ms
  zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 0c8001530037 +5ms
  zigbee-herdsman:adapter:ezsp:ezsp <== 0x53: {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":false,"status":55} +0ms
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
```

Editing the assertion to be more verbose I got:
```
Assertion failed: Command (setConfigurationValue) EzspConfigId.CONFIG_PACKET_BUFFER_COUNT = 255 returned unexpected state: {"_cls_":"setConfigurationValue","_id_":83,"_isRequest_":false,"status":55}
```

stevenh, 2023-12-14:
Seems to be related to [this addon issue](https://github.com/home-assistant/addons/issues/3166)

merlinpimpim, 2023-12-16:
Some news ?

`[01:04:38] INFO: Preparing to start...
[01:04:39] INFO: Socat not enabled
[01:04:44] INFO: Starting Zigbee2MQTT...
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: 55`

With Skyconnect / Silicon Labs Multiprotocol

PetziAt, 2023-12-19:
[12:17:22] INFO: Preparing to start...
[12:17:22] INFO: Socat not enabled
[12:17:25] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-19 12:17:31: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2023-12-19 12:17:31: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-19.12-17-31' filename: log.txt
Zigbee2MQTT:info  2023-12-19 12:17:31: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-19 12:17:31: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-19 12:17:31: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[112,15,219,255,208,218,208,223],"networkKey":"HIDDEN","panID":18429},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202092823-if00"}}'
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Zigbee2MQTT:info  2023-12-19 12:17:34: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2023-12-19 12:17:34: Coordinator firmware version: '{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"3","product":12,"revision":"7.3.1.0 build 176"},"type":"EZSP v12"}'
Zigbee2MQTT:debug 2023-12-19 12:17:34: Zigbee network parameters: {"channel":11,"extendedPanID":112,"panID":18429}
Zigbee2MQTT:info  2023-12-19 12:17:34: Currently 0 devices are joined:
Zigbee2MQTT:info  2023-12-19 12:17:34: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2023-12-19 12:17:35: Connecting to MQTT server at mqtt://192.168.192.10:1883
Zigbee2MQTT:debug 2023-12-19 12:17:35: Using MQTT login with username: mqtt
Zigbee2MQTT:info  2023-12-19 12:17:35: Connected to MQTT server
Zigbee2MQTT:info  2023-12-19 12:17:35: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2023-12-19 12:17:35: Started frontend on port 8099
Zigbee2MQTT:info  2023-12-19 12:17:35: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":"1 ","majorrel":"7","minorrel":"3","product":12,"revision":"7.3.1.0 build 176"},"type":"EZSP v12"},"log_level":"debug","network":{"channel":11,"extendedPanID":112,"panID":18429},"permit_join":false,"version":"1.34.0"}'
Zigbee2MQTT:info  2023-12-19 12:17:35: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2023-12-19 12:17:35: Zigbee2MQTT started!

dierochade, 2024-01-02:
I got this error (for the first time) after upgrade to 1.35.01. 

my hardware is skyconnect, multipan firmwore on ha supervised/vmware. 

Reverted to prior version, what worked

IgnacioHR, 2024-01-02:
@dierochade which version works for you? 

Here are my thoughts on this:
- It appears in EZSP devices when Coordinator Firmware version is 7.3.1.0. 
- As stated [here](https://github.com/home-assistant/addons/issues/3166#issuecomment-1667824342) the problem is related with the initialisation of the coordinator

I'm not sure if this is related with z2m or related with the Zigbee-herdsman library. I don't know how is responsible of the coordinator initialisation in the code.

I believe it could be considered a WARNING only. This message shouldn't block the coordinator to work ok at all.

merlinpimpim, 2024-01-02:
> I got this error (for the first time) after upgrade to 1.35.01.
> 
> my hardware is skyconnect, multipan firmwore on ha supervised/vmware.
> 
> Reverted to prior version, what worked

I have the same hardware and the same configuration (supervised,  but on a Khadas) and I already had the problem with previous versions (and it continues with version 1.35).

merlinpimpim, 2024-01-02:
> @dierochade which version works for you?
>
> Here are my thoughts on this:
>
> * It appears in EZSP devices when Coordinator Firmware version is 7.3.1.0.
> * As stated [...]

OK but I think that in previous versions of Z2M (<1.35) this error message was blocking.
I remember having to run Z2M a few times in debug mode to get the coordinator to work properly.

xhemp, 2024-01-05:
Just started seeing this on my side as well with the latest zigbee2mqtt, multipan firmware on a dongle E.

JigSawFr, 2024-01-05:
Also facing this bug with SkyConnect and latest upgrade of the Z2M addon... but Z2M don't work anymore.

MarkMikkelsen83, 2024-01-05:
Same issues here :(

nyok92, 2024-01-05:
Same issue here with latest zigbee2mqtt version ( 1.35.0-1 ), multipan firmware (2.4.0)on a dongle E.

sixfact, 2024-01-05:
Same issue with zigbee2mqtt version ( 1.35.0-1 ), multipan firmware (2.4.0)on a dongle E.
Correct by reverse to multipan firmware 2.3.2

aLimp86, 2024-01-05:
I have the same problem now. Sky connect and multiprozocol enabled. 
is there a solution for this? Downgrading what and how?

sixfact, 2024-01-05:
> I have the same problem now. Sky connect and multiprozocol enabled. is there a solution for this? Downgrading what and how?

In my case, I downgrade Silicon Labs Multiprotocol addon from 2.4.0 to 2.3.2 by restoring a backup :  addon_core_silabs_multiprotocol_2.3.2 (System > Backup >addon_core_silabs_multiprotocol_2.3.2 >restore)

yvos, 2024-01-05:
Same here since this morning. After upgrading HA to the latest version.

aLimp86, 2024-01-05:
> In my case, I downgrade Silicon Labs Multiprotocol addon from 2.4.0 to 2.3.2 by restoring a backup :  addon_core_silabs_multiprotocol_2.3.2 (System > Backup >addon_core_silabs_multiprotocol_2.3.2 [...]

Worked also for me. After downgrade und restart of Home assistant it works again. That was an easy fix.

aLimp86, 2024-01-05:
Ok at least it started. But I can't send a command now. No lamp is turning on. 
Someone still with the same problem?

hensjl, 2024-01-05:
> Worked also for me. After downgrade und restart of Home assistant it works again. That was an easy fix.
Easy fix but one hour without services. All is OK now.

samoz83, 2024-01-05:
> Edit: Ok at least it started. But I can't send a command now. No lamp is turning on. Someone still with the same problem?

Yep, same here, commands time out (multipan firmware, E dongle)

aLimp86, 2024-01-05:
There is a weird solution for our problem in another forum. 
You have to shutdown everything, unplug your device for 5s (e.g. raspberry) and startup again. That should fix it, but have not tried yet.

Seger85, 2024-01-05:
Join me, Sky connect and multiprozocol disable. Latest firmware, new Z2M version

nyok92, 2024-01-05:
Downgrading did not solved this issue for me. timeout: Z2MQTT still not working.

Maybe the next Firmware release 0.2.0 from [multipan_flasher](https://github.com/darkxst/multipan_flasher/tree/4.4) with the update for GeckoSDK 4.4.0 will solve the issue as the new Silicon Labs Multiprotocol addon 2.4.0 use this new GeckoSDK 4.4.0.

Seger85, 2024-01-05:
> Ein Downgrade hat dieses Problem für mich nicht gelöst. Timeout: Z2MQTT funktioniert immer noch nicht.
> 
> Möglicherweise wird die nächste Firmware-Version 0.2.0 von [multipan_flasher](https://github.com/darkxst/multipan_flasher/tree/4.4) mit dem Update für GeckoSDK 4.4.0 das Problem lösen, da das neue Silicon Labs Multiprotocol Addon 2.4.0 dieses neue GeckoSDK 4.4.0 verwendet.

unfortunately it will not, I have just installed the new firmware, nothing works.

Seger85, 2024-01-05:
Es ist auch nicht das Multiprotocol, ich z.B. laufe ohne, der selbe Fehler.
Es gab ja auch einen Hotfix von Z2M, aber der hat nichts gebracht.

nyok92, 2024-01-05:
By the way, my thread/Matter network is working correctly with Silicon Labs Multiprotocol addon 2.4.0.
Switched from Z2MQTT to ZHA as a temporary workaround (to have some light :) ): working correctly

kurdybanek, 2024-01-05:
Same problem here.
After downgrade of Silabs multiprotocol I am continously receiving ping errors to various devices.
As a result if I try to switch some device on/off it sometimes work, sometimes not. Looks like traffic is jammed and sometimes command goes through that jam and sometimes it is lost.
Perhaps in that case a quasi soft reset of the stick (plugging it off for few seconds) may help, I can't check right now.

In my case the update went at the same time as the system update, and I'm not sure if that update was not part of the issue. For example I had issues with the second HA instance, where I didn't perform update of the Silabs Multiprotocol addon. 
In that case rebooting HA AND unpluging/pluging the stick solved the problem. But I have only 2 zigbee devices at that location.

> There was also a hotfix from Z2M, but it did nothing.
@Seger85  - what hotfix do you mean, and how to install it?

aLimp86, 2024-01-05:
>  Z2M 1.35.0-1 1 was added later and installed automatically

Is it then possible to update Multiprotocol software to 2.4.0?

kurdybanek, 2024-01-05:
Don't think so. I have 1.35.0-1 already (so I probably missed the original 1.35.0) and after upgrading multiprotocol I had the issue of "unexpected state: [object Object]" :(

ucito, 2024-01-05:
Also for me: 
Sonoff-E /Silicon laps multiprotocol for thread and after update 2.4.0 zigbee2mqtt could not start again, or in the log all zigbee devices had lost contact.

Repair attempts carried out:
-partial restore multiprotocol to 2.3.2 and zigbee2mqtt -did not help

-shutdown hass, physically remove dongle for a few minutes and reconnect -did not help
-after some time at least the zigbee2mqtt device overview page was displayed again, but no device could be controlled

-after waiting half an hour, I tried to register  a new bulb in zigbee2mqtt for fun and it worked. 

And the other zigbee devices worked again too!       (multiprotocol 2.3.2)

samoz83, 2024-01-05:
> ﻿And the other zigbee devices worked again too! (multiprotocol 2.3.2)
After seeing your post, tried my devices again too and are now working on 2.3.2

loganfuller, 2024-01-05:
I have had this issue for several weeks as well. I solve it by restarting my raspberry pi or restarting the Z2M addon over and over until it randomly works.

Drealine, 2024-01-05:
Same issue here since update multipan addon.

nic0dk, 2024-01-05:
same here, how do i downgrade multipan addon ?

jdesai61, 2024-01-05:
Same here. Need info on downgrading multipan addon.

aLimp86, 2024-01-05:
> Also for me:
>
> Sonoff-E /Silicon laps multiprotocol for thread and after update 2.4.0 zigbee2mqtt could not start again, or in the log all zigbee devices had lost contact.
>
>
>
> Repair attempts carried [...]

Worked for me. Everything is up and running again. 
But what strange is that if you delete the newly added device again everything stops working. At least for me.

jdesai61, 2024-01-05:
Just downgraded multipan addon to 2.3.2 and can confirm that z2m is working again.
Steps to downgrade in HA:
```
1. Bring up list of backups using Settings -> System -> Backups
2. Click a backup from the list that was created before 2.4.0 upgrade (I picked one from last week - hope you have one available)
3. Resulting window (see attached), choose "Silicon Labs Multiprotocol (2.3.2)" and click "Restore"
4. Confirm that you only want the partial restore.

```
[image]

puddly, 2024-01-05:
I **strongly** suggest you migrate away from multi-protocol + Z2M (or disable auto updates). The multi-protocol addon tracks the absolute latest version of the SDK and without constant updates to Z2M's EZSP driver, you will continue having breakage.

loganfuller, 2024-01-05:
@jdesai61 

> Just downgraded multipan addon to 2.3.2 and can confirm that z2m is working again. Steps to downgrade in HA:
>
> ```
> 1. Bring up list of backups using Settings -> System -> Backups
> 2. Click a backup [...]

I tried this and it wiped out my Z2M config - all devices gone.

jdesai61, 2024-01-05:
> I tried this and it wiped out my Z2M config - all devices gone.

Restoring Silicon Labs Multiprotocol Add-on should have no effect on z2m config.  I followed above steps and didn't have any issues.

loganfuller, 2024-01-05:
I thought the same, but no dice. And of course I didn't bother with a full backup before I tried it...

samoz83, 2024-01-05:
I notice while certain functions do now work after downgrading, all my devices last seen are N/A

loganfuller, 2024-01-05:
For anyone who has more understanding of Zigbee2MQTT and Home Assistant than me: where is the Z2M database stored? After rolling back the silicon labs multiprotocol addon back it caused all devices but one to disappear from the Z2M UI, and it seems like no matter which partial backup I roll back to with Zigbee2MQTT it always just loads with that single device. My original setup had 21.

jdesai61, 2024-01-05:
> I thought the same, but no dice. And of course I didn't bother with a full backup before I tried it...

I assume the devices are still paired.  You just lost their custom names.

loganfuller, 2024-01-05:
> > I thought the same, but no dice. And of course I didn't bother with a full backup before I tried it...
> 
> I assume the devices are still paired. You just lost their custom names.

If that was the case, would they not still appear in the Z2M add-on UI?

jdesai61, 2024-01-05:
> If that was the case, would they not still appear in the Z2M add-on UI?

Then something else is going on - because backup/restore doesn't unpair devices. Somehow you managed to 'reset' or 'reinitialize' the zigbee stick. I don't think backup/restore can cause this.

loganfuller, 2024-01-05:
Well at least the devices still show up in Home Assistant, though marked as unconnected to MQTT. Time to get re-pairing...

loganfuller, 2024-01-05:
On second look, the devices ARE still there under the `Settings -> Raw` tab of the Z2M UI, they just aren't populating under the dashboard. I'm also continuing to get the following error on startup:

```
Zigbee2MQTT:info  2024-01-05 16:31:10: Starting zigbee-herdsman (0.30.0)
Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]
Assertion failed: Command (setValue) returned unexpected state: 55
```

I wonder if they are related?

jdesai61, 2024-01-05:
Also, it seems like a new 2.4.1 Silicon Labs Multiprotocol add-on version 2.4.1 has been pushed that should work with HA and Z2M again.
```
Version 2.4.1
Revert back to Silicon Labs Gecko SDK 4.3.1 to maintain compatibility with Home Assistant Core 2023.12.0 and Zigbee2MQTT. If you are running 2024.1.0 with ZHA, you don't have to downgrade.
```

Seger85, 2024-01-05:
In my opinion, the whole thing only has something to do with the multiprotocol, I and others who did not have the multiprotocol installed (adapter EZ..) have the same problem. 

My solution was as follows: 

I stopped the addon Z2M, deleted almost all files in the folder `config/zigbee2mqtt/`. I left the `configuration.yaml` file in there. After that, the addon starts cleanly, **but then everything had to be learnt again**. But this was quicker than usual :-), as the user-defined names for HA are taken from `Configuration.yaml`. After that, all the automation worked straight away, as the names were identical to the devices again. Time-consuming, but then it worked again for me.

kkuriata, 2024-01-05:
I can confirm that removing all files except `configuration.yaml` in `config/zigbee2mqtt/` worked for me. I have multriprotocol enabled on SkyConnect

JigSawFr, 2024-01-05:
> In my opinion, the whole thing only has something to do with the multiprotocol, I and others who did not have the multiprotocol installed (adapter EZ..) have the same problem.
> 
> My solution was as follows:
> 
> I stopped the addon Z2M, deleted almost all files in the folder `config/zigbee2mqtt/`. I left the `configuration.yaml` file in there. After that, the addon starts cleanly, **but then everything had to be learnt again**. But this was quicker than usual :-), as the user-defined names for HA are taken from `configuration.yaml`. After that, all the automation worked straight away, as the names were identical to the devices again. Time-consuming, but then it worked again for me.
> 
> Edit: I'm always talking about the `configuration.yaml` by which I mean the one in the `config/zigbee2mqtt/` folder, as written above. Not that you are thinking of the other one.

Done the same, pretty straight forward :) Better than many hours troubleshooting issues, upgraded all addons, and followed your advice 👍

[38 more replies]

closed 2024-10-13 by github-actions[bot]