Koenkk/zigbee2mqtt #19169: [New device support]: Cable outlet with pilot wire and consumption measurement

yoan-m, 2023-10-04:
### Link

https://www.leroymerlin.fr/produits/chauffage-et-ventilation/radiateur/radiateur-electrique/kit-de-demarrage-economie-energie/recepteur-connecte-chauffage-fil-pilote-87766424.html

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x187a3efffeb1f880","nwkAddr":1984,"manufId":4727,"manufName":"Adeo","powerSource":"Mains (single phase)","modelId":"SIN-4-FP-21_EQU","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,4096,64512],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-FP-21_EQU","manufacturerName":"Adeo","powerSource":1,"zclVersion":8,"appVersion":3,"stackVersion":103,"hwVersion":20,"dateCode":"2021"}},"haElectricalMeasurement":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}},"piAnalogInputReg":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff048357","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff048357","endpointID":1},{"cluster":64576,"type":"endpoint","deviceIeeeAddress":"0x00212effff048357","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":103,"hwVersion":20,"dateCode":"2021","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1696430383834,"defaultSendRequestWhen":"immediate"}

### Comments

It will probably looks like Legrand 064882

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SIN-4-FP-21_EQU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SIN-4-FP-21_EQU', // Vendor model number, look on the device for a model number
    vendor: 'ADEO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Cable outlet with pilot wire and consumption measurement', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.legrand_cluster_fc01, fz.legrand_cable_outlet_mode, fz.on_off, fz.electrical_measurement, fz.power_on_behavior],
    toZigbee: [tz.legrand_deviceMode, tz.legrand_cableOutletMode, tz.on_off, tz.electrical_measurement_power, tz.power_on_behavior],
    exposes: [e.binary('device_mode', ea.ALL, 'pilot_on', 'pilot_off'),
            e.enum('cable_outlet_mode', ea.ALL, ['comfort', 'comfort-1', 'comfort-2', 'eco', 'frost_protection', 'off']),
            e.switch().withState('state', true, 'Works only when the pilot wire is deactivated'),
            e.power().withAccess(ea.STATE_GET),
            e.power_apparent(),
            e.power_on_behavior().withDescription(`Controls the behavior when the device is powered on. Works only when the pilot wire is deactivated`)],
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'manuSpecificLegrandDevices2']);
      await reporting.onOff(endpoint);
      await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
      await reporting.activePower(endpoint);
      await reporting.apparentPower(endpoint);
    },
};

module.exports = definition;
```

hcharbonnier, 2023-10-06:
This device seems to be the Nodon SIN-4-FP-21.
The only reference on nodon website is on this page: https://nodon.fr/gateway-compatibility/

yoan-m, 2023-10-06:
I'm able to metering & On/Off now but I don't know how to change  the heat mode.
Any idea ?

yoan-m, 2023-10-06:
It seems to be on cluster 0xfc40 but i'm not able to find the type and format

yoan-m, 2023-10-06:
I have some data on 0xfc00

yoan-m, 2023-10-07:
When I turn on the module, 1::64512::0 is set to 1, and when I turn off 1::64512::0 is set to 3.
Now I would like to manualy change this value to control the pilot wire.
I tried to create a toZigbee function and write on the cluster without succes
``

embryooo, 2023-10-08:
Hi, I'm new to HA with Z2M but would like to manage my heaters with this device. I bought one but it's not supported obviously. It appears as SIN-4-FP-21_EQU from Adeo.

[image]

If I can help by testing it's with pleasure (unfortunately don't ask me to code plz 👎 )

hcharbonnier, 2023-10-10:
We may get hint on the device configuration at this URL:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7280

Zoa-home, 2023-10-24:
> Hi, I'm new to HA with Z2M but would like to manage my heaters with this device. I bought one but it's not supported obviously. It appears as SIN-4-FP-21_EQU from Adeo.
> 
> [image]
> If I can help by testing it's with pleasure (unfortunately don't ask me to code plz 👎 )

Follow this link : [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)

and replace @yoan-m code "External converter"

yoan-m, 2023-10-27:
Hi, I just received an email from nodon company.

> Integration is in progress. It will be done within a few days.

hcharbonnier, 2023-10-27:
> Hi, I just received an email from nodon company.
> 
> > Integration is in progress. It will be done within a few days.

Same here

yoan-m, 2023-11-01:
It's moving here https://github.com/Koenkk/zigbee-herdsman-converters/pull/6381

krikristoophe, 2023-11-08:
Hello @yoan-m and others. Both pending PRs (https://github.com/Koenkk/zigbee-herdsman/pull/796 and https://github.com/Koenkk/zigbee-herdsman-converters/pull/6434) allow to change mode of `SIN-4-FP-21-EQU` ! Just waiting feedback or merge.

## Zigbee infos

I wrote Zigbee infos of the module here if anyone need them one day :

- Cluster : (`manuSpecificUbisysDeviceSetup2`) `0xfc00`
- endpoint: `1`
- command: `0x00`
- manufacturer code : `(NodOn) 0x128b`

### data

- off: `0x00`
- hors gel: `0x03`
- eco: `0x02`
- confort -1 : `0x04`
- confort -2 : `0x05`
- confort : `0x01`

yoan-m, 2023-11-08:
Thanks @krikristoophe, here is the PR for the SIN-4-FP-21 from NodOn
 https://github.com/Koenkk/zigbee-herdsman-converters/pull/6429
It seems to be the same device

krikristoophe, 2023-11-10:
Hello everyone
[The PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6429) that integrate this device has been merged. But it look like that not working because of the bad payload sent to Zigbee 🥲

I will make another PR to fix this !

yoan-m, 2023-11-10:
[Work in progress here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6429)

closed 2023-11-10 by yoan-m

syl20cjs, 2023-12-08:
Hello I successfully pair it on zigbee2mqtt, I Can see the consumption value on home assistant but I am not able to change the state or to use the switch...

It works when I push the physical button to change mode.

Is there a solution ?

Here are the logs I have in case of an error
Error 2023-12-08 19:56:48Publish 'set' 'state' to 'radiateur_leo' failed: 'Error: Command 0xf4b3b1fffeb3e52b/242 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 32138 - 242 - 165 - 6 - 11 after 10000ms)'

Regards

carlofrancese, 2023-12-14:
Exactly in the same situation as @syl20cjs
All visible but as soon as I try to switch on/off or change mode I've a timeout. The physical button works thus no issue with the cabling

baz-fr, 2023-12-14:
Hello @syl20cjs , @carlofrancese 
Have you try with a simple MQTT command ( like MQTT Explorer ) ?
I use this device with a Jeedom integration. Work perfectly.

carlofrancese, 2023-12-14:
> Hello @syl20cjs , @carlofrancese Have you try with a simple MQTT command ( like MQTT Explorer ) ? I use this device with a Jeedom integration. Work perfectly.

Hi @baz-fr 
after you mentioned MQTT Explorer i tried to publish messages as raw and json like "{ "mode": "comfort","state": "ON"}" (if it makes sense - i took inspiration from the read messages) but nothing changed.

baz-fr, 2023-12-17:
No need to set state, this is implicit.
`{"mode":"comfort"}` change state to "on" (if you set "state" to "off" , this turn "mode"  to "anti-freeze" )
I think you need to reset the device and retry the association.
[image]

syl20cjs, 2023-12-17:
On my side it started working without any action on my side After like 48h, but since then it works flawlessly

baz-fr, 2023-12-17:
@carlofrancese have you the latest version? (correction about this device)

carlofrancese, 2023-12-20:
Hi Everyone,
don't know how but as for @syl20cjs it all started to work properly one day after without any rewiring or modification in configuration.

dinde, 2024-01-24:
Hello,
Exactly the same behaviour here.
Adding + interview = OK
Trying to control results a permanent fail
I pressed the physical button on module and state is reported as below:
    "energy": 0.004,
    "last_seen": "2024-01-24T08:58:13.385Z",
    "linkquality": 102,
    "mode": "anti-freeze",
    "power": 515,
    "state": "ON",
    "pilot_wire_mode": "comfort"
}
Tried to add / delete / force interview / pair with another router all fails.
Some people also reported that at some point it should start to work (outside this thread).

MichaelMure, 2024-03-24:
Same here. I have three of those (same firmware). 2 work just fine, but the third has that timeout behavior.

It is plugged into a different brand of heater, but I don't see how that can affect anything.

MichaelMure, 2024-04-03:
I believe I've found a big clue for that timeout issue: https://github.com/Koenkk/zigbee2mqtt/issues/22068

morcus, 2024-04-04:
I found a solution for the timeout errors here: https://github.com/Koenkk/zigbee2mqtt/discussions/15683

You need to shut down zigbee2mqtt, edit `database.db` and modify the entry of the broken device from `[242,1]` to `[1,242]`.