Koenkk/zigbee2mqtt #19173: BRT-100-TRV flash ZT5

jonathan-gatard, 2023-10-04:
### What happened?

The problem is already known : https://github.com/Koenkk/zigbee2mqtt/issues/17313
My device is not responding after bad OTA update.

I have arduino, esp32, UART/USB key, everything what I need to flash the ZT5.

I just need :
- 1 dump of good version
- Pin schema of ZT5

Is someone able to help me ?

Thanks.

### What did you expect to happen?

Flash the ZT5 module with good firmware

### How to reproduce it (minimal and precise)

Don't do it, but to reproduce it, just update via OTA your TRV

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (zStack3x0) (ZBDongle-P)

sgilyin, 2023-10-06:
Yesterday I run ota update on my 3 TRV 's.
All of them are blocked zigbee now.
How can I downgrade firmware or enable zigbee now?
[image]
[image]
[image]
[image]

jonathan-gatard, 2023-10-06:
Hi @sgilyin,
We just need someone able to dump the ZT5 firmware for the moment.
After that, i'm pretty sure we'll find a solution to upload the firmware to our ZT5 :)

henkiejan1, 2023-10-08:
Yeah, ik broke also my TRV with the firmware upgrade. Maybe a good idea to block the OTA update for this valve @Koenkk ? 

Anyways. @jonathan-gatard but there is no procedure to dump the firmware? You can't use blackmagic or any tool to connect to this module what i read... So if there was a firmware dump it's impossible to write it back?

jonathan-gatard, 2023-10-08:
Hi @henkiejan1 

Check this doc : [https://developer.tuya.com/en/docs/iot/overview-of-download-firmware-and-authorization?id=Ka4zcqfo0eb5u](url)
If I understand well we have 2 possibilities:
[image]

-UART
VCC: 5 or 10
GND: 7 or 12
RX: 4 (or 8?)
TX: 3 (or 11?)

-"J-Link" module

I never did it, I just read the docs, maybe it's more complicated than I think... or maybe not 😄

Koenkk, 2023-10-08:
Could you provide the data/database.db entry of this device?

MattWestb, 2023-10-08:
Looks like its possible reading and writing the Z2 chip and one user have doing it with one tuya ZT3L so shall being possible do the same with the ZT5.
https://github.com/doctor64/tuyaZigbee/blob/master/docs/flash.md

jonathan-gatard, 2023-10-10:
@Koenkk, we don't have it in database because we cannot pair it anymore
However I found some people who share their data/database.db with their working TRV :
`{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c13808df96d72d","nwkAddr":43117,"manufId":4417,"manufName":"_TZE200_b6wax7g0","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_b6wax7g0","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1696957911447,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2023-10-10:
@jonathan-gatard thanks, I've removed it yesterday as part of https://github.com/Koenkk/zigbee2mqtt/issues/18840

jonathan-gatard, 2023-10-10:
Thanks 👍🏻
Sorry but could we open again the topic, we must repair our devices, we still need a dump, and a way to flash our ZT5. I think it's the best place to ask this, if it's not a problem for you.

ikarisan, 2023-10-11:
How can I open the BRT-100-TRV device? I removed the two srews, but nothing else is moveable. :(

jonathan-gatard, 2023-10-12:
Hi @ikarisan,
Which screws are you talking about ?

What I have done :
Immerse the head of the valve in boiling water, surrounding it with cling film to prevent it from getting wet for a few minutes.
Once hot, with a scalpel blade I took off the touchscreen. You will see 2 screws, t is these one that must be removed.
Then you just have to pull on it.

ikarisan, 2023-10-12:
> Hi @ikarisan, Which screws are you talking about ?
[image]
 This two screws 😊

ikarisan, 2023-10-14:
> ...we still need a dump, and a way to flash our ZT5. ...

I think the problem is, that nobody wants to open (and maybe damage) its well working device to dump the firmware. :(

jonathan-gatard, 2023-10-16:
It's not a problem for me to buy a new one, open it and dump the firmware, but I just need to know how to dump it

ikarisan, 2023-10-19:
> It's not a problem for me to buy a new one and open it, I just need to know how to dump it.

I guess, this should work: https://github.com/Koenkk/zigbee2mqtt/issues/10140#issuecomment-1006876103

jonathan-gatard, 2023-10-19:
I'm not sure, the chip is not the same, mine is ZT5 and pins are on the left and bottom.

ikarisan, 2023-10-19:
> Maybe but the chip is not the same. Mine is a ZT5 and the pins are on the left and the bottom.

Yes. But maybe all you need is VCC, GND, RST and the SWS pin.
https://developer.tuya.com/en/docs/iot/ZT5-Module-Datasheet?id=Kan388vhu20ar 

The TYZS3 module in the linked post is connected to

+3.3V
GND
RESET
SWDIO
SWCLK

But I have no experiences with these Tuya modules.

jonathan-gatard, 2023-10-26:
Hello,
I tried but without success...
[image]

I tried with VCC/GND/RX/TX + shunt RST to ground to "enable flash mode"

MattWestb, 2023-10-26:
Its looks like the Comport method is not reable and must using SWD / J-Tag for doing it  with success if reading the linked pages.

sktoroid, 2023-11-24:
Has anyone found a working way to revive the ZT5 module?
I also stepped on a rake with OTA in z2m :( and now I have 8 bricks :(

ilvits, 2023-11-24:
I have working one. But I don't know how to dump it and then upload the firmware to non-working ones.

ikarisan, 2023-11-24:
I was not able to find out quickly how to flash the ZT5 modules at all. There is a lot of documentation on developer.tuya.com. But if and how you can read out a flashed module I have not found. It seems that the module makes use of a Tuya Z2 chip and a ARM Cortex-M33 processor. There is an IDE available for the Z2: http://wiki.telink-semi.cn/wiki/IDE-and-Tools/IDE-for-TLSR8-Chips/ and you can subscribe for a "Tuya Smart Production Management System" platform where you can get the TYDA Software. As far as I understand it, this software can at least flash the firmware (wherever it comes from). I'm afraid you have to remove the heatspreader from the ZT5 to be able to solder something directly to the chip.

LKozlov, 2023-12-09:
I made bump from worked device and successfully restored broken TRV
Warning! Do not erase all flash!
from address 0xFE000 it contains Factory configuration and MAC address

[Moes-BRT100-TRV-tl8258-dump.bin.gz](https://github.com/Koenkk/zigbee2mqtt/files/13623277/Moes-BRT100-TRV-tl8258-dump.bin.gz)

sgilyin, 2023-12-09:
> I made bump from worked device and successfully restored broken TRV Warning! Do not erase all flash! from address 0xFE000 it contains Factory configuration and MAC address
> 
> [Moes-BRT100-TRV-tl8258-dump.bin.gz](https://github.com/Koenkk/zigbee2mqtt/files/13623277/Moes-BRT100-TRV-tl8258-dump.bin.gz)

How can we restore our devices? Can you create instructions?

LKozlov, 2023-12-09:
Connect ZT5 to any usb-ttl converter:
[image]
then flash attached firmware. I used this one (works with google chromium):
https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
Warning! Do not erase all flash!

jonathan-gatard, 2023-12-09:
Well done @LKozlov 
Thanks you so much

closed 2023-12-09 by jonathan-gatard

MattWestb, 2023-12-10:
## Great  work @LKozlov !!!
Now all users with  broken devices can getting them back working only need little hardware skills for getting the connections OK after using the hammer ;-)

sktoroid, 2023-12-10:
I will make a small addition to the scheme. +3.3V and GND must be taken from the TTL adapter, otherwise I could not flash it.

almirus, 2024-10-31:
any idea to add external temperature sensor support?

jonathan-gatard, 2024-11-03:
> any idea to add external temperature sensor support?

+1

frickua, 2024-11-08:
> any idea to add external temperature sensor support?

1. U can extend the existing sensor with a cable
2. [crutch] U can use a digital potentiometer like X9C103S with some ESP - (01 for example) on a real room sensor update u need to change the potentiometer value via esp. so TRV will think that the temperature changed. But it is a very "crutched" way, and mostly applicable when u have a manifold with all trv installed on it in the boiler room, to optimize usage of power supply and controllers.
3. There are trvs that use external sensors, like sonoff and aquara