Koenkk/zigbee2mqtt #19174: Connecte thermostat - Different manufacturer

Andreas-Froyland, 2023-10-04:
### What happened?

I have a Connecte smart thermostat, that is connected to Zigbee.

In zigbee2mqtt this device comes up as Not supported, with Zigbee model TS0601 and manufacturer _TZE204_e5hpkc6d 

[image]

I see that in zigbee herdsman converters there is a Connecte device, with the exact same type and the same model number as the one I got. However, the manufacturer differs.

https://github.com/Koenkk/zigbee-herdsman-converters/blob/465413a49df8ae413ca1605b841525620a9da5cc/src/devices/connecte.ts#L10C1-L12

### What did you expect to happen?

I expected that this device would come up as supported, since it is the same model that is listed under supported devices. https://www.zigbee2mqtt.io/devices/4500994.html

### How to reproduce it (minimal and precise)

1. Add device to zigbee2mqtt
2. Device comes up as not supported

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

0x26720700

### Adapter

Conbee II

Koenkk, 2023-10-05:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-05 by Koenkk

Andreas-Froyland, 2023-10-05:
Awesome, thank you!