Koenkk/zigbee2mqtt #19177: Zigbee2MQTT losing network connection consistently after recent updates (Hassio)

allgrinder, 2023-10-05:
### What happened?

Hey,

I've been using Zigbee2MQTT via HomeAssistant for over 2 years, and for a little over a year with the Sonoff Zigbee 3.0 stick.

So far, everything has been running very smoothly and reliably. However, since the last update, and also with the current one, I've been constantly facing issues:

Zigbee2MQTT loses connection to the entire network. No device responds, and the error message is:

Failed to ping "Kitchen" (attempt 1/2, Read 0xd*****/1 genBasic(["zclVersion])... Timeout /... dataRequest)

Once the error occurs, the only solution is to restart. After that, all devices reconnect as usual.

### What did you expect to happen?

I have no idea. The regular log just indicates that no device can be pinged. I have now activated Zigbee-herdsman.

### How to reproduce it (minimal and precise)

The issue is not reproducible. It occurs intermittently, sometimes after hours, and sometimes only after a day.

### Zigbee2MQTT version

1.33 1-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 stick

DorjeDorf, 2023-10-05:
I am seeing the same thing on Slaesh's stick with the newes firmware. After a couple hours, all communication stops and all I see in logs is the same as OP reported.

CzPeti, 2023-10-05:
same error with conbee 2

Palthe, 2023-10-23:
I have the same problem with a number of my devices. Once the power of the devices are on and off again or randomly after a period of time, the responce in HAS works again. 

The error log:

Zigbee2MQTT:error 2023-10-23 22:36:56: Error: CommandResponse 0x001788010cd9d740/2 genOta.queryNextImageResponse({"status":152}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)
Zigbee2MQTT:error 2023-10-23 22:36:57: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:error 2023-10-23 22:36:57: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error)'
Zigbee2MQTT:error 2023-10-23 22:37:00: Publish 'set' 'state' to 'back garden - at the house - light switch' failed: 'Error: Command 0xa4c1383b78913732/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 14009 - 1 - 21 - 6 - 11 after 10000ms)'

rdl987, 2023-10-24:
A solution could be downgrade zigbee2mqtt version?

I have the same problem

allgrinder, 2023-10-24:
I understand it not. Since my post, everything ran stable, and today 3 crashes... I changed nothing. Also, the log is somehow not informative....

mjaensch, 2023-11-08:
I have the same problem. Lost connections to half of my zigbee devices over night.
I will try to downgrade zigbee2mqtt since pairing all these devices again and changing all related home assistant things is a really big pain.

N3rdix, 2023-11-08:
> pairing all these devices again and changing all related home assistant things is a really big pain.

This is usually not a problem if you just pair them again in Z2M. First thing I would try is putting them in pairing mode without deleting the device, only if this does not help delete it in Z2M. All entities in HA should be kept/recreated as before, I did that a lot of times without any hassle in HA

mjaensch, 2023-11-08:
> This is usually not a problem if you just pair them again in Z2M. First thing I would try is putting them in pairing mode without deleting the device, only if this does not help delete it in Z2M. [...]

Thank you! I managed to re-pair all devices without deleting them first.
I downgraded to version 1.33.0 and I hope that it will be a bit more stable now.

aambroze, 2023-12-25:
Version 1.34, drops started happening recently once or twice a day. I’ve confirmed that restarting the addon helps. Before that I have these issues reported
[image]

If restart helps does it mean it is indeed a software problem ?

tclemos, 2024-01-01:
I'm also facing a similar issue. I'm running it as a home assistant add-on and at some point some devices simply stop responding. If I click on a button, the entity state is toggled but the physical switch doesn't toggle the state. Generally a restart to the add-on fixes the problem. Also I have some scene buttons that struggle to communicate, not sure if it is related, I'll try to downgrade it to the previous version and check if it is more stable

Ricconnect, 2024-02-20:
Anyone has an update on this issue? I'm experiencing the same problem since more frequently updating the docker instance of Z2M. It works after a restart, but only for some time (minutes or hours) and then all my devices stop responding for some reason

timwar, 2024-03-01:
Same issue here :(.
@mjaensch Did a downgrade the magic(did it help?)

dodo124, 2024-03-06:
Same problem here. Sonoff Zigbee 3.0 and Zigbee2MQTT. 
This is happening since I switched from IO Broker to HomeAssistant in Dezember.
One more thing though:
Some devices not only loose connection but HA tells me that the whole entitie is gone.
So far this problem mainly occurs when the Zigbee3 device is completely turned off, e.g. there is a switch to turn it off completely (for the case that HA is not working). 
And additionaly I do have a Tuya light which randomly looses connection and starts pairing mode.
One thing I recognized is that the problem occours when I interact with HA (especially Zigbee2MQTT).

Greetings Dom

cracyfloyd, 2024-04-27:
Same here with Conbee III.
For me are only some bulbs from Ikea,Philips and Osram with this problem With my xiaomi sensors not. it is with the latest release version and with the latest beta too.
Zigbee2MQTT-Version
1.36.1-dev commit: [245afb1](https://github.com/Koenkk/zigbee2mqtt/commit/245afb1)
Coordinator-Typ
ConBee3
Coordinator-Version
0x26510900

JSkier21, 2024-05-04:
1.37.0 seems to have fixed connectivity issues with the Peanut plugs I have. It's been two days and no connectivity issues so far 🤞

closed 2024-12-01 by github-actions[bot]