Koenkk/zigbee2mqtt #19178: [New device support]:

RusselIce, 2023-10-05:
### Link

https://aliexpress.ru/item/1005002776442806.html?spm=a2g2w.orderdetail.0.0.6ed64aa6Aq7B61&sku_id=12000034189168907&_ga=2.53168235.2123802470.1696432918-1411164799.1661953562

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xa4c138d74b6d163d","nwkAddr":15199,"manufId":4417,"manufName":"_TZE204_6fk3gewc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u001d�,f�\u001d�,\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_6fk3gewc","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1696509678244,"defaultSendRequestWhen":"immediate"}

### Comments

[image]

closed 2023-11-13 by github-actions[bot]

phoenixckua, 2023-11-18:
I need this device integrate too

Fenixekb, 2023-12-10:
Yes, support for this device is very necessary!

BenJamesAndo, 2024-01-10:
For now you can use custom code from https://github.com/Koenkk/zigbee2mqtt/issues/19147
I might see if I can create a pull request to get this officially added.