Koenkk/zigbee2mqtt #19184: Interview failed of Danfoss Ally after few days

messani, 2023-10-05:
### What happened?

After few days of working I saw red triangle in zigbee2mqtt web interface at my Danfoss Ally device [](https://www.zigbee2mqtt.io/devices/014G2461.html#danfoss-014g2461) . Device seems to be working, but I think that this should not happen. It happens only with Danfoss Ally devices. Other devices are working well (for approximatelly one year)

### What did you expect to happen?

No triangle in web interface

### How to reproduce it (minimal and precise)

I do not know. I saw it few times in the past, but I have log now.

### Zigbee2MQTT version

1.33.0 commit: 4d53ac4

### Adapter firmware version

20230922

### Adapter

CC2652P from china: https://www.aliexpress.com/item/1005004149549758.html?spm=a2g0o.order_list.order_list_main.17.371f1802JYQJhL

### Debug log

[interview_failed.txt](https://github.com/Koenkk/zigbee2mqtt/files/12821934/interview_failed.txt)

lux73, 2023-10-06:
are you sure that pairing is 100% comlete before?

AFAIK Interview is not not more necessary after succesful pairing...

messani, 2023-10-06:
> are you sure that pairing is 100% comlete before?
> 
> AFAIK Interview is not not more necessary after succesful pairing...

All my Danfoss Ally devices has been paired approximately one week ago. All devices were successfully "Configured" and I think that it means that all were suceessfully Interviewed. I did not noticed interview error when device was successfully configured. 

I admit that I had to pair devices multiple times, but last pairing was finished successfully for all Danfoss devices.

I am 100% sure that the red triangle was not there before. And I am 100% sure that I did nothing at that time of interview.
I do not know the reason why it happened. Did you see the log? There is something about a quirk and timeout (I do not know what it means)

lux73, 2023-10-11:
sounds really strange. But glad you got it working

closed 2024-05-09 by github-actions[bot]