Koenkk/zigbee2mqtt #19186: Transition ignored if both target brightness and color temperature specified

kszys, 2023-10-05:
### What happened?

When specyfying all three attributes during `light.turn_on` service call in Home Assistant:

- `transition`
- `brightness`
- `color_temp`

The transition is completely ignored - the light directly reaches specified color temperature and brightness. 
Tested on various Ikea lights. 

### What did you expect to happen?

I expected that the light would _transition_ to the new state during the time specified. This actually works fine if only `brightness` or only `color_temp` attributes are specified - but not when both are there. 

### How to reproduce it (minimal and precise)

Execute in HA Developer Tools -> Services (substitute light entity with your own):

```
service: light.turn_on
data:
  transition: 16
  brightness: 250
  color_temp: 357
target:
  entity_id: light.office_lamp
```

Note - if either `brightness` or `color_temp` attributes are removed - everything works fine. 

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

Debug 2023-10-05 23:22:17Received MQTT message on 'zigbee2mqtt/Office Lamp/set' with data '{"state":"ON","transition":16.0,"brightness":249,"color_temp":357}'
Debug 2023-10-05 23:22:17Publishing 'set' 'brightness' to 'Office Lamp'
Debug 2023-10-05 23:22:17Publishing 'set' 'transition' to 'Office Lamp'
Debug 2023-10-05 23:22:17Publishing 'set' 'color_temp' to 'Office Lamp'
Debug 2023-10-05 23:22:17Received Zigbee message from 'Office Lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":249}' from endpoint 1 with groupID 0
Info 2023-10-05 23:22:17MQTT publish: topic 'zigbee2mqtt/Office Lamp', payload '{"brightness":249,"color_mode":"color_temp","color_options":null,"color_temp":357,"level_config":{"on_level":"previous"},"linkquality":176,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
Info 2023-10-05 23:22:17MQTT publish: topic 'zigbee2mqtt/Office Lamp', payload '{"brightness":249,"color_mode":"color_temp","color_options":null,"color_temp":357,"level_config":{"on_level":"previous"},"linkquality":172,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'

Koenkk, 2023-10-06:
This is because Ikea lights don't support a transition on both simultaneously, so the observed behavior is expected.

kszys, 2023-11-08:
I guess I cannot expect a fix then. But just to notice that this is what HA
scenes are doing (setting both target brightness and color temperature) and
it works when using ZHA... For me the trigger to discover it was when my
existing Scenes stopped working correctly after moving my lights to Z2M. So
_there is a way_... Just saying...

3limin4tor, 2024-01-09:
Having the same issue and would be totally nice if this somehow can be fixed.

kineticscreen, 2024-06-08:
ZHA has a setting to automatically adjust for this: "Enable enhanced light colour/temperature transition from an off-state".

Surprising that it isn't also built into Z2M

kineticscreen, 2024-12-11:
Just posting to keep this open, unless someone has an update on this being fixed.

closed 2025-03-14 by github-actions[bot]

ArbiterGR, 2025-04-06:
I'm surprised that this is not built into z2m. Can this somehow be fixed?