Koenkk/zigbee2mqtt #19188: Difference in TS0601 Single and Double Clamp Settings (specific) Tab.

congkai88, 2023-10-06:
### What happened?

Difference in Single and Double Clamp Settings (specific) Tab. 

**Single Clamp:** 
Zigbee Model TS0601 
Zigbee Manufacturer _TZE204_cjbofhxw
[image]
[image]

**Double Clamp:**
Zigbee Model TS0601 
Zigbee Manufacturer _TZE204_81yrt3lo
[image]
[image]

**Issue**
Unable to adjust Power Precision or Calibration for the double clamp devices
[image]

### What did you expect to happen?

Double clamp should have the Settings to adjust the Power Precision and Calibration of 2 clamps. 

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20220219

### Adapter

Sonoff Usb3.0 Zigbee

closed 2023-11-14 by github-actions[bot]

Emilef22, 2023-11-14:
Hi! Have you found a solution to adjust the other specific parameter?

PetersPlace, 2024-01-19:
Same here only voltage setting.

Other question,  for very small currents i would like to change te CT ratio setting 1:80 change to 1:20   ( 4 windings through the CT clamp)