Koenkk/zigbee2mqtt #19191: [New device support]: Zigbee Combustible Gas Detector

rednet1, 2023-10-06:
### Link

https://it.aliexpress.com/item/1005005936314298.html?spm=a2g0o.order_list.order_list_main.71.41b236964M0Oco&gatewayAdapt=glo2ita

### Database entry

{"id":49,"type":"Router","ieeeAddr":"0xa4c138bd4cabd1e4","nwkAddr":643,"manufId":4417,"manufName":"_TZE204_zougpkpy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1696608818753,"defaultSendRequestWhen":"immediate"}

### Comments

Wrong model Not supported

kerimcelik, 2023-10-24:
Aynı sorun bende de var. Cihazı buluyor eşleşme yapıyor. Fakat cihaz tanınmıyor. Çözüm için bekliyorum.

gvlad01, 2023-11-05:
I have the same gas detector called TUYA DY-RQ500A, which works for me

closed 2024-06-03 by github-actions[bot]