Koenkk/zigbee2mqtt #19193: Zibbee2mqtt crash?

j0hanSE, 2023-10-07:
### What happened?

After recent update the addon on HA crashes after awhile?

The only report i can see in the logs is this when i have to manual start it again.

```
23-10-07 08:09:53 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:14:58 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:20:02 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:23:59 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-10-07 08:25:06 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:30:10 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:35:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:37:55 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:37:58 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)23-10-07 08:09:53 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:14:58 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:20:02 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:23:59 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-10-07 08:25:06 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:30:10 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:35:14 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:37:55 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:37:58 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:37 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:38 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:38 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application
23-10-07 08:38:38 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:49 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-aarch64 with version 1.33.1-1
23-10-07 08:38:49 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:50 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:39:37 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt_user'
23-10-07 08:39:39 INFO (MainThread) [supervisor.auth] Successful login for 'mqtt_user'
23-10-07 08:40:19 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:37 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:38 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:38 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_45df7312_zigbee2mqtt application
23-10-07 08:38:38 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:49 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on zigbee2mqtt/zigbee2mqtt-aarch64 with version 1.33.1-1
23-10-07 08:38:49 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:38:50 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
23-10-07 08:39:37 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt_user'
23-10-07 08:39:39 INFO (MainThread) [supervisor.auth] Successful login for 'mqtt_user'
23-10-07 08:40:19 WARNING (MainThread) [supervisor.addons.options] Unknown option 'base_topic' for Zigbee2MQTT (45df7312_zigbee2mqtt)
```

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20220713083000

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

rdeneer, 2023-10-10:
Also experienced a crash last night, running z2m 1.33.1 in a container using Docker. No clues in the z2m log...
In the Docker daemon I observe the following:

`Oct 10 02:33:04 zigbeebridge dockerd[581]: time="2023-10-10T02:33:04.370192962+02:00" level=info msg="ignoring event" container=5010f09ca1608fc812c0749d8f7a83deaab4842a0f39b58f08c92e01922540f6 module=libcontainerd namespace=moby topic=/tasks/delete type="*events.TaskDelete"`

If this happens again anytime soon I'll enable debug logging to see if I can gather some more clues.

tech2mar, 2024-02-10:
Exactly the same thing is happening to me on the HASSos. Anything new here where the problem originated?

closed 2024-09-08 by github-actions[bot]