Koenkk/zigbee2mqtt #19196: [Feature request]: Please support zigbee repeater TS0207 for manufacturer _TZ3000_nlsszmzl

CrewingTop, 2023-10-07:
### Is your feature request related to a problem? Please describe

The device is detected as https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector but should be https://www.zigbee2mqtt.io/devices/TS0207_repeater.html. 

### Describe the solution you'd like

Add manufacturer into database

### Describe alternatives you've considered

Add manufacturer into database

### Additional context

**Zigbee2MQTT version**
1.33.1

**Adapter**
Itead SONOFF Zigbee 3,0 USB Dongle Plus (SONOFF ZBDongle-P)
https://a.aliexpress.com/_mtLH5LC

**Adapter firmware version**
20230507

**Zigbee Repeater TS0207**
https://a.aliexpress.com/_mOMO0Bk

**Manufacturer**
_TZ3000_nlsszmzl

**Debug log**
Failed to configure 'ZB роутер 1', attempt 2 (Error: ConfigureReporting 0x943469fffe57265b/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 20087 - 1 - 204 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

Koenkk, 2023-10-08:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-08 by Koenkk

cloudian-mgerecke, 2023-10-09:
I have the exact same issue but the OEM of the ZigBee repeater is Marmitek.
Zigbee Manufacturer:  _TZ3000_wlquqiiz
Zigbee2MQTT version: 1.33.1
https://www.amazon.co.uk/Marmitek-SE-Link-Hotspot-Repeater/dp/B09V7K4DBC

The device completed the interview and is working as a repeater in my ZigBee network.

IEEEE Address:
0x04cd15fffe0c4c9d
(0x7CBD)

[image]

grobihh, 2023-10-09:
I have the exact same issue but the OEM of the ZigBee repeater is Marmitek.
Zigbee Manufacturer: _TZ3000_wlquqiiz
Zigbee2MQTT version: 1.33.1
https://www.amazon.co.uk/Marmitek-SE-Link-Hotspot-Repeater/dp/B09V7K4DBC

The device completed the interview and is working as a repeater in my ZigBee network.

IEEEE Address:
0x04cd15fffe0c4c9d
(0x7CBD)

[image]

Koenkk, 2023-10-09:
@grobihh also added this.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)