Koenkk/zigbee2mqtt #19197: Aqara Magic Cube is not reporting the correct side

heisenberg2980, 2023-10-07:
### What happened?

The aqara cube reports an incorrect value for "side" every time the cube is rotated to a difference side in a way that doesn´t trigger an action (e.g. rotating the cube in diagonal). After this the cube keeps reporting the wrong side even after performing other actions (e.g. rotating it, shaking it) and it is only fixed when the cube is flipped (either 90 or 180 degrees) and the side reported is correct again.

### What did you expect to happen?

The cube should report the correct side in every action (e.g. rotating it, shaking it) and not only after flip90 and flip180

### How to reproduce it (minimal and precise)

Change the side the cube is in a way that doesn´t trigger an action and then perform any other action like rotating it

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20220219

### Adapter

sonoff zigbee 3.0

### Debug log

debug 2023-10-07 12:43:26: Received Zigbee message from '0x00158d000231e246', type 'attributeReport', cluster 'genAnalogInput', data '{"65285":500,"presentValue":70.3499984741211}' from endpoint 3 with groupID 0
info  2023-10-07 12:43:26: MQTT publish: topic 'zigbee2mqtt/0x00158d000231e246', payload '{"action":"rotate_right","action_angle":70.34,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":100,"current":0,"device_temperature":22,"linkquality":58,"power":724,"power_outage_count":5,"side":2,"voltage":3025}'
info  2023-10-07 12:43:26: MQTT publish: topic 'zigbee2mqtt/0x00158d000231e246', payload '{"action":"","action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":100,"current":0,"device_temperature":22,"linkquality":58,"power":724,"power_outage_count":5,"side":2,"voltage":3025}'
info  2023-10-07 12:43:26: MQTT publish: topic 'zigbee2mqtt/0x00158d000231e246/action', payload 'rotate_right'

KHV8, 2023-10-08:
This is the expected behaviour of the cube. It will only report on actions.

heisenberg2980, 2023-10-08:
> This is the expected behaviour of the cube. It will only report on actions.

Yes but it should report the correct side in all the actions and not only in flip90 and flip180 actions. Due to this issue any automation based on the side the cube is very unreliable, as the cube can easily report a wrong side. As an example I wanted to use different sides of the cube for different actions, for instance one side to control the volume (so rotate turn the volume up/down), other sides for different lights (so rotate increase/decrease the brightness of that specific light), but I found the cube was reporting the wrong side very often, and after some investigation I realised the issue happens every time the side of the cube is changed without triggering an action.

KHV8, 2023-10-10:
Z2M or any other Zigbee system can only report what the HW reports. The cube only reports when three is a action. You can check this in the states part in Z2M.
I have a cube, I do not use it at all, due to this problem.

closed 2024-05-09 by github-actions[bot]