Koenkk/zigbee2mqtt #19198: Ikea TRADFRI Driver 10W - ICPSHC24-10EU-IL-1

chris-1243, 2023-10-07:
### What happened?

Hello everyone,

Since version 1.33.1 (stable and dev) this device is able to pair correctly but then it does not answer to any mqtt commands and timeout errors are sent.

I tried with different coordinator firmware (20230507 and 20230923) with the same behavior.

Any help would be much appreciated

### What did you expect to happen?

The same behavior as version 1.33.0 in which the device is perfectly working.

### How to reproduce it (minimal and precise)

As it is a spare part, i bring the device as close as possible of the coordinator ~10cm, bring it in paring mode and i do allow joining device in Zigbee2mqtt.
After a few seconds, device is recognized and paired. Then, while trying to send a command via the frontend, I do receive those errors.
I have to force the device to be removed from the network.

Doing the same in version 1.33.0 and everything goes right. I even do not need to force the device the removed from the network.

### Zigbee2MQTT version

1.33.1 stable and dev

### Adapter firmware version

20230923

### Adapter

Sonoff dongle P

### Debug log

error 2023-10-07 13:33:14: Publish 'set' 'power_on_behavior' to '0x50325ffffe4548c4' failed: 'Error: Write 0x50325ffffe4548c4/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
debug 2023-10-07 13:33:14: Error: Write 0x50325ffffe4548c4/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
debug 2023-10-07 13:33:24: Received MQTT message on 'zigbee2mqtt/0x50325ffffe4548c4/get' with data '{"state":""}'
debug 2023-10-07 13:33:24: Publishing get 'get' 'state' to '0x50325ffffe4548c4'
debug 2023-10-07 13:33:26: Received MQTT message on 'zigbee2mqtt/0x50325ffffe4548c4/set' with data '{"state":"OFF"}'
debug 2023-10-07 13:33:26: Publishing 'set' 'state' to '0x50325ffffe4548c4'
error 2023-10-07 13:33:32: Publish 'get' 'power_on_behavior' to '0x50325ffffe4548c4' failed: 'Error: Read 0x50325ffffe4548c4/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
debug 2023-10-07 13:33:32: Error: Read 0x50325ffffe4548c4/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
error 2023-10-07 13:33:50: Publish 'set' 'state' to '0x50325ffffe4548c4' failed: 'Error: Command 0x50325ffffe4548c4/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'
debug 2023-10-07 13:33:50: Error: Command 0x50325ffffe4548c4/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
debug 2023-10-07 13:34:01: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"error"}},"transaction":"kyd0z-7"}'
error 2023-10-07 13:34:08: Publish 'set' 'state' to '0x50325ffffe4548c4' failed: 'Error: Command 0x50325ffffe4548c4/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'

MattWestb, 2023-10-07:
`'MAC no ack'` = radio channel is blocked of strong interference so take one look how the WiFi is around your network and adjusting it or the Zigbee channel.

chris-1243, 2023-10-08:
Thanks for the advice. I had a look and my wifi is on channel 1 and my zigbee network on channel 25. It should normally be well separated.

After some tries, I moved away the tradfri driver against the recommandations made by the zigbee2mqtt's team for pairing this device. My wifi router is not so far of the coordinator.
Paring went fine and i didn't get errors.

Problem solved. I did learn something. Thanks for your help

closed 2023-10-08 by chris-1243