Koenkk/zigbee2mqtt #19201: [New device support]: IKEA LED2107C4 TRADFRI bulb E14 WS candle 470lm

jmccrohan, 2023-10-07:
### Link

https://www.ikea.com/pt/pt/p/tradfri-lampada-led-e14-470-lumenes-ilum-regulavel-s-fios-espectro-branco-lampada-chama-branco-opala-60545499/

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0x5cc7c1fffe39941d","nwkAddr":39296,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E14 WS candle 470lm","epList":[1,242],"endpoints":{"1":{"prof Id":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64636],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E14 WS candle 470lm","manufacturerName":"IKEA of Sweden","powerSource":1,"z clVersion":8,"appVersion":2,"stackVersion":106,"hwVersion":1,"dateCode":"20230421","swBuildId":"2.4.4"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"in ClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":106,"hwVersion":1,"dateCode":"20230421","swBuildId":"2.4.4","zclVersion":8,"interviewCompleted":true,"meta": {},"lastSeen":1696689494183,"defaultSendRequestWhen":"immediate"}

### Comments

I was going to raise a PR against zigbee-herdsman-converters myself, but I clear whether LED2107C4 should be added as its own entry definition, or appended to either LED1903C5/LED1835C6 or LED1949C5.

IKEA provides a single manual for LED1835C6, LED1949C5 and LED2107C4.

Koenkk, 2023-10-08:
Since the zigbeeModel is unique (`TRADFRI bulb E14 WS candle 470lm`) it should be added as it's own entry.

jmccrohan, 2023-10-08:
@Koenkk Thanks. I see @ggtimtom filed a parallel request (https://github.com/Koenkk/zigbee-herdsman-converters/issues/6253) which has since been merged.

closed 2023-10-08 by jmccrohan

Koenkk, 2023-10-09:
@zillion42 try updating again in 1 hour from now

zillion42, 2023-10-09:
Hi,

Updated, it's now commit #9cff19b
```
info  2023-10-09 22:21:21: Starting Zigbee2MQTT version 1.33.1-dev (commit #9cff19b)
info  2023-10-09 22:21:21: Starting zigbee-herdsman (0.19.1)
```
unfortunately I still got the same issue:

```
Debug 2023-10-09 22:25:44Received Zigbee message from '0x5cc7c1fffe5ad01e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816580,"imageType":8704,"manufacturerCode":4476}' from endpoint 1 with groupID 0
Warning 2023-10-09 22:25:44Received message from unsupported device with Zigbee model 'TRADFRI bulb E14 WS candle 470lm' and manufacturer name 'IKEA of Sweden'
Warning 2023-10-09 22:25:44Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
```

zillion42, 2023-10-09:
Should I delete any files from here, between updates? I was coming from non edge Version before.
[image]

martinhthomsen, 2023-10-30:
> Hi,
> 
> I force deleted the device 0x5cc7c1fffe5ad01e. I then updated, it's now commit #9cff19b
> 
> ```
> info  2023-10-09 22:21:21: Starting Zigbee2MQTT version 1.33.1-dev (commit #9cff19b)
> info  2023-10-09 22:21:21: Starting zigbee-herdsman (0.19.1)
> ```
> 
> Then I factory reset the bulb, 6x times on off, and Interviewed it again
> 
> Unfortunately I still got the same issue:
> 
> ```
> Debug 2023-10-09 22:25:44Received Zigbee message from '0x5cc7c1fffe5ad01e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816580,"imageType":8704,"manufacturerCode":4476}' from endpoint 1 with groupID 0
> Warning 2023-10-09 22:25:44Received message from unsupported device with Zigbee model 'TRADFRI bulb E14 WS candle 470lm' and manufacturer name 'IKEA of Sweden'
> Warning 2023-10-09 22:25:44Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
> ```
> 
> [image]

I'm having the exact same problem. Should the "TRADFRI bulb E14 WS candle 470lm" be supported?

WeMakeComputersWork, 2023-11-07:
Since the last firmware update (3.0.10), the state switch reacts incorrectly.
Both events (state=true, state=false) lead to toggling instead of setting the status.

quenthal, 2023-11-30:
I'm having trouble with these bulbs also. I have many IKEA bulbs, but only these are quite a pain to work with.

After multiple pairing and unpairing and manually setting reporting on "OnOff" and "CurrentLevel"I managed to get them work - partially.

(I have ceiling lamp that has three of these. Naturally I made group out of these. Binded remote is nearly impossible with group consisting of these bulbs. Making group out of any  other Ikea or Philips bulbs works fine.)

Aragur, 2023-12-08:
Is this device supported?
I cannot get it to pair

TomComm27, 2023-12-08:
> Hello,
> 
> 
> 
> the "TRADFRI bulb E14 WS candle 470lm" is also not recognized correctly, although in my opinion everything is already up-to-date.
> 
> The adapter is operated under a Raspberry Pi4 (Debian version: Bullseye) with a ConBee2.
> 
> 
> 
> Here hopefully all necessary information:
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 
> 
> 
> 
> /opt/zigbee2mqtt/data/configuration.yaml 
> 
> ```                                      
> 
> homeassistant: false
> 
> permit_join: true
> 
> frontend:
> 
>   port: 80xx
> 
>   host: 192.xxx.xxx.xxx
> 
> mqtt:
> 
>   base_topic: zigbee2mqtt
> 
>   server: mqtt://192.xxx.xxx.xxx:xxxx
> 
>   user: xxx
> 
>   password: xxxxxxxxxxx
> 
>   keepalive: 60
> 
>   reject_unauthorized: true
> 
>   version: 4
> 
>   client_id: 
> 
> serial:
> 
>   port: /dev/ttyACM0
> 
> advanced:
> 
>   pan_id: 10328
> 
>   ext_pan_id:
> 
[... 165 lines cut ...]
> 
> Warning 2023-11-29 14:13:43Received message from unsupported device with Zigbee model 'TRADFRI bulb E14 WS candle 470lm' and manufacturer name 'IKEA of Sweden'
> 
> Warning 2023-11-29 14:13:44Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
> 
> Info 2023-11-29 14:13:44MQTT publish: topic 'zigbee2mqtt/0xb43522fffe814999', payload '{"last_seen":"2023-11-29T14:13:43+01:00","linkquality":255}'
> 
> Warning 2023-11-29 14:13:46Received message from unsupported device with Zigbee model 'TRADFRI bulb E14 WS candle 470lm' and manufacturer name 'IKEA of Sweden'
> 
> Warning 2023-11-29 14:13:46Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
> 
> ```
> 
> 
> 
> I really don't know anymore!
> 
> 
> 
> Thank you!

I solved the problem with:
`# Run the update script from the Zigbee2MQTT directory
cd /opt/zigbee2mqtt
./update.sh`