Koenkk/zigbee2mqtt #19205: Aqara roller shade companion E1 (Xiaomi ZNJLBL01LM) state reporting

z-master42, 2023-10-07:
### What happened?

I have an Aqara roller shade companion E1 (Xiaomi ZNJLBL01LM) which works so far with Z2M. However, what does not work is the transmission of the position of the curtain and the battery level to Home Assistant. In Z2M, this can be updated manually, but there is no transmission. See also discussion #16910. However, I can not imagine that it is due to the channel.

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EFR32MG21 (Home Assistant Yellow)

cromelex, 2023-10-09:
I have a number of E1s and using channel 20 and don't see the autoreporting. Doing it manually works.

llamaonaskateboard, 2023-11-12:
I have this issue also. 
I've written an extension as very crude workaround which picks up any position set command and triggers a read after 10 seconds (basic timeout for the write to go through).
https://gist.github.com/llamaonaskateboard/1371619bbad59e87e19bd38562ddeee4

z-master42, 2023-11-12:
Thanks @llamaonaskateboard Better than nothing ;-)

cromelex, 2023-11-12:
> I have this issue also. I've written an extension as very crude workaround which picks up any position set command and triggers a read after 10 seconds (basic timeout for the write to go through). https://gist.github.com/llamaonaskateboard/1371619bbad59e87e19bd38562ddeee4

Sorry can I ask how I would use this?

z-master42, 2023-11-12:
On your Z2M page, you will find the Extensions tab on the far right. Create a new one using the green +. And overwrite the example code with the code from the Github Gist and then simply save it. If you now move the curtain and it stops at some point, the status will be updated after approx. 10 seconds, including that of the battery.

llamaonaskateboard, 2024-03-30:
In case anyone is still using my workaround extension above, I've updated the gist to handle a cluster name change from https://github.com/Koenkk/zigbee-herdsman/pull/892

closed 2024-10-28 by github-actions[bot]