Koenkk/zigbee2mqtt #19211: Certain IKEA bulbs switching to lowest brightness on toggle

JudahBella, 2023-10-08:
### What happened?

Some of our IKEA bulbs are switching on to the lowest brightness setting when toggled. This happened suddenly after Z2M autoupdated to 1.33.1, and it has never happened before. No OTA was done on the bulbs either, so there was no other change. 

This happens regardless of being toggled from HomeKit via HomeKit Bridge, Home Assistant, or directly from Z2M GUI. 

For one of the bulbs having this issue, setting the transition to a numerical value (instead of leaving it blank) solved the issue.

However, certain bulbs in groups still power on to 1% brightness on toggle. Not all bulbs are affected, only some. 

### What did you expect to happen?

The bulb should turn on to the last brightness setting when toggling from off to on

### How to reproduce it (minimal and precise)

1. Turn the bulb on, set the brightness (e.g. 100%)
2. Toggle the bulb off
3. Toggle the bulb back on (and the brightness is at 1%, even though the UI says 100%)

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230507

### Adapter

tubeszb-cc2652-eth

JudahBella, 2023-10-08:
Referencing https://github.com/Koenkk/zigbee2mqtt/issues/3333, I decided to try adding `transition: 1` to `groups.yaml`. Strangely enough, `genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10})` is sent and all bulbs in the group turn on to the correct previous brightness:

```
Zigbee2MQTT:debug 2023-10-08 18:40:59: Received MQTT message on 'zigbee2mqtt/Common Toilet Lights/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-10-08 18:40:59: Publishing 'set' 'state' to 'Common Toilet Lights'
2023-10-08T10:40:59.670Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-10-08T10:40:59.698Z zigbee-herdsman:controller:group Command 11 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10})
2023-10-08T10:40:59.700Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x000000000000000b","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":29,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,10,4,254,10,0]}}
2023-10-08T10:40:59.702Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,1,11,0,0,0,0,0,0,0,255,0,0,1,8,0,29,0,30,6,0,17,10,4,254,10,0,46]
2023-10-08T10:40:59.736Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2023-10-08T10:40:59.736Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2023-10-08T10:40:59.736Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2023-10-08T10:40:59.737Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2023-10-08T10:40:59.737Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-10-08T10:40:59.752Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,29,219]
2023-10-08T10:40:59.753Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,29,219]
2023-10-08T10:40:59.753Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,29] - 219
2023-10-08T10:40:59.753Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":29}
2023-10-08T10:40:59.754Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2023-10-08 18:40:59: MQTT publish: topic 'zigbee2mqtt/Common Toilet Ceiling', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_options":{"execute_if_off":false},"color_temp":370,"color_temp_startup":370,"level_config":{"on_level":0},"linkquality":25,"power_on_behavior":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2023-10-08 18:40:59: MQTT publish: topic 'zigbee2mqtt/Common Toilet Lights', payload '{"brightness":254,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2023-10-08 18:40:59: MQTT publish: topic 'zigbee2mqtt/Common Toilet Counter', payload '{"brightness":254,"level_config":{"on_level":"previous"},"linkquality":25,"power_on_behavior":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2023-10-08 18:40:59: MQTT publish: topic 'zigbee2mqtt/Common Toilet Cove', payload '{"brightness":254,"linkquality":98,"state":"ON"}'
Zigbee2MQTT:info  2023-10-08 18:40:59: MQTT publish: topic 'zigbee2mqtt/Ceiling Lights', payload '{"brightness":254,"color":{"x":0.4607,"y":0.4108},"color_mode":"color_temp","color_temp":371,"state":"ON"}'
```

cwildfoerster, 2023-10-14:
Im experiencing the same issue for a couple of days on one group probably since the update to 1.33.1. Switching the group/lights on, through a bound remote or directly, turns them on at 1%.

Koenkk, 2023-10-15:
This looks similar to https://github.com/Koenkk/zigbee2mqtt/issues/16388#issuecomment-1749489713 and at first glance looks to be a bug of the bulb (off -> on should give the same state back).

- I find it interesting that this happens only with z2m 1.33.1, what was the last known working z2m version?
- Does `transition: 0` also work?

JoernBrauns, 2023-10-15:
> * Does `transition: 0` also work?

I can confirm that the bug is also present for a group of LED2005R5 (IKEA Tradfri GU10) and that adding `transition: 0` or `transition: 1` to the group configuration leads to a correct brightness value after powering on. If only the transition time is added to the lightbulb configuration, the standalone switching is correct, but the group switching behavior is still wrong.

Before the zigbee2mqtt update, the lightbulbs (either standalone or as a group) behaved completely normal for several months. Hence, the bug may be introduced with some of the last zigbee2mqtt updates. Unfortunately, I cannot name the last working version from my side.

JudahBella, 2023-10-15:
> This looks similar to [#16388 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/16388#issuecomment-1749489713) and at first glance looks to be a bug of the bulb (off -> on should give the same state back).
> 
> * I find it interesting that this happens only with z2m 1.33.1, what was the last known working z2m version?
> * Does `transition: 0` also work?

Last known working is 1.33.0. I haven’t tried `transition: 0` yet but I think https://github.com/Koenkk/zigbee2mqtt/issues/19211#issuecomment-1763364246 response should be sufficient!

Koenkk, 2023-10-15:
I'm wondering what caused this issue, @JudahBella how are you running z2m? Then I can provide some instructions to downgrade zigbee-herdsman-converters (which is responsible for the device control)

JudahBella, 2023-10-16:
> I'm wondering what caused this issue, @JudahBella how are you running z2m? Then I can provide some instructions to downgrade zigbee-herdsman-converters (which is responsible for the device control)

I'm running z2m as a HA addon.

Koenkk, 2023-10-16:
Firmware build date: 20230308
Firmware version: 24.4.6

Koenkk, 2023-10-16:
@JudahBella can you ssh into your system and execute:

```bash
# Enter z2m container
docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/bash
cd app
# Install zigbee-herdsman-converters version that with z2m 1.33.0
npm install zigbee-herdsman-converters@15.67.1
exit
# Restart z2m container
docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
```
check if it works now

JudahBella, 2023-10-17:
> @JudahBella can you ssh into your system and execute:
> 
> ```shell
> # Enter z2m container
> docker exec -it $(docker ps | grep zigbee2mqtt | cut -d" " -f 1) /bin/bash
> cd app
> # Install zigbee-herdsman-converters version that with z2m 1.33.0
> npm install zigbee-herdsman-converters@15.67.1
> exit
> # Restart z2m container
> docker restart $(docker ps | grep zigbee2mqtt | cut -d" " -f 1)
> ```
> 
> check if it works now

I'm running HAOS on a VM and don't have any way to SSH into the OS layer at the moment, but let me see what I can do.

JudahBella, 2023-10-17:
Don't mind me poking around, but it seems like between 1.33.0 and 1.33.1, ikea.js was refactored to ikea.ts. Could the refactor have caused anything? 

I notice that in [ikea.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/250f731ba988f9d37d2ac1992e1a9833997111fc/src/devices/ikea.js), `bulbOnEvent` tested for `undefined` quite aggressively whereas in [ikea.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/fec45acfcde1848955f2654d338d26eadf582dfe/src/devices/ikea.ts) it seems to have all but disappeared. I do notice that now in the UI, if the `transition` variable is blank and I click on it but not enter any value, it saves, 

[image]

but when I try to turn it on or off, the following error occurs:

[image]

Also, from my very limited understanding, bulbs seem like they should be turned on with the `genLevelCtrl` command via `bulbOnEvent`, yet without `transition` set, the logs stated that `genOnOff` was sent instead.

Koenkk, 2023-10-17:
- Can you do a `apk add nodejs` before executing the `npm` command?
- Could it be that you had a `transition: ''` with z2m 1.33.0?

cwildfoerster, 2023-10-17:
Just to chime in, i had all my groups configured without `transition` at all. setting `transition: 1` actually didn't help with it.

```
docker compose exec zigbee2mqtt sh

apk add nodejs npm
npm install zigbee-herdsman-converters@15.67.1
exit

docker compose restart zigbee2mqtt
```

```
groups:
...
  '2':
    friendly_name: Deckenleuchte Wohnzimmer
    devices:
      - 0xd0cf5efffeec790b/1
      - 0xbc33acfffe16bc4c/1
      - 0xd0cf5efffef15440/1
```

Had no effect. Still turning on with 0% brightness.

JudahBella, 2023-10-19:
> * Can you do a `apk add nodejs` before executing the `npm` command?

Ok will try again when I have time

> * Could it be that you had a `transition: ''` with z2m 1.33.0?

Not sure for `devices.yaml` (likely not) but definitely not for `groups.yaml`

xeonixo, 2023-10-20:
I am having the same issue with my IKEA ICPSHC24-30EU-IL-1. After turning on the brightness is on 1. HASS still has 255 as value.

cwildfoerster, 2023-10-21:
strangely, it only affects one group, other groups behave correctly.

xeonixo, 2023-10-21:
tried a bit with different versions of z2m. defenitly not a version issue. 
when turning on, it still gets brightness 1. but if i turn it on with an brightness value, the value is correct.

maybe use the known value as attribute while turning on?

alex3305, 2023-11-06:
Over the past few weeks more and more IKEA devices started acting up. Starting at the lowest brightness when turning on. Also Zigbee2MQTT was complaining about not having a valid `transition` property:

```
Error 2023-11-06 11:52:57 Publish 'set' 'state' to 'Closet Bulb 2' failed: 'Error: 'transition' is not a number, got string ()'
```

Which I didn't even touch. I have set the `transition` property on all my devices to 0 now. I hope this keeps working 😐.

But what I also noticed is that this behavior only happens when toggling a light from a Zigbee group. Whenever I toggle the light independently, it just works 🤷. I also tried downgrading from 1.33.1 to 1.33.0 but to no avail.

So for now I just created the same groups within Home Assistant. Which is AFAIK not ideal for my Zigbee network. But as a workaround would be sufficient for now, I guess.

helljol, 2023-11-09:
Hello I just seen I have the Same issue. I have same 5 Ikea LED1837R5 and issue happen on 2 of them. 
It happens when switching them on individually if transition value is blank. If transition is set to 0 then it works. 

Looks like it have start happening after I upgrade from 1.33.0 to 1.33.2

schumi2004, 2023-11-18:
I'm facing the same issue but not using z2m groups at all.
Can you also set transition for a single bulb?
What I did notice is that all affected lights have     "color_temp_startup": 250,  set which I can't remove.

Jabe, 2023-11-24:
A few weeks ago this problem started for 1 of 3 lamps (IKEA T2037 v1.0.006) in a group. Fast forward, 2/3 lamps have this problem, then all of them. I usually use a Ubisys S2 bound to toggle the group with the lamps (using zigbee; they stay powered on). On each transition from OFF to ON the brightness is reset to 1. I set up reporting for "brightness" so that's easy to verify. This happens all the time when using the switch. Keep in mind that the switch only sends toggle to the OnOff cluster.

It also happens when I toggle the group using MQTT directly (simulating the switch) unless I specify "transition: 0".

```jsonc
{"state": "ON", "brightness":200} // Reset
{"state": "OFF"}
{"state": "ON"} // Brightness = 1

{"state": "ON", "brightness":200} // Reset
{"state": "OFF"}
{"state": "ON", "transition":0} // Brightness = 200
```

The switch triggers this behavior even when z2m is stopped (so it can't possibly inject a message). I actually suspect a volatile bug in the firmware because I was able to temporarily fix the issue by powering down the devices (for a good while!). But the behavior quickly reappears unfortunately.

My (very weak) guess is that some recent change to z2m triggers this bug in the firmware that breaks until powered down.

Looking at the ZCL it states under "3.10.2.1.1 Effect of On/Off Commands on the CurrentLevel Attribute":

For ON:  
> Temporarily store CurrentLevel.
> Set CurrentLevel to the minimum level allowed for the device.
> Move CurrentLevel to OnLevel, or to the stored level if OnLevel is not defined, over the
time period OnOffTransitionTime.

So it appears as if the last step simply fails. The big question is: why now? And how could z2m trigger this bug.

Koenkk, 2023-12-06:
@Jabe I've compared to toZigbee code in 1.33.0 vs 1.33.2, but there are not differences regarding this. Do you know from which version this bug starts ocurring?

KizzyCode, 2023-12-06:
I'm afraid I also cannot offer more insight except that for me the problem occurred the upgrade to koenkk/zigbee2mqtt:1.33.2, whereas downgrading to koenkk/zigbee2mqtt:1.33.1 seems to have fixed the problem. Symptoms that occurred are the lowest brightness issue here in the thread, as well as repeated connectivity issues (6000ms or 10000ms timeouts, sadly don't remember exactly) after 24-48h.

After the downgrade, all problems vanished "magically" without any change in config etc.

ovidiubute, 2023-12-06:
I am also having this issue since a few versions now. If I can help with debugging let me know, I have a bunch of IKEA light bulbs and they are all having the brightness problem. 

For me if I kill the power to the lights while they are turned off from HA, then when I turn the power on again the brightness is set to a minimum even though if I check HA the brightness is reported as maximum.

Numpienick, 2023-12-06:
Having the same issue suddenly since a few days, sadly downgrading to 1.33.1 didn't solve it for me, neither did all the transition stuff. In my case the lights turn on at brightness 1 no matter what method I use, be it from HA, my STYRBAR remote or the Z2M frontend. Both bulbs are in a group, but turning them on separately has the same behaviour.

villyvitz, 2023-12-08:
I'm also suffering with this issue. I have just recently started with Home Assistant so I have no prior version history with Z2M and started on 1.34.0. I have LED1949C5 and LED2003G10 bulbs. If the transition value is blank in the Z2M setup GUI then the bulbs will turn on at 1 brightness, and report 255 brightness to HA, despite the previous state. Setting a transition of 0 or 1 resolves the issue. However, I still have this problem with groups of lights. From what I can tell there is no way to set the transition of groups through the GUI but when I set the transition in the configuration.yaml that doesn't seem to work.

Jabe, 2023-12-08:
@Koenkk it's been too long to say when it started, probably with 1.33.0 or even earlier. I usually don't need these lights until the days get shorter, so I might not have noticed until around October. Also, since it only affected 1/3 of the lights in the group I kinda dismissed it in the beginning. 

I still don't think the issue can be fully explained by z2m alone and it's very odd. I can't come up with a reason why this happens except though a firmware bug.

Koenkk, 2023-12-08:
@Jabe could you provide the herdsman debug log of doing the [sequence which triggers the bug](https://github.com/Koenkk/zigbee2mqtt/issues/19211#issuecomment-1826110011)?

Jabe, 2023-12-10:
@Koenkk  okay I've captured three scenarios. There is some noise in the data from other devices which I've XXX'ed out. Please see the the list of relevant device addresses at the end too. Here is my setup:

* The offending lamps are called "**Diele Deckenlampe 1**" though 3.
* They are in a group called "**Diele Deckenlampen**". I usually use the group only.
* They are also in a group called "**Tag Nacht Wechsel EG**" (= day/night switching ground floor) which has 2 scenes but that's not relevant I think; just for clarity.
* There is also the physical toggle switch called "**Diele/Flur Wandschalter**" which is bound to group "**Diele Deckenlampen**".

There is a 1 sec delay between each call. Each call was made using mosquitto_pub. 

Test 1 was made with this:
```jsonc
{"state": "ON", "brightness":200} // Reset
{"state": "OFF"}
{"state": "ON"} // Brightness = 1
```
[test1.log](https://github.com/Koenkk/zigbee2mqtt/files/13627778/test1.log)

Test 2:
```jsonc
{"state": "ON", "brightness":200} // Reset
{"state": "OFF"}
{"state": "ON", "transition":0} // Brightness = 200
```
[test2.log](https://github.com/Koenkk/zigbee2mqtt/files/13627780/test2.log)

Test 3 I made with the physical toggle switch. I started with the 
lights ON at brightness=203 and toggled OFF then ON using the switch. The lights were at brightness=1 after that as expected. (I'm not sure I captured everything here.)
[test3.log](https://github.com/Koenkk/zigbee2mqtt/files/13627781/test3.log)

Addresses:  
Diele Deckenlampe 1 - 0x70ac08fffef6f69f (IKEA STOFTMOLN ceiling/wall lamp WW37)  
Diele Deckenlampe 2 - 0xf4b3b1fffe250f1f (ditto)  
Diele Deckenlampe 3 - 0x70ac08fffef6e86a (ditto)  
Diele/Flur Wandschalter - 0x001fee000000642d (Ubisys S2)

Koenkk, 2023-12-10:
What value is returned when reading the `onLevel`?

[image]

Jabe, 2023-12-11:
`Read result of 'genLevelCtrl': {"onLevel":0}` Oh!

I've restored full functionality by setting it to 0xff (previous), including my scene switching when OFF. 😃 Somehow the attrib must have slipped in over time.

JudahBella, 2023-12-11:
> I'm also suffering with this issue. I have just recently started with Home Assistant so I have no prior version history with Z2M and started on 1.34.0. I have LED1949C5 and LED2003G10 bulbs. If the [...]

You'll have to add `transition: 1` to `groups.yaml` for each group of lights you want to manually override transition, i.e.:

```yaml
'1':
  friendly_name: Dry Kitchen Pendants
  transition: 1
  devices:
    - 0x804b50fffefb392a/1
    - 0x84fd27fffe258a5a/1
    - 0x84fd27fffe03dab1/1
```

@Koenkk I'm so sorry but I haven't gotten to downgrading herdsman and providing logs, we've just had a newborn and I haven't had the time, we kinda need the house to just work during this season. Hope the logs from the rest of the community has been sufficient for you. I'll try to circle back again when I have some downtime.

JudahBella, 2023-12-11:
> Over the past few weeks more and more IKEA devices started acting up. Starting at the lowest brightness when turning on. Also Zigbee2MQTT was complaining about not having a valid `transition` [...]

This happens when you click into the transition input box in the GUI, but enter nothing, and click away. The GUI autosaves it as an empty string (rather than do nothing/null/zero), causing this error. 

[image]

alex3305, 2023-12-11:
@JudahBella Like I explicitly said in my comment, which you have even quoted, I didn't touch the transition property input. I barely use Zigbee2MQTT's UI because I control all of my devices through Home Assistant. The behavior also appeared on more and more devices as time went by. I even manually set the transition to `1` or `0` to exclude any misconfiguration on my end, like I also mentioned. But to no avail.

What did happen though, is that some devices that I 100% knew I had set the transition property on, reverted to an empty field after a day or so. And re-setting the value would sometimes make it disappear again 🤷. I'm also confident that everything worked perfectly prior and I didn't make any configuration changes on my end. Except updating Zigbee2MQTT.

However after switching from Zigbee to Home Assistant groups I haven't been affected by this issue since. Some devices within groups also seem to respond quicker. Which I find a tad peculiar.

Koenkk, 2023-12-11:
Thanks @Jabe! So to all who are experiencing this issue, you can fixed it by doing a write in the z2m frontend like this (for attribute select `onLevel`)

[image]

I'm still puzzled why the `onLevel` was set to 1 in the first place, looking at the whole z2m code base, this is only done when the users sets the `level_config`.

villyvitz, 2023-12-11:
@Koenkk is there a way to set this attribute at the group level? I've tried this fix for the individual bulbs, which has worked, but z2m grouped bulbs are still affected.

Jabe, 2023-12-12:
@Koenkk thank you for your hard work! As always.

I've previously stated that I was able to restore the correct behavior with a long power cycle, but later I couldn't reproduce that. In hindsight, there might be a possible explanation.

The bug stopped for a short time when my home had a power outage of about 5 minutes, so the bulbs came back online without z2m being online (the host is much slower). I might have tested it immediately without z2m being able to listen to deviceAnnounce. On later tests I just power cycled the bulbs with z2m being online, which didn't work. Looking at the code introduced in https://github.com/Koenkk/zigbee-herdsman-converters/commit/c99f72d96dbc63b3c5c19982b9b9849ca933b48f, maybe I had a value (like null or "") for `state.level_config.on_level` in my DB, causing z2m to set onLevel to 1 on deviceAnnounce. This could close the loop considering the behavior I noticed. How would one check the DB value? And obviously, how could such a value end up there? 

@villyvitz no there's not. The attribute is stored on the device and it shouldn't make a difference how you get the device to turn on, either directly or through a group.

Jabe, 2023-12-12:
And the story continues.

I had a look in the database.db of a backup from last week and indeed it has onLevel = 0 persisted:
[image]

And now for the fun part: it came back! While checking my current database.db I noticed this:
[image]

This reflects the actual state right in this moment: One light is indeed at brightness = 1, the other two are at 203. Somehow onLevel got back in the DB. 

Now I read the value using the dev console and only *one* device reported 0, contrary to the DB.

alex3305, 2023-12-12:
@Jabe Like I said in my comment yesterday, that also happened to me on the transition property. But only when I use Zigbee groups. Although like you said earlier, that shouldn't make a difference. 

I also tried to read out the `onLevel` but that doesn't seem to work for me 😒 
[image]

Clicking Lezen (or Read) doesn't do anything. Same for Write. Reading an unsupported attribute yields an error.

Koenkk, 2023-12-12:
@Jabe this should only happen when there is an `on_level` for this device in the `data/state.json` file, can you check that?

Jabe, 2023-12-12:
@Koenkk looking at the backup again, it's set to 0 in state.json:

```json
"0x70ac08fffef6f69f": {
    "update": {
        "state": "idle",
        "installed_version": 65542,
        "latest_version": 65542
    },
    "state": "OFF",
    "last_seen": "2023-12-07T16:27:32.352Z",
    "brightness": 203,
    "linkquality": 94,
    "power_on_behavior": "off",
    "level_config": {
        "on_level": 0
    },
    "color_mode": "color_temp",
    "color": {
        "x": 0.4348,
        "y": 0.4033
    },
    "color_temp": 330
}
```

Right now it's set to `"previous"` because I've reset the lights again earlier today.

doerakje, 2023-12-12:
> So to all who are experiencing this issue, you can fixed it by doing a write in the z2m frontend like this (for attribute select `onLevel`)

Thanks! this fixed the problem for me (2x IKEA LED 1836G9 with firmware 2.3.093).

ovidiubute, 2023-12-13:
I finally solved this. I don't know why, but the config value for `LevelCtrl -> startUpCurrentLevel` was set to 255. This is an invalid value since the maximum is 254. So whenever my IKEA lights went through a cold start while being off in HA, they would start at level 1 brightness. As a note `LevelCtrl -> onLevel` was already correctly set to 254 but it made absolutely no difference. 

After I wrote 254 to `startUpCurrentLevel` the problem is now gone!

Koenkk, 2023-12-13:
@Jabe try to:

- stop z2m
- remove the following from your state.json
```
    "level_config": {
        "on_level": 0
    },
```
- start z2m
- write `onLevel`
- see if issue is fixed.

Xitro01, 2023-12-16:
> @Jabe try to:
> 
> * stop z2m
> * remove the following from your state.json
> 
> ```
>     "level_config": {
>         "on_level": 0
>     },
> ```
> 
> * start z2m
> * write `onLevel`
> * see if issue is fixed.

Writing On_Level 0 did not work for me.
After writing 255 to On_Level my bulb worked correctly again, finally saving my brightness.
Have this issue with [LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html#ikea-led1624g9) and [LED1836G9](https://www.zigbee2mqtt.io/devices/LED1836G9.html#ikea-led1836g9)

marazmarci, 2023-12-16:
@Xitro01, if you power-cycle the bulb, does the issue return?

Xitro01, 2023-12-16:
> @Xitro01, if you power-cycle the bulb, does the issue return?

Ah yeah it does, that is kind of worthless.

Jabe, 2023-12-17:
@Koenkk I've been busy these past few days but I just tried it.

1. To verify I power-cycle the lights, dev console reads 0
1. Stop z2m.
4. Check state.json and all of them are set to `"level_config": { "on_level": "previous" }` *confused ??*
5. Start z2m, lights are broken still.
6. Write 255 to all lights using dev console.
7. Stop z2m.
8. state.json now says `"level_config": { "on_level": 0 }`. *confused ??*
9. Remove `level_config` from state.json
10. Start z2m.
11. Lights work, dev console reads 255.
12. Power-cycle lights, dev console reads 0, state.json not updated.
13. Power-cycle lights again, dev console reads 0, state.json not updated.
14. Stop z2m, state.json now says `"level_config": { "on_level": 0 }`
15. Start z2m.

Now it gets wild, but I *did see it happen*: When I power cycle the lights, for a short moment the dev console reads 255 after the devices come back online, then it quickly changes to 0. But that *at least* explains why state.json sometimes has a 0 and sometimes a previous, even in a mixed state. 

I could imagine that there's some fighting going on between the int values and "previous", maybe on device read/write and when applying the value from state.json. Also 0 gets written on deviceAnnounce to the device when state.json has no value for level_config (see  step 11).

SgtZapper, 2023-12-17:
I have the same problem with a pair of IKEA [LED1925G6], that will turn on at 1 
Tried setting the onLevel to 255 in the dev console and that works. But haven't tried power cycling anything yet.

This might also be related.
I have noticed that if I set transition: 0 or anything on a group it will behave strange.
1: Have a group with lights that have different brightness levels.
2: Change the brightness of one light in the group in HA
3: Turn the whole group off in HA
4: Turn the group on again and all members will have the same brightness as the last set member had when turned off.

Koenkk, 2023-12-17:
@Jabe could you give me the herdsman debug logging between step 9 and 13, I don't understand why the level config gets added back to state.json.

mrmaximas, 2023-12-26:
> @Jabe could you give me the herdsman debug logging between step 9 and 13, I don't understand why the level config gets added back to state.json.

it's possible to add on_level and startUpCurrentLevel values manageable via WebUI?

Koenkk, 2023-12-26:
@mrmaximas yes, see https://github.com/Koenkk/zigbee2mqtt/issues/19211#issuecomment-1850725742

mrmaximas, 2023-12-26:
Meanwhile, I got the firmware for the _TZ3210_ngqk6jia via this instructions https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html ;)
https://images.tuyaeu.com/smart/firmware/upgrade/20220915/1663234985-oem_ztu_dimmer1_klt_OTA_1.0.6.bin

mrmaximas, 2023-12-26:
> @mrmaximas yes, see [#19211 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/19211#issuecomment-1850725742)

I didn't mean through the dev console, but through exposes.

danishru, 2023-12-28:
I have the same problem with IKEA TRADFRI Driver 30W and TRADFRI TRADFRI bulb E27 WS opal 1000lm.

My setup.

Zigbee2MQTT version - 1.34.0
Coordinator type - zStack3x0
Coordinator version - 20230507

Model TRADFRI Driver 30W - ICPSHC24-30EU-IL-1
Firmware version TRADFRI Driver 30W - 2.3.086 (20211025)

Model TRADFRI bulb E27 WS opal 1000lm - LED1732G11
Firmware version TRADFRI bulb E27 WS opal 1000lm - 2.0.022 (20190308)

I have not used these devices before, so I cannot say when this started, but since the firmware versions of the devices differ greatly, I conclude that it does not depend on this.
I ran some tests and came up with a stable dependency problem. After physically turning off the power, the lamp and driver always reset the onOffTransitionTime and onLevel attributes in the LevelCtrl cluster to some of their default settings. onOffTransitionTime always returns to 5, and onLevel always returns to 0. When using direct binding, the switches begin to use exactly these parameters, which leads to turning on at minimum brightness and with the specified Transition, while other attributes are not reset. I also disabled zigbee2mqtt to eliminate its influence, the devices also reset these attributes. Using group or device binding also does not affect behavior in any way.

As a temporary workaround, you can use automations in Home Assistant. But to do this, you need to configure reports for the onLevel and onOffTransitionTime attributes. Automation will catch changes in the values of the specified attributes and send the required attribute values to zigbee2mqtt.

Example of automation via link.
1. [Edit an attribute if the value of the required attributes has changed on the device.](https://gist.github.com/danishru/59feea546d06b9fae0a1d1265624edc6)
2. [Always edit the attribute after changing the device status from the unavailable state.](https://gist.github.com/danishru/61e771a07c335eb51575f01ea057f52d)

I'm looking forward to solving this problem, as IKEA devices are of very good quality and functional.

pedrolamas, 2024-01-11:
I have an [IKEA LED1733G7](https://www.zigbee2mqtt.io/devices/LED1733G7.html#ikea-led1733g7) lamp that has been showing this behavior (I believe since the latest release, before was working ok)

My setup is using the latest Z2M Docker image.

I've tried writing `255` to `onLevel` and `254` to `startUpCurrentLevel`:

[image]

This is the state for this lamp:

```json
"0xd0cf5efffe311aa5": {
    "state": "OFF",
    "update_available": false,
    "update": {
        "state": "idle",
        "installed_version": 587814449,
        "latest_version": 587814449
    },
    "color_temp": 338,
    "color_mode": "color_temp",
    "brightness": 254,
    "linkquality": 58,
    "level_config": {
        "on_level": "previous",
        "current_level_startup": 254
    },
    "power_on_behavior": "previous",
    "color_temp_startup": 65535
}
```

Having said that, I have seen a couple of times `"on_level": "0"` on the above state file, which I find odd as that is AFTER I manually wrote the `254` value...

mrmaximas, 2024-01-11:
> Having said that, I have seen a couple of times `"on_level": "0"` on the above state file, which I find odd as that is AFTER I manually wrote the `254` value...
Does this lamp belong in any group?
I often observe this behaviour with the first or last lamp in a group of four, and have never seen it with the second and third.
Only when the mains to this lamps is switched off, on_level & current_level_startup is resets.

pedrolamas, 2024-01-11:
> Does this lamp belong in any group?

No, this specific lamp is not in any group.

mrmaximas, 2024-01-11:
> > Does this lamp belong in any group?
> 
> No, this specific lamp is not in any group.

Okay, I don't have single Ikea gu10 lamps.

baderas, 2024-01-12:
This bug is really annoying. As a workaround i'm just writing "onLevel":255 and "onOffTransitionTime":0 once per hour to evey ikea bulb via mqtt. So far this seems to be working quite well (did not see a low brightness bulb on toggle since then) and not damaging anything (also it seems that it does not change anything?).
Sadly, at least to my knowledge, onLevel and onOffTransitionTime cannot be received via mqtt automatically, otherwise we could react to the change instead of pushing it evey hour (or so).

[57 more replies]