Koenkk/zigbee2mqtt #19218: [Feature request]: broaden calibration range in AVATTO ME167 TRV

wookash, 2023-10-08:
### Is your feature request related to a problem? Please describe

I can't set calibration limits more than -3 +3 range this throws errors

### Describe the solution you'd like

I'd like to be able to set limits in local_temperature_calibration in z2m in larger scale (e.g. +-20)

### Describe alternatives you've considered

using alternative custom converter, but it does not seem to work with calibration

### Additional context

n/a

IvanRebroff, 2023-10-08:
I think the easiest way to solve this would be to use the well known and great working external converter instead the official support for this device. But sadly external converters don't work as soon the official support is integrated.
I'm not a programmer, but to integrate a choose option for official vs. external converters is probably easier than debug every little complaint of a few users...

wookash, 2023-11-08:
> I think the easiest way to solve this would be to use the well known and great working external converter instead the official support for this device. But sadly external converters don't work as [...]

Agreed, I tried this converted and could not get it to work properly. If you know how to do it - please point me in the right direction

closed 2024-06-07 by github-actions[bot]