Koenkk/zigbee2mqtt #19222: Zigbee2mqtt is not working on the iHost

christianbalz, 2023-10-08:
### What happened?

I've installed three Docker containers on iHost, including HA, eclipse-mosquitto and zigbee2mqtt. All of them are up and running. 

I've made numerous attempts, installations and flashed Tasmota on two Sonoff Zigbee Bridge devices, one regular and the other PRO.

I can't perform many checks from iHost and I can only gather information from the logs.

iHost version 1.10.1
Home Assistant 2023.10.0
eclipse-mosquitto version 2.0.18
Zigbee2MQTT version 1.33.1
Sonoff Bridge Tasmota Version 13.1.0 (zbbridge)

Any suggestions on what might be happening?

### What did you expect to happen?

Zigbee2mqtt is not accessing the specified IP and port, showing 'ERR_CONNECTION_REFUSED.'

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

13.1.0

### Adapter

Sonoff Bridge Tasmota Version 13.1.0 (zbbridge)

### Debug log

      p2023-10-08T18:31:51.397635151Z Zigbee2MQTT:debug 2023-10-08 18:31:51: Saving state to file /app/data/state.json
      d2023-10-08T18:31:51.409015512Z Zigbee2MQTT:error 2023-10-08 18:31:51: Not connected to MQTT server!
      �2023-10-08T18:31:51.412149713Z Zigbee2MQTT:error 2023-10-08 18:31:51: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
      e2023-10-08T18:31:51.416385521Z Zigbee2MQTT:info  2023-10-08 18:31:51: Disconnecting from MQTT server
      ^2023-10-08T18:31:51.550418522Z TypeError: Cannot read properties of undefined (reading 'end')
      P2023-10-08T18:31:51.550676060Z     at MQTT.disconnect (/app/lib/mqtt.ts:130:21)
      U2023-10-08T18:31:51.550755683Z     at Controller.stop (/app/lib/controller.ts:198:9)
      A2023-10-08T18:31:51.550800891Z     at stop (/app/index.js:111:5)
      O2023-10-08T18:31:51.550846390Z     at process.handleQuit (/app/index.js:117:9)
      C2023-10-08T18:31:56.787980745Z Using '/app/data' as data directory
      r2023-10-08T18:32:10.874699885Z Zigbee2MQTT:debug 2023-10-08 18:32:10: Loaded state from file /app/data/state.json
      �2023-10-08T18:32:10.892404938Z Zigbee2MQTT:info  2023-10-08 18:32:10: Logging to console and directory: '/app/data/log/2023-10-08.18-32-10' filename: log.txt
      |2023-10-08T18:32:10.976677186Z Zigbee2MQTT:info  2023-10-08 18:32:10: Starting Zigbee2MQTT version 1.33.1 (commit #7e63039)
      h2023-10-08T18:32:11.000281395Z Zigbee2MQTT:info  2023-10-08 18:32:10: Starting zigbee-herdsman (0.19.0)
     �2023-10-08T18:32:11.016052145Z Zigbee2MQTT:debug 2023-10-08 18:32:11: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6760},"serialPort":{"adapter":"ezsp","path":"tcp://192.168.15.7:8888"}}'

christianbalz, 2023-10-08:
Here my configuration.yaml

frontend:
  port: 8099
  host: 0.0.0.0
  auth_token: admin
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.15.95:18883
serial:
  port: tcp://192.168.15.7:8888
  adapter: ezsp
advanced:
  log_level: debug
  zigbee_herdsman_debug: true
  channel: 20
  pan_id: 6760
  network_key: GENERATE

closed 2023-10-08 by christianbalz

darkeagle76, 2023-11-01:
Hello,
Would you please describe with settings you entered in the iHost docker configuration for zigbee2mqtt?
I'm getting an error on startup:

FATAL: A required add-on configuration option is missing!
FATAL: Please set a value for the 'data_path' option.

A detailed description of your setup (including eclipse-mosquito) will be very appreciated.
D

rdaraujo, 2023-12-20:
> Hello @christianbalz, Would you please describe with settings you entered in the iHost docker configuration for zigbee2mqtt? I'm getting an error on startup:

Hello @darkeagle76

The docker image for zigbee2mqtt is **koenkk/zigbee2mqtt**.
For configuration I'm using as:
[image]

Also, install filebrowser/filebrowser docker image to be able to browse the configuration files.
[image]
[image]

And eclipse-mosquitto:
[image]
[image]

Hope it helps.

txitxo0, 2023-12-23:
Any update on this?

yesuaambros, 2024-01-25:
Hello @rdaraujo 

I have tried your steps and I have the mosquitto-server working. I have tested devices with Tasmota and Home Assistant, seeing them correctly.

However, if I try to configure the zigbee2mqtt docker to be able to integrate my zigbee devices, it does not launch the docker, I get an error:

info  2024-01-25 17:19:08: Logging to console and directory: '/app/data/log/2024-01-25.17-19-08' filename: log.txt
info  2024-01-25 17:19:09: Starting Zigbee2MQTT version 1.35.1 (commit #9eaaa0f)
info  2024-01-25 17:19:09: Starting zigbee-herdsman (0.30.0)
error 2024-01-25 17:19:10: Error while starting zigbee-herdsman
error 2024-01-25 17:19:10: Failed to start zigbee
error 2024-01-25 17:19:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-25 17:19:10: Exiting...
error 2024-01-25 17:19:10: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18

Could the ihost device be another?

How could I give the "koenkk/zigbee2mqtt docker" the option of:

     devices:
       - /dev/ttyACM0:/dev/ttyACM0

thanks

rdaraujo, 2024-01-25:
Hello @yesuaambros

[Here](https://forum.ewelink.cc/t/more-hardware-information-of-ihost-for-helping-porting-haos/21641/56) they say the device is `/dev/ttyS4`, but it didn't work for me.

To change it, you edit the file `zigbee2mqtt > configuration.yaml` with filebrowser addon.

yesuaambros, 2024-01-26:
error 2024-01-26 07:48:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2024-01-26 07:48:16: Exiting...
error 2024-01-26 07:48:16: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyS4'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/dist/index.js:82:22)
    at /app/node_modules/@serialport/stream/dist/index.js:118:18

No, it doesn't work either.

I understand that the original problem is that the reference to the device must be shared when creating the container. And in this case, the record would be missing:

  devices:
    - /dev/ttyACM0:/dev/ttyACM0

I don't know if there is a way to put it.

Have you not made zigbee2mqtt work on the sonoff ihost?

rdaraujo, 2024-01-26:
Unfortunately, I didn't succeed.

`Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'`

yesuaambros, 2024-01-26:
After contacting Sonoff they gave me the solution, (although I don't like it very much, it is functional)

They tell me that it is not possible to use the zigbee interface for two entities and it can only be used in the ewelink Cube.

But you can put a zigbee USB stick, create a new docker image of zigbee2mqtt and when you give it advanced options, the stick and the docker interface to associate it with appear.

You can see the thread here:

https://forum.ewelink.cc/t/ihost-devices-in-home-assistant/27626/6

Greetings

rios0rios0, 2025-12-30:
Hi @yesuaambros 
I'm having almost the same error.
Here is the Sonoff iHost tutorial: https://sonoff.tech/en-br/blogs/news/how-to-use-ihost-on-board-zigbee-chip-to-add-zigbee2mqtt-or-zha-in-home-assistant?srsltid=AfmBOoo6pDxTc05XKS6U1j3PrddjHYXimFw_8aFTMJEHvWR9uvkVyVId

Here is the error I'm having:
```
[2025-12-29 21:40:01] debug: 	zh:ezsp:uart: -?- waiting reset
[2025-12-29 21:40:11] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2025-12-29 21:40:11] error: 	zh:ezsp:ezsp: Connection attempt 3 error: Error: Failure to connect
[2025-12-29 21:40:26] debug: 	zh:ezsp:ezsp: Next attempt 4
[2025-12-29 21:40:26] debug: 	zh:ezsp:uart: Opening SerialPort with {"path":"/dev/ttyS4","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: Port closed. Error? null
[2025-12-29 21:40:41] debug: 	zh:ezsp:ezsp: onSerialClose()
[2025-12-29 21:40:41] debug: 	zh:ezsp:driv: onEzspClose()
[2025-12-29 21:40:41] debug: 	zh:ezsp: onDriverClose()
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: Serialport opened
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: Uart reseting
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: --> Write reset
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: --> [1ac038bc7e]
[2025-12-29 21:40:41] debug: 	zh:ezsp:uart: -?- waiting reset
[2025-12-29 21:40:51] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":-1} after 10000ms
[2025-12-29 21:40:51] error: 	zh:ezsp:ezsp: Connection attempt 4 error: Error: Failure to connect
[2025-12-29 21:40:51] debug: 	zh:ezsp:driv: EZSP could not connect: Error: Failure to connect
[2025-12-29 21:40:51] error: 	z2m: Error while starting zigbee-herdsman
[2025-12-29 21:40:51] error: 	z2m: Failed to start zigbee-herdsman
[2025-12-29 21:40:51] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-12-29 21:40:51] error: 	z2m: Exiting...
[2025-12-29 21:40:52] error: 	z2m: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:366:19)
    at Driver.startup (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:186:13)
    at EZSPAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:151:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@6.3.2/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:67:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
```