Koenkk/zigbee2mqtt #19224: Nodon SIN-4-1-20 update fails

chris-1243, 2023-10-08:
### What happened?

I tried to update this device as it seems to be an update available but I always get a timeout error.

It looks like the issue #19129 regarding the Ubisys update

### What did you expect to happen?

To be able to update

### How to reproduce it (minimal and precise)

While trying to update via the frontend

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230923

### Adapter

Sonoff dongle-P

### Debug log

info  2023-10-07 18:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"0uuef-4"}'
debug 2023-10-07 18:48:34: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":534,"imageType":10,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-07 18:48:34: Device '0x04cd15fffea5776b' requested OTA
debug 2023-10-07 18:48:36: Received Zigbee message from '0x04cd15fffea5776b', type 'raw', cluster 'greenPower', data '{"data":[25,26,2,0],"type":"Buffer"}' from endpoint 242 with groupID 0
debug 2023-10-07 18:48:36: No converter available for 'SIN-4-1-20' with cluster 'greenPower' and type 'raw' and data '{"data":[25,26,2,0],"type":"Buffer"}'
debug 2023-10-07 18:48:37: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":266,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-07 18:48:37: Device '0x04cd15fffea5776b' requested OTA
debug 2023-10-07 18:48:48: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x04cd15fffea5776b","transaction":"0uuef-5"}'
info  2023-10-07 18:48:48: Updating '0x04cd15fffea5776b' to latest firmware
debug 2023-10-07 18:48:48: Received Zigbee message from '0x04cd15fffea5776b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2020"}' from endpoint 1 with groupID 0
debug 2023-10-07 18:48:48: Updating to latest '0x04cd15fffea5776b' (SIN-4-1-20)
debug 2023-10-07 18:48:48: Using endpoint '1'
debug 2023-10-07 18:48:50: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":534,"imageType":10,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-07 18:48:50: Got OTA request '{"fieldControl":1,"manufacturerCode":4747,"imageType":10,"fileVersion":534}'
debug 2023-10-07 18:48:50: ZigbeeOTA: downloaded main index
debug 2023-10-07 18:48:50: getNewImage for '0x04cd15fffea5776b', meta {"fileVersion":541,"fileSize":401402,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-000A-021D-700_p1771_nodon_sin_1_fw_efr32_V021D.zigbee","sha512":"fb81608a3cc6e7ffd4efcfc409449d0f75bcea27e6ee1405b8f30e4c59f72346f203d97d1ecc549a2145138f494c1fcdcf65c048cb537bc420989026c97ed40c"}
debug 2023-10-07 18:48:50: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-000A-021D-700_p1771_nodon_sin_1_fw_efr32_V021D.zigbee
debug 2023-10-07 18:48:51: OTA update checksum validation succeeded for '0x04cd15fffea5776b'
debug 2023-10-07 18:48:51: getNewImage for '0x04cd15fffea5776b', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4747,"imageType":10,"fileVersion":541,"zigbeeStackVersion":2,"otaHeaderString":"nodon_sin_efr32_ota\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":401402}
debug 2023-10-07 18:48:51: Got new image for '0x04cd15fffea5776b'
debug 2023-10-07 18:48:51: Starting upgrade
debug 2023-10-07 18:48:51: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":266,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-07 18:48:54: Received Zigbee message from 'Cuisine_SAM', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":36701218,"imageType":31539,"manufacturerCode":4338}' from endpoint 232 with groupID 0
debug 2023-10-07 18:51:21: Update of '0x04cd15fffea5776b' failed (Error: Timeout: device did not request any image blocks)
info  2023-10-07 18:51:21: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffea5776b', payload '{"device":{"applicationVersion":2,"dateCode":"2020","friendlyName":"0x04cd15fffea5776b","hardwareVersion":34,"ieeeAddr":"0x04cd15fffea5776b","manufacturerID":4747,"manufacturerName":"NodOn","model":"SIN-4-1-20","networkAddress":62216,"powerSource":"Mains (single phase)","stackVersion":103,"type":"Router","zclVersion":8},"linkquality":29,"power_on_behavior":null,"state":"OFF","update":{"installed_version":534,"latest_version":541,"state":"available"},"update_available":null}'
info  2023-10-07 18:51:21: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x04cd15fffea5776b"},"error":"Update of '0x04cd15fffea5776b' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"0uuef-5"}'
error 2023-10-07 18:51:21: Update of '0x04cd15fffea5776b' failed (Timeout: device did not request any image blocks)
debug 2023-10-07 18:51:21: Error: Timeout: device did not request any image blocks
    at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:371:28
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)

chris-1243, 2023-10-09:
When posting, I was on the dev branch as I was impacted by the ubisys firmware misbehavior.

I just had a try on the latest dev (1.33.1-dev commit: 9cff19b) and the error remains unfortunately

chris-1243, 2023-10-30:
After some time, I just had a try on the new dev branch (commit e2fccc2) and unfortunately the problem still persists.

Below, a copy/paste of the log

debug 2023-10-30 09:27:08: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"0x04cd15fffea5776b","transaction":"lisi6-9"}'
info  2023-10-30 09:27:08: Updating '0x04cd15fffea5776b' to latest firmware
debug 2023-10-30 09:27:08: Received Zigbee message from '0x04cd15fffea5776b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2020"}' from endpoint 1 with groupID 0
debug 2023-10-30 09:27:08: Updating to latest '0x04cd15fffea5776b' (SIN-4-1-20)
debug 2023-10-30 09:27:08: Using endpoint '1'
debug 2023-10-30 09:27:10: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":534,"imageType":10,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-30 09:27:10: Got OTA request '{"fieldControl":1,"manufacturerCode":4747,"imageType":10,"fileVersion":534}'
debug 2023-10-30 09:27:10: ZigbeeOTA: downloaded main index
debug 2023-10-30 09:27:10: getNewImage for '0x04cd15fffea5776b', meta {"fileVersion":543,"fileSize":401588,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-000A-021F-700_nodon_sin_1_fw_efr32_V021F.zigbee","sha512":"e40fdab5bc41b290ba85b8b1da167b3d4b1698b73b7231e6122a9e8f6e17ddd15b58dbb0a81b0fc91d7098a7d0b094a25ed894d5b77fc5311da4473992e36ea5"}
debug 2023-10-30 09:27:10: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/NodOn/128b-000A-021F-700_nodon_sin_1_fw_efr32_V021F.zigbee
debug 2023-10-30 09:27:11: OTA update checksum validation succeeded for '0x04cd15fffea5776b'
debug 2023-10-30 09:27:11: getNewImage for '0x04cd15fffea5776b', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4747,"imageType":10,"fileVersion":543,"zigbeeStackVersion":2,"otaHeaderString":"nodon_sin_efr32_ota\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":401588}
debug 2023-10-30 09:27:11: Got new image for '0x04cd15fffea5776b'
debug 2023-10-30 09:27:11: Starting upgrade
debug 2023-10-30 09:27:11: Received Zigbee message from '0x04cd15fffea5776b', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65794,"imageType":266,"manufacturerCode":4747}' from endpoint 1 with groupID 0
debug 2023-10-30 09:29:41: Update of '0x04cd15fffea5776b' failed (Error: Timeout: device did not request any image blocks)
info  2023-10-30 09:29:41: MQTT publish: topic 'zigbee2mqtt/0x04cd15fffea5776b', payload '{"device":{"applicationVersion":2,"dateCode":"2020","friendlyName":"0x04cd15fffea5776b","hardwareVersion":34,"ieeeAddr":"0x04cd15fffea5776b","manufacturerID":4747,"manufacturerName":"NodOn","model":"SIN-4-1-20","networkAddress":55631,"powerSource":"Mains (single phase)","stackVersion":103,"type":"Router","zclVersion":8},"linkquality":178,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":534,"latest_version":543,"state":"available"},"update_available":null}'
info  2023-10-30 09:29:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x04cd15fffea5776b"},"error":"Update of '0x04cd15fffea5776b' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"lisi6-9"}'
error 2023-10-30 09:29:41: Update of '0x04cd15fffea5776b' failed (Timeout: device did not request any image blocks)
debug 2023-10-30 09:29:41: Error: Timeout: device did not request any image blocks
    at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:391:28
    at runNextTicks (node:internal/process/task_queues:60:5)
    at listOnTimeout (node:internal/timers:538:9)
    at processTimers (node:internal/timers:512:7)
debug 2023-10-30 09:30:04: Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"warn"}},"transaction":"lisi6-10"}'

I hope we may find a resolution. Anyway, many thanks for your time and effort

jzecca, 2023-12-01:
I've been having the very same issue for maybe 2 or 3 months now (currently running 1.34.0 w/ a Sonoff Dongle-P).

Looks like the SIN 4-2-20 also have the same problem:
https://github.com/Koenkk/zigbee2mqtt/issues/19516

sbourdette, 2023-12-08:
Same for me

ericsche, 2023-12-11:
I'm also facing the same issue.

chris-1243, 2023-12-30:
@Koenkk 

How could I help to solve this issue ?

I would like to help but I honestly have no idea where to begin or what to look for

Koenkk, 2023-12-30:
Could you provide the herdsman debug log when starting an OTA update?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

chris-1243, 2024-01-01:
Here are the logs.

I have waited for the zigbee2mqtt update to see if there were some code changes in the 1.35.0 version

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13805456/log.txt)
[z2m_log-herdsman.txt](https://github.com/Koenkk/zigbee2mqtt/files/13805457/z2m_log-herdsman.txt)

Koenkk, 2024-01-02:
The device doesn't start the OTA indeed, any chance someone can sniff the traffic when updating the device via the original gateway? ([sniffing tutorial](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

ericsche, 2024-01-03:
Hello HNY,
this is tricky they dont have a native gateway they only offer devices.
they have a Android app to update devices via bluetooth but i guess it would not help.
May be it has a different behavior with ZHA?

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

You are receiving this email because you commented on the thread.

chris-1243, 2024-01-03:
I just had a look on the apple store and there is an app available as well for ios (nodon connect)

According to it, my module is up to date...

[image]

merlinpimpim, 2024-02-02:
> I just had a look on the apple store and there is an app available as well for ios (nodon connect)
> 
> According to it, my module is up to date...
> 

Same problem for me.
The Nodon application says that this module is up to date but the OTA via Z2M says that an update is available but it can never be installed.
[image]
[image]

moshiko2312, 2024-02-25:
same problem update 
[image]

raffoul, 2024-04-11:
> Same problem for me. The Nodon application says that this module is up to date but the OTA via Z2M says that an update is available but it can never be installed. [...]

Hello,
Same issue for me. The notification come back every time I ignore the update.

jzecca, 2024-08-07:
Mine finally updated successfully, but I have no clue to why there are so many different versions.
They're the same product, most of them bought from the same site in a single order, yet I get 4 (?!) different version numbers:

- `3.3.0-1.1.2`
- `3.3.0-1.4.4`
- `3.3.0-1.5.0`
- `3.0.0-1.5.0`

I rechecked for updates but as far as z2m is concerned, everything's up to date.

[image]

The same thing's happening with my FP-21s too, no clue to what's going on here:

[image]

I had no luck looking for official firmwares descriptions or changelogs (and haven't tried NodOn mobile app yet).

moshiko2312, 2024-08-07:
Not working fo me 
￼

nanostra, 2024-10-17:
ONE year later...and a lot of crappy or incomplete firmware updates!!! 

[image]

[image]

moshiko2312, 2024-11-25:
> Same issue here

Me to
Anybody can write the solution please...

Gw3n4, 2025-01-03:
Hi,
I meet the same issue. I get the messsage 'no new update available' when trying to update my SIN-4-1-20 device's firmware.
What I see on the mqtt topic:
"update": {
"installed_version": 65794,
"latest_version": 66816,
"state": "available"
},
"update_available": null
}
Installed version is not the latest, but no update available. Weird.

raffoul, 2025-01-06:
Same issue with Z2M 2.0

SentenciaLaMalice, 2025-01-28:
Same here with this device ->

https://www.zigbee2mqtt.io/devices/SIN-4-1-20.html#nodon-sin-4-1-20

CaptainKrby, 2025-03-29:
Come on nodon devs .... Still waiting

CaptainKrby, 2025-05-02:
OTA succeeded to version 773, yesssssss !!! 
Before, I installed Z2M to 2.3.0.

chris-1243, 2025-05-02:
I definitely need to try

lennvilardi, 2025-05-03:
error 2025-05-03 08:53:40z2m: Update of 'interrupteur_filpilote.salledebain' failed (No image currently available) with latest update

[image]