Koenkk/zigbee2mqtt #19230: Silvercrest  LIDL HG08673-BS IAN 390881_2201 SAPZ 1 A2 recognised as TS011F _TZ3000_3uimvkn6

cromelex, 2023-10-09:
### What happened?

I bought a LIDL Silvercrest HG08673-BS (-BS, ie British style plug) as I saw the HG08673 as being supported on Zigbee2Mqtt. 

My Zigbee2MQTT is recognising it as a Tuya TS011F _TZ3000_3uimvkn6 instead. https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3
 
It says Smart plug (with power monitoring by polling). I've noted the comments regarding the polling only on the Tuya plug, but there are no such comments on the LIDL HG08673.

The datapoints seem to work, but not instant, ie, only by polling.

{
    "child_lock": "UNLOCK",
    "current": 0.07,
    "energy": 0,
    "indicator_mode": "off",
    "last_seen": "2023-10-09T11:33:40+01:00",
    "linkquality": 255,
    "power": 8,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": 242,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}

### What did you expect to happen?

I expected that the plug would report the datapoints without polling - just trying to understand if it is being caused by being mis-identified or if it is caused by the underlying unit itself.

Also if someone could confirm that the Energy figure is "read" from the datapoints directly, rather than calculated based on the Power reported, as it is obviously not updating very often, I'd be grateful. I assume that is the case, but would not know.

Thank you in advance

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

 0x26660700

### Adapter

ConBee2

Koenkk, 2023-10-09:
> I expected that the plug would report the datapoints without polling - just trying to understand if it is being caused by being mis-identified or if it is caused by the underlying unit itself.

This is not because mis identification, the firmware on the plug doesn't support reporting, therefore polling is used.

cromelex, 2023-10-09:
> This is not because mis identification, the firmware on the plug doesn't support reporting, therefore polling is used.

Thank you.

Should the plug be identified differently, regardless, in terms of the model in ZigBee2mqtt or should I just close this issue instead?

Koenkk, 2023-10-09:
To do a final check:

Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/86a0d194398e97d5adccda3ba87629fe

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
  - energy updates
  - voltage, power and current update (this is not polled with this converter)

cromelex, 2023-10-10:
> To do a final check:
>
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/86a0d194398e97d5adccda3ba87629fe
>
> * save this as file next to [...]

It worked! 

Tested with my air purifier and changing the settings I can see the power cycling between 0, 17W, 18W, 36W. 

Video showing the states updating multiple times within 1m30s

https://github.com/Koenkk/zigbee2mqtt/assets/96779452/7b04de55-4a81-47d9-b75e-bc1803675807

I've done a reset on the energy measurement and waited to see it update back from 0 to 0.01.  
Can confirm it works as well.
`{
    "child_lock": "UNLOCK",
    "current": 0.1,
    "energy": 0.01,
    "indicator_mode": "off/on",
    "last_seen": "2023-10-10T09:42:51+01:00",
    "linkquality": 215,
    "power": 18,
    "power_outage_memory": "off",
    "state": "ON",
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null,
    "voltage": 245
}`

On the external converted you listed the model as // FR, suppose it can be updated as BS, same with the description, if you include this.

Koenkk, 2023-10-10:
Great, added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-10 by Koenkk