Koenkk/zigbee2mqtt #19233: [New device support]: HS1MIS-3.0 Smart Occupancy Sensor

mario42004, 2023-10-09:
### Link

https://www.heimantech.com/product/?type=detail&id=72

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x84fd27fffe8636a6","nwkAddr":34563,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"PIRILLSens$

### Comments

We have developed the converter based on the documentation and other converters examples.

### External converter

```shell
// Version 1.3
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PIRILLSensor-EF-3.0'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'HS1MIS-3.0' , // Vendor model number, look on the device for a model number
    vendor: 'HEIMAN', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart occupancy sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

    fromZigbee: [fz.occupancy, fz.battery, fz.illuminance],
    toZigbee: [],

   configure: async (device,cordinatorEndpoint,logger)=>{
        const endpoint0 = device.getEndpoint(1)
	const endpoint1 = device.getEndpoint(1)
	await reporting.bind(endpoint0, cordinatorEndpoint, ['msOccupancySensing','genPowerCfg']);
	await reporting.batteryPercentageRemaining(endpoint0)
	await reporting.occupancy(endpoint0)
	await reporting.bind(endpoint1, cordinatorEndpoint, ['msIlluminanceMeasurement']);
	await reporting.illuminance(endpoint1)

},
    exposes: [e.occupancy(), e.battery(),e.illuminance()],

};
module.exports = definition;
```

Koenkk, 2023-10-09:
Added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-09 by Koenkk

mario42004, 2023-10-09:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

You are welcome!!

Regards,

Mario

El presente correo no representa la opinión o consentimiento por parte de la Universidad del Norte por lo que no adquiere responsabilidad alguna por el contenido del presente mensaje. Este correo proviene de una cuenta que la Universidad ofrece a los egresados de la Institución, evento en el cual tanto el mensaje como sus anexos son confidenciales. Este mensaje ha sido verificado con software antivirus; no obstante lo anterior, la Universidad del Norte no garantiza que sea seguro o que no contenga errores o virus por lo que nos e hace responsable por su transmisión.

aelsharawi, 2024-03-15:
need to add similar one here please
https://github.com/Koenkk/zigbee2mqtt/discussions/21796