Koenkk/zigbee2mqtt #19235: Touchlink scan fails with RaspBee II

bitfl0wer, 2023-10-09:
### What happened?

I installed the drivers required for my RaspBee and installed Zigbee2MQTT. The RaspBee and the Zigbee2MQTT connected and seem to know about each other. I wanted to start a Touchlink scan using the Webinterface and got the errors you can see in the debug log.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

0x26690700

### Adapter

RaspBee 2

### Debug log

```
debug 2023-10-09 21:48:33: Loaded state from file /app/data/state.json
info  2023-10-09 21:48:33: Logging to console and directory: '/app/data/log/2023-10-09.21-48-33' filename: log.txt
debug 2023-10-09 21:48:33: Removing old log directory '/app/data/log/2023-10-09.21-28-17'
info  2023-10-09 21:48:33: Starting Zigbee2MQTT version 1.33.1 (commit #7e63039)
info  2023-10-09 21:48:33: Starting zigbee-herdsman (0.19.0)
debug 2023-10-09 21:48:33: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":>
info  2023-10-09 21:48:33: zigbee-herdsman started (resumed)
info  2023-10-09 21:48:33: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":105,"product":0,"revision":"0x26690700","transportrev":0},"type":"ConBee2/RaspBee2"}'
debug 2023-10-09 21:48:33: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2023-10-09 21:48:33: Currently 0 devices are joined:
warn  2023-10-09 21:48:33: `permit_join` set to  `true` in configuration.yaml.
warn  2023-10-09 21:48:33: Allowing new devices to join.
warn  2023-10-09 21:48:33: Set `permit_join` to `false` once you joined all devices.
info  2023-10-09 21:48:33: Zigbee: allowing new devices to join.
info  2023-10-09 21:48:35: Connecting to MQTT server at mqtt://mqtt
debug 2023-10-09 21:48:35: Using MQTT anonymous login
info  2023-10-09 21:48:35: Connected to MQTT server
info  2023-10-09 21:48:35: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2023-10-09 21:48:35: Started frontend on port 0.0.0.0:8080
info  2023-10-09 21:48:35: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"7e63039","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":105,"product":0,"revision":"0x26690700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"debug","network":{"channel":1>
info  2023-10-09 21:48:35: Zigbee2MQTT started!
debug 2023-10-09 21:48:41: Received MQTT message on 'zigbee2mqtt/bridge/request/touchlink/scan' with data '{"transaction":"euzlc-1","value":true}'
info  2023-10-09 21:48:41: Start Touchlink scan
error 2023-10-09 21:48:41: Request 'zigbee2mqtt/bridge/request/touchlink/scan' failed with error: 'not supported'
debug 2023-10-09 21:48:41: Error: not supported
    at DeconzAdapter.restoreChannelInterPAN (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1007:15)
    at Touchlink.scan (/app/node_modules/zigbee-herdsman/src/controller/touchlink.ts:53:32)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
info  2023-10-09 21:48:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{},"error":"not supported","status":"error","transaction":"euzlc-1"}'
```

Koenkk, 2023-10-10:
Touchlink is not supported by the Conbee/Raspbee

bitfl0wer, 2023-10-10:
Through the native deCONZ PhosCon Webinterface, you can issue a Touchlink Reset Command and pair devices with it - I have done so successfully in the past with an older Philips Hue Iris Lamp. Does this not qualify as Touchlink functionality? Curious if so.

Koenkk, 2023-10-10:
I'm not sure, but according to the latest update in https://forum.phoscon.de/t/feature-request-touchlink-support-on-conbee-ii/340/2 it's not.

bitfl0wer, 2023-10-10:
> I'm not sure, but according to the latest update in https://forum.phoscon.de/t/feature-request-touchlink-support-on-conbee-ii/340/2 it's not.

This is very strange.
On my RaspBee II, after setting up the officially endorsed [Docker deCONZ setup](https://github.com/ebaauw/homebridge-deconz#readme), I can access the older WebUI via an intended-for button. Doing this yields access to a button which lets you reset a nearby Touchlink Device. I have made a screen recording which shows this:

https://github.com/Koenkk/zigbee2mqtt/assets/39242991/56e339ff-0d26-4b34-bdf5-e8357ceb9adf

And, not to repeat myself, but this *did* work for me. I was able to reset a Philips LivingColors Iris Lamp using this (which does not support the regular Hue/Zigbee Ecosystem)

Koenkk, 2023-10-10:
Looks it might be supported for Raspbee 2 but not the Conbee 2: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3485

MattWestb, 2023-10-11:
In the video its one RaspBee (not 2) and if have reading right from DE is both Rasp and CornBee 2 have not implanting Touch Linking in the firmware and they is having nearly identical hardware so if putting it in one its only copy and pasting to the other project and it shall working and easier maintainer then using the same code base.

closed 2024-05-09 by github-actions[bot]