Koenkk/zigbee2mqtt #19240: Zigbee Thermostat - Cannot read properties of undefined" (reading 'toString')

GotraSYS, 2023-10-09:
### What happened?

I have installed the following Zigbee Thermostat in Home Assistant:
https://www.amazon.de/dp/B09H6L84BB?ref=ppx_yo2ov_dt_b_product_details&th=1

The Thermostat was recognized as “TuYa TV02-Zigbee“ 
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html

All configurations and settings under "Zigbee2MQTT/Details/Schedule" work well
except the schedule setting, where I get the following error message:

‘TypeError: Cannot read properties of undefined" (reading ‘toString’)’
.
[image]

### What did you expect to happen?

I expected to schedule the temperature for each day..

### How to reproduce it (minimal and precise)

First open under -> Zigbee2MQTT -> Devices 
Then choose the device and klick on -> Exposes 
Then scroll down to -> Schedule
Then choose a day and the error message appear

### Zigbee2MQTT version

1.33.1-1

### Adapter firmware version

20230508135944

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230508135944

### Debug log

Debug 2023-10-09 23:58:30Received MQTT message on 'zigbee2mqtt/Heizung Arbeitszimmer/set' with data '{"schedule":{"week_day":"tuesday"}}'
Debug 2023-10-09 23:58:30Publishing 'set' 'schedule' to 'Heizung Arbeitszimmer'
Error 2023-10-09 23:58:30Publish 'set' 'schedule' to 'Heizung Arbeitszimmer' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'
Debug 2023-10-09 23:58:30TypeError: Cannot read properties of undefined (reading 'toString') at Object.assertString (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:501:111) at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:634:19) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:883:57) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:127:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20)

MahrWe, 2023-10-11:
I encountered the same issue when trying to set the schedule on my Thermostats.

My Devices are Tuya TV05-ZG which are also recognised as TV02 and have been working fine last winter. 

Z2M Versions 1.33.0 and 1.33.1 are at least effected.

Error message:
Zigbee2MQTT:error 2023-10-11 20:08:40: Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of [object Object],[object Object],[object Object]}
Zigbee2MQTT:error 2023-10-11 20:08:42: Exception while calling fromZigbee converter: Value '2' is not allowed, expected one of [object Object],[object Object],[object Object]}
Zigbee2MQTT:error 2023-10-11 20:09:01: Publish 'set' 'schedule' to 'Wohnzimmer Thermostat ' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'

iatzek, 2023-10-12:
1.33.1 in docker with DEBUG=*
```
debug 2023-10-12 14:26:23: Received MQTT message on 'zigbee2mqtt/M_TRV/set' with data '{"schedule":{"schedule":"04:00/23 24:00/21"}}'
debug 2023-10-12 14:26:23: Publishing 'set' 'schedule' to 'M_TRV'
error 2023-10-12 14:26:23: Publish 'set' 'schedule' to 'M_TRV' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'
debug 2023-10-12 14:26:23: TypeError: Cannot read properties of undefined (reading 'toString')
    at Object.assertString (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:501:111)
    at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:606:19)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:883:57)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52)
    at EventEmitter.emit (node:events:526:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:127:27)
    at WebSocket.emit (node:events:514:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20)
debug 2023-10-12 14:26:24: Received MQTT message on 'zigbee2mqtt/M_TRV/set' with data '{"schedule":{"week_day":"monday"}}'
debug 2023-10-12 14:26:24: Publishing 'set' 'schedule' to 'M_TRV'
error 2023-10-12 14:26:24: Publish 'set' 'schedule' to 'M_TRV' failed: 'TypeError: Cannot read properties of undefined (reading 'toString')'
debug 2023-10-12 14:26:24: TypeError: Cannot read properties of undefined (reading 'toString')
    at Object.assertString (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:501:111)
    at Object.to (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:634:19)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:883:57)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52)
    at EventEmitter.emit (node:events:526:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:141:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:127:27)
    at WebSocket.emit (node:events:514:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20)
```

huettenwirt, 2023-10-12:
I just installed a new TuYa TV02-Zigbee compatible one and encountered the same issue. My two old ones (same devices) are working with the schedule set then, but I can't adjust the schedule anymore. Getting the "Cannot read properties of undefined (reading 'toString')'" message as well.

Koenkk, 2023-10-15:
There is currently a bug in the frontend, I will look into it later. For the time being you can set it by publishing an MQTT message.

- Topic: `zigbee2mqtt/my_device_name/set` (change `my_device_name` to your z2m device `friendly_name`)
- Payload: `{"schedule":{"week_day":"saturday", "schedule": "04:00/20 08:30/22 10:10/18 18:40/24 22:50/19.5"}}`

HA users can do this via the developers tools tab:

[image]

JurajSK, 2023-10-17:
for TS0601 it requires different topic
zigbee2mqtt/FRIENDLY_NAME/set/schedule
front end sends wrong topic/payload
I can confirm that it works manualy with developers tools tab with correct topic/payload according [https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html](url)

iatzek, 2023-10-18:
I can confirm it workarounds my problem with TV02 just fine. 
mosquitto_pub -h msqtt.local -p 8883  -t zigbee2mqtt/M_TRV/set -m '{"schedule":{"week_day":"monday", "schedule": "08:00/18 18:00/21 24:00/21" }}'

Zatow, 2023-10-19:
I also get this message when I modify "UnitOfMeasure" for Zipulse device
[image]

Koenkk, 2023-10-24:
This should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

osmeest, 2023-10-25:
Great news ! Thanks Koen. Looking forward testing the dev branch when I find a minute.

huettenwirt, 2023-10-25:
I can confirm, it's working with a TS0601. I used the docker image.

Thanks a lot for fixing this!!!

git-tail-d, 2023-10-25:
Hello,

`# Update
git fetch origin dev
git checkout dev # Change 'dev' to 'master' to switch back to the release version 
git pull
npm ci
`
git checkout dev give the error Pfadspezifikation 'dev' stimmt mit keinen git-bekannten Dateien überein

how it it correct?

closed 2023-10-25 by Koenkk

jiquem, 2023-10-27:
Hello @Koenkk I have the same issue with the Zipulses [https://www.zigbee2mqtt.io/devices/ZiPulses.html#lixee-zipulses](url)

needles77, 2023-11-06:
Hello @Koenkk ,
Same issue than @jiquem  with the lixee-ZiPulse . Same error messages when i want to change divisor or UnitofMeasure.
:(

florianbelhomme, 2023-11-06:
@jiquem @needles77 You can try this workaround https://github.com/fairecasoimeme/ZiPulses/issues/21#issuecomment-1788820015

jiquem, 2023-11-07:
Thanks @Koenkk : I will test this weekend

joe67, 2024-03-15:
Is this bug still present in the version: 1.36.0-1? 
When i publish the schedule in an HA Automation i get the error message "invalid message "null" skipping"

My Automation looks so far:

> service: mqtt.publish
> data_template:
>   topic: zigbee2mqtt/TRV_BZ/set
>   payload: >
>      '{"schedule":{"week_day":"monday", "schedule": "08:00/18 18:00/21 24:00/21" }}'

i also tried over the mqtt exploerer app with this topic 

> zigbee2mqtt/TRV_BZ/set

  and this json data 

> '{"schedule":{"week_day":"monday", "schedule": "08:00/18 18:00/21 24:00/21" }}'

Or do i make something wrong?