Koenkk/zigbee2mqtt #19241: [New device support]: Liwokit _TZE200_hmqzfqml - Light switch with FAN speed regulator switch

tribalRu, 2023-10-09:
### Link

https://aliexpress.ru/item/1005005835561401.html

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0xa4c13822922397ab","nwkAddr":53282,"manufId":4417,"manufName":"_TZE200_hmqzfqml","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_hmqzfqml","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1696887350241,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to add support by myself, but it can't be done without DPID.

tribalRu, 2023-10-23:
For now it has this code:
```
const` fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
 
const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_hmqzfqml',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Power switch with Fan speed switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            e.binary('status_indication', ea.STATE_SET, 'ON', 'OFF').withDescription('Light switch'),
            tuya.exposes.switch(), 
            e.power_on_behavior(['off', 'on']).withAccess(ea.STATE_SET),
            e.enum('fan_speed', ea.STATE_SET, ['minimum', 'medium', 'maximum']).withDescription('Speed off the fan'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff],
                [101, 'fan_speed', tuya.valueConverterBasic.lookup({'minimum': tuya.enum(0), 'medium': tuya.enum(1), 'maximum': tuya.enum(2)})],
                [11, 'power_on_behavior', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'on': tuya.enum(1)})],
                [5, 'status_indication', tuya.valueConverter.onOff],
            ],
        },
};
 
module.exports = definition;
```
All works fine. 
Except for **_power_bihavor_**, I haven't had a chance to test it properly yet.
I'm not even sure it's present in this device. XD

Koenkk, 2023-10-24:
Looks good, could you make a PR?

tribalRu, 2023-10-26:
> Looks good, could you make a PR?

Done... I think ))

Koenkk, 2023-10-27:
Merged, thanks!

closed 2023-10-27 by Koenkk