Koenkk/zigbee2mqtt #19242: BTICINO FC80GCS (Legrand Smart Shedder Modul e)

Lollos60, 2023-10-09:
### Link

https://www.google.com/url?sa=t&source=web&cd=&ved=2ahUKEwjewcq0h-qBAxXS7rsIHaN0CyMQFnoECBUQAQ&url=https%3A%2F%2Fcatalogo.bticino.it%2Fprodotto%2Fsoluzioni-per-la-smart-home%2Fquadro-elettrico-smart-btdin-with-netatmo%2FBTI-FC80GCS-IT&usg=AOvVaw2Y__5W4pfnVhbZUZQbvlua&opi=89978449

### Database entry

{"id":125,"type":"Router","ieeeAddr":"0x00047400008464db","nwkAddr":14094,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Smart shedder module\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":13,"inClusterList":[0,3,4,5,64513,2820],"outClusterList":[6,0,64513,5,25],"clusters":{"genBasic":{"attributes":{"modelId":" Smart shedder module\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":66,"hwVersion":4,"dateCode":" \u000020211013\u0000\u0000\u0000\u0000\u0000","swBuildId":"003b"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":4,"dateCode":" \u000020211013\u0000\u0000\u0000\u0000\u0000","swBuildId":"003b","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1696892624782,"defaultSendRequestWhen":"immediate"}

### Comments

I think that this device Is quite the same as the Legrand   412015 Din Power consumption module (Bticino F20T60A) supported by Zigbee2mqtt. I've tried to start from this devices to make an external converter but didn't work.

Rjevski, 2023-10-13:
I copy-pasted the "DIN power consumption module"'s config from the library and adjusted the model name - seems to work:

Save the below under some name next to your Zigbee2MQTT `configuration.yaml`:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
        zigbeeModel: [' Smart shedder module\u0000\u0000', ' Smart shedder module\u0000', ' Smart shedder module', 'Smart shedder module'],
        model: '412172',
        vendor: 'Legrand',
        description: 'Smart shedder module',
        ota: ota.zigbeeOTA,
        fromZigbee: [fz.identify, fz.metering, fz.electrical_measurement, fz.ignore_basic_report, fz.ignore_genOta,
            fz.legrand_power_alarm, fz.legrand_led_in_dark],
        toZigbee: [tz.legrand_settingEnableLedInDark, tz.legrand_identify, tz.electrical_measurement_power, tz.legrand_powerAlarm],
        exposes: [
            e.power().withAccess(ea.STATE_GET),
            e.power_apparent(),
            e.binary('power_alarm_active', ea.STATE, true, false),
            e.binary('power_alarm', ea.ALL, true, false).withDescription('Enable/disable the power alarm'),
        ],
        onEvent: async (type, data, device, options, state) => {
            /**
             * The DIN power consumption module loses the configure reporting
             * after device restart/powerloss.
             *
             * We reconfigure the reporting at deviceAnnounce.
             */
            if (type === 'deviceAnnounce') {
                for (const endpoint of device.endpoints) {
                    for (const c of endpoint.configuredReportings) {
                        await endpoint.configureReporting(c.cluster.name, [{
                            attribute: c.attribute.name, minimumReportInterval: c.minimumReportInterval,
                            maximumReportInterval: c.maximumReportInterval, reportableChange: c.reportableChange,
                        }]);
                    }
                }
            }
        },
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['haElectricalMeasurement', 'genIdentify']);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
            await reporting.activePower(endpoint);
            await endpoint.read('haElectricalMeasurement', ['activePower']);
            try {
                await reporting.apparentPower(endpoint);
                await endpoint.read('haElectricalMeasurement', ['apparentPower']);
            } catch (e) {
                // Some version/firmware don't seem to support this.
            }
            // Read configuration values that are not sent periodically.
            await endpoint.read('haElectricalMeasurement', [0xf000, 0xf001, 0xf002]);
        },
    };

module.exports = definition;
```

Then in `configuration.yaml` add/amend the following:

```
external_converters:
  - smart_shedder_module.js
```

(note - `external_converters` should be a top-level key and not nested under anything else).

If you're using Home Assistant, install both Z2M and the "File Editor" as add-ons; then using the file editor, do the above in the `zigbee2mqtt` directory.

Restart Z2M and you should be good to go.

Koenkk, 2023-10-14:
@Rjevski you have multiple `zigbeeModel`s , which one is required?

Rjevski, 2023-10-14:
@Koenkk good question - is there a way to get a raw binary dump of the model somewhere from a Zigbee2MQTT install as a HA add-on (so no easy CLI access)?

I've put the multiple ones with the extra null bytes just in case based on patterns I saw in the other Legrand models.

Koenkk, 2023-10-14:
@Rjevski try with just `Smart shedder module` and see if it works.

Rjevski, 2023-10-14:
@Koenkk confirmed that just using `Smart shedder module`, with no _leading_ nor trailing spaces/null bytes works.

Out of interest, is there a way to get the true name anyway? I'm puzzled as to why the "unsupported" log messages had a leading space in the model:

```
Received message from unsupported device with Zigbee model ' Smart shedder module' and manufacturer name ' Legrand'
```

The existing Legrand module in the herdsman library also has leading spaces for some models?

Koenkk, 2023-10-14:
@Rjevski thanks, integrated!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-14 by Koenkk