Koenkk/zigbee2mqtt #19243: Lights turn on if coordinator is disconnected and reconnected

Lollos60, 2023-10-09:
### What happened?

Some devices like Miboxer led controller and Ikea, Osram bulbs turn on if coordinator is disconnected from Power and reconnected. Main problemi Is that new state (on) Is not shown in Zigbee2mqtt, as lights remain on without the possibility to know the effective state.

### What did you expect to happen?

Lights remain in previous state (or in the predefined state) when they reconnect to the coordinator. At least they should show the "on" state when they reconnect.

### How to reproduce it (minimal and precise)

Shut down the coordinator and turn on the coordinator.

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20231226

### Adapter

Sonoff Zigbee Dongle P

lux73, 2023-10-10:
the main Problem: your Coordinator should not disconnect without your Integration eg pulling from Socket

You must solve this Problem in my Opinion

Brumhilde, 2023-10-26:
@lux73 I don't agree. Either way Z2M should show the correct state when devices reconnect to the network.

For example I have 2 thermostats for floor heating that are connected on a separate "heating electricity box" (together with electrical radiators and ventilation). So I can lose power to the house (sockets and lamps) but still have power to heating/ventilation devices. When power is restored I expect the thermostats to report the correct state when they reconnect to the network.

lux73, 2023-10-26:
i agree IF A DEVICE leave the Network

but in this case THE COORDINATOR is disconnected from power ^^

Brumhilde, 2023-10-26:
My case is also the coordinator losing power. For example if I have a ground fault all sockets, lights and appliances will be disconnected from power (the coordinator). But 2 floor heating thermostats will remain powered.

closed 2024-05-24 by github-actions[bot]