Koenkk/zigbee2mqtt #19246: [New device support]:

Killi77, 2023-10-10:
### Link

https://moeshouse.com/products/zigbee-programmable-thermostat-006?variant=44869537956155

### Database entry

'homeassistant/switch/0xa4c13897d0e2954a/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/Water Valve 01/set","device":{"identifiers":["zigbee2mqtt_0xa4c13897d0e2954a"],"manufacturer":"TuYa","model":"Smart water/gas valve (ZN231392)","name":"Water Valve 01"},"name":null,"object_id":"water_valve_01","origin":{"name":"Zigbee2MQTT","sw":"1.33.1","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Water Valve 01","unique_id":"0xa4c13897d0e2954a_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'

### Comments

Dear guys, I need help. I'm completely useless as a programmer trying to read all posts but got lost in midway... I have successfully  jointed my TYBAC-006 Wall-mount thermostat for 2-pipe fan-coil unit from _TZE204_mpbki2zm manufacturer to Home Assistant zigbee2mqtt and all is working just perfect except I can't control 
1. Manual mode manual = ON or Schedule = OFF and 
2. Min temperature limit ECO Cooling energy-saving temperature (default: 26 ºC)
I can see the status in MQTT and is changing as per on device settings but won't work reversely thus can't do the setting form HA. Can you please help?
Thank you.

closed 2023-11-05 by Killi77

pcrednet, 2023-11-27:
Hello, good morning.
I have this thermostat at home and the only problem I see is that home assistant does not show the status of heating or cooling.
It only shows the mode it is in, whether it is cooling or heating.
Is there any way to change this?

Thank you.

pcrednet, 2024-06-28:
Good afternoon.
Is there any way that this thermostat in Home assistant I can see if it is running or not?
This thermostat is active when the valve is open and when it is off it is closed.
In home assistant it is not possible to see when the thermostat is active... is there any way to modify it?

Thank you.