Koenkk/zigbee2mqtt #19248: Ubisys C4 firmware update fail - "device did not request any image blocks"

Ulrar, 2023-10-10:
### What happened?

Z2M reports my Ubisys C4 has a firmware update available, but then fails to perform it with a timeout error.

### What did you expect to happen?

Update the firmware. I have already updated this specific device through Z2M before, it used to work I believe.

### How to reproduce it (minimal and precise)

- Get a Ubisys C4
- Try to udpate the firmware

### Zigbee2MQTT version

1.33.1 commit: 7e63039

### Adapter firmware version

20230507

### Adapter

TubesZB poe (older model with cc chip, not the newer one)

### Debug log

```json
debug 2023-10-10 10:31:53: getNewImage for '0x001fee0000008c3e', meta {"hardwareVersionMin":0,"hardwareVersionMax":4,"fileVersion":36701218,"url":"http://fwu.ubisys.de/smarthome/OTA/release/10F2-7B39-0000-0004-02300422-spo-fmi4.ota.zigbee"}
debug 2023-10-10 10:31:53: getNewImage for '0x001fee0000008c3e', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":60,"otaHeaderFieldControl":4,"manufacturerCode":4338,"imageType":31545,"fileVersion":36701218,"zigbeeStackVersion":2,"otaHeaderString":"ubisys C4 2.3.0\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","totalImageSize":128510,"minimumHardwareVersion":0,"maximumHardwareVersion":4}
debug 2023-10-10 10:31:53: Received Zigbee message from 'Switch - Kitchen - Remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31545,"manufacturerCode":4338}' from endpoint 232 with groupID 0
debug 2023-10-10 10:33:28: Received Zigbee message from 'Switch - Kitchen - Remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":35718162,"imageType":31545,"manufacturerCode":4338}' from endpoint 232 with groupID 0
info  2023-10-10 10:34:23: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Switch - Kitchen - Remote"},"error":"Update of 'Switch - Kitchen - Remote' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"1sxbp-4"}'
    at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:386:28
```

closed 2023-10-10 by Koenkk

m0wlheld, 2023-10-16:
> Already fixed in dev #19129

No, it's not. This issue is new with latest-dev, previously the error message was different. Please re-open.

m0wlheld, 2023-10-16:
Affects 1.33.1-dev commit: [81a8df9](https://github.com/Koenkk/zigbee2mqtt/commit/81a8df9)