Koenkk/zigbee2mqtt #19253: half of my network goes offline

BeFygo, 2023-10-10:
### What happened?

Half my zigbee network goes offline, even devices with LQI > 150.
Rebooting the host (Odroid N2+) resolves the issue.

### What did you expect to happen?

That all the devices stay online and responsive.

### How to reproduce it (minimal and precise)

random

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20230507

### Adapter

Sonoff ZBdongle-P CC2652P 

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12860902/log.txt)

tech2mar, 2023-10-15:
This also happens to me, which is very frustrating, the problem is when I am not at home and I do something that the network would not fix after a reboot.

Koenkk, 2023-10-15:
You are using a fw which is not considered stable for larger networks, please try the new fw from https://github.com/Koenkk/Z-Stack-firmware/discussions/478 and provide feedback there.

closed 2023-10-15 by Koenkk

BeFygo, 2023-10-15:
Thanks for the fast reply.
I will try those firmware asap and give feedback.

Great work!

tech2mar, 2023-10-16:
Currently testing 0923.. I was getting crazy that half of the network goes randomly offline... Thanks