Koenkk/zigbee2mqtt #19256: [New device support]: Tuya Zigbee Sensor de Gas Natural para el hogar, Detector de fugas de Gas GLP inteligente, protección contra incendios

Waterbrain, 2023-10-11:
### Link

https://es.aliexpress.com/item/1005005936314298.html

### Database entry

{"1":"Gas Detection State","2":"Detected Gas","6":"Alarm Ringtone","7":"Alarm Time","8":"Auto-Detect","9":"Auto-Detect Result","10":"Preheat","13":"Alarm Switch","16":"Silence"}

### Comments

Hello, I followed the guide to add new zigbee devices for Tuya, and diving into the tuya.ts repository I wrote this converter. I wonder if before adding it as an external converter, what I have put in would work and not break my Zigbe2MQTT.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_zougpkpy',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Combustible gas detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('alarm_ringtone', ea.STATE_SET, ['1', '2', '3', '4', '5']).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas', tuya.valueConverter.trueFalseEnum0],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({'1': 0, '2': 1, '3': 2, '4': 3, '5': 4})],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

Koenkk, 2023-10-11:
does z2m start and does the device work when using this ext converter?

Waterbrain, 2023-10-12:
This is what I want to know if I activate it. Know if it is well done?

Waterbrain, 2023-10-12:
[08:11:59] INFO: Preparing to start...
[08:12:00] INFO: Socat not enabled
[08:12:02] ERROR: Got unexpected response from the API: Service not enabled
[08:12:02] INFO: Starting Zigbee2MQTT...

Waterbrain, 2023-10-12:
[image]

Waterbrain, 2023-10-12:
But, it detects the gas and does not activate the sensor....

[image]

Waterbrain, 2023-10-12:
I'm going to link it to yours and see something else.....

Waterbrain, 2023-10-12:
[image]
[image]

Waterbrain, 2023-10-12:
May be             [1, 'gas', tuya.valueConverter.trueFalseEnum0],?

Waterbrain, 2023-10-12:
It doesn't work either

[image]

Waterbrain, 2023-10-12:
[1, 'gas', tuya.valueConverterBasic.lookup({'alarm': tuya.enum(0), 'normal': tuya.enum(1)})]]

Waterbrain, 2023-10-12:
Well the gas is sue works:
[image]
Detected:
[image]
But at startup it gives this error:
[image]

And the tone selection doesn't work, any ideas?

Waterbrain, 2023-10-13:
Works perfectly

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_zougpkpy',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Combustible gas detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('alarm_ringtone', ea.STATE_SET, [
            'melody1', 'melody2', 'melody3', 'melody4', 'melody5',
        ]).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas', tuya.valueConverterBasic.lookup({'alarm': tuya.enum(0), 'normal': tuya.enum(1)})],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({
                'melody1': tuya.enum(0), 'melody2': tuya.enum(1), 'melody3': tuya.enum(2), 'melody4': tuya.enum(3),
                'melody5': tuya.enum(4), })],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

Waterbrain, 2023-10-13:
What is the next step?

Koenkk, 2023-10-13:
@Waterbrain integrated it, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-13 by Koenkk

Waterbrain, 2023-10-14:
This is the current status of my converter:
```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_zougpkpy',
        },
    ],
    model: 'TZE204_zougpkpy',
    vendor: 'TuYa',
    description: 'Combustible gas detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('gas_detection_state', ea.STATE, ['alarm', 'normal']).withDescription('State of sensor'),
        e.enum('alarm_ringtone', ea.STATE_SET, [
            'melody1', 'melody2', 'melody3', 'melody4', 'melody5',
        ]).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas_detection_state', tuya.valueConverterBasic.lookup({ 'alarm': tuya.enum(0), 'normal': tuya.enum(1) })],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({
                'melody1': tuya.enum(0), 'melody2': tuya.enum(1), 'melody3': tuya.enum(2), 'melody4': tuya.enum(3),
                'melody5': tuya.enum(4),
            })],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does 
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```
But the detector status does not work for me:

[image]

Although it is detecting gas, the status is still false (**Gas value 13** LEL and **Gas detection state alarm**).

And in Home Assitant states:

[image]

I don't know if it's because of doing so many tests or why this happens.
The installation of zigbee2MQTT, as you can see, is in Homeassitant.

In addons store I see:

[image]

If I install Zigbee2MQTT Edge could you help me?

The thing is that in the documentation to debug the Zigbee protocol it talks about installing a tool that I cannot install on my HomeAssitant.

kerimcelik, 2023-10-24:
Maybe we can manage to close the gas valve by reading the mqtt lel level via nodered.

kerimcelik, 2023-10-24:
Maybe this way we can reset the alarm when it activates.

Waterbrain, 2023-10-25:
But if we do it as I have it, in "Gas detecto state" it does change the states of "alarm" or "normal".
[image]

kerimcelik, 2023-10-25:
I can close the gas valve according to the nodered lel level. Successful.
[https://hizliresim.com/biej18e](url)

Waterbrain, 2023-10-26:
Perfect. Are you using it as an external converter?

Waterbrain, 2023-10-26:
If you use the last abobe use Gas detection state get "alarm" or "normal"

kerimcelik, 2023-10-26:
I did it, I'll check it when I get home.

[image]

kerimcelik, 2023-10-26:
Hello I tried. I enter alarm - normal states. But gas state is entity state It doesn't change. In this case, triggering automation will not be possible.

Waterbrain, 2023-10-27:
The same thing happens with the old one, doesn't it?
The case is that the datapoints that you expose are:
```
{"1":"Gas Detection State","2":"Detected Gas","6":"Alarm Ringtone","7":"Alarm Time","8":"Auto-Detect","9":"Auto-Detect Result","10":"Preheat","13":"Alarm Switch","16":"Silence"}
```
The gas that does not work in any of the two is already exposed in:

```
exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('gas_detection_state', ea.STATE, ['alarm', 'normal']).withDescription('State of sensor'),
        e.enum('alarm_ringtone', ea.STATE_SET, [
            'melody1', 'melody2', 'melody3', 'melody4', 'melody5',
        ]).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
```

Waterbrain, 2023-10-27:
The my last one:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_zougpkpy',
        },
    ],
    model: 'TZE204_zougpkpy',
    vendor: 'TuYa',
    description: 'Combustible gas detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('gas_detection_state', ea.STATE, ['alarm', 'normal']).withDescription('State of sensor'),
        e.enum('alarm_ringtone', ea.STATE_SET, [
            'melody1', 'melody2', 'melody3', 'melody4', 'melody5',
        ]).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas_detection_state', tuya.valueConverterBasic.lookup({ 'alarm': tuya.enum(0), 'normal': tuya.enum(1) })],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({
                'melody1': tuya.enum(0), 'melody2': tuya.enum(1), 'melody3': tuya.enum(2), 'melody4': tuya.enum(3),
                'melody5': tuya.enum(4),
            })],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does 
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```

kerimcelik, 2023-10-27:
The last converter you wrote works fine. Thank you. I have been waiting for a month for such a solution, thank you very much. I got rid of buying a Tuya hub. Thank you again.

Koenkk, 2023-10-27:
@Waterbrain could you make a pr to add out-of-the-box support for this device?

Waterbrain, 2023-10-28:
Of course

Koenkk, 2023-10-28:
Click the pencil/edit icon on [`tuya.ts`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)

Waterbrain, 2023-10-28:
Forgive my ignorance, clumsiness, and age.
You mean that one?
[image]
Because this is the result:
[image]

Koenkk, 2023-10-28:
- Yes, please create a fork, after that you can follow [this guide](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork)
- Also please use `model: 'DY-RQ500A',` instead of `model: 'TZE204_zougpkpy',` (this is what is advertised on AliExpress)

Waterbrain, 2023-10-28:
So I add this to the branch?
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_zougpkpy',
        },
    ],
    model: 'DY-RQ500A',
    vendor: 'TuYa',
    description: 'Combustible gas detector',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
        tuya.exposes.silence(),
        e.enum('gas_detection_state', ea.STATE, ['alarm', 'normal']).withDescription('State of sensor'),
        e.enum('alarm_ringtone', ea.STATE_SET, [
            'melody1', 'melody2', 'melody3', 'melody4', 'melody5',
        ]).withDescription('Ringtone of the alarm'),
        e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
            .withUnit('s').withDescription('Alarm time'),
        e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'gas_detection_state', tuya.valueConverterBasic.lookup({ 'alarm': tuya.enum(0), 'normal': tuya.enum(1) })],
            [2, 'gas_value', tuya.valueConverter.divideBy10],
            [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({
                'melody1': tuya.enum(0), 'melody2': tuya.enum(1), 'melody3': tuya.enum(2), 'melody4': tuya.enum(3),
                'melody5': tuya.enum(4),
            })],
            [7, 'alarm_time', tuya.valueConverter.raw],
            [8, 'self_test', tuya.valueConverter.raw],
            [9, 'self_test_result', tuya.valueConverter.selfTestResult],
            [10, 'preheat', tuya.valueConverter.raw],
            [13, null, null], // alarm_switch; ignore for now since it is unclear what it does 
            [16, 'silence', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;
```
and then?

Koenkk, 2023-10-29:
> Or do I just modify the datapoints in?

Add that indeed, you can use `DYGSM` as the `vendor`.

Waterbrain, 2023-10-29:
I leave it at that:

```
    {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_yojqa8xn', '_TZE204_zougpkpy']),
        model: 'TS0601_gas_sensor_2',
        vendor: 'DYGSM',
        description: 'Gas sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.gas(), tuya.exposes.gasValue().withUnit('LEL'), tuya.exposes.selfTest(), tuya.exposes.selfTestResult(),
            tuya.exposes.silence(),
            e.enum('alarm_ringtone', ea.STATE_SET, ['1', '2', '3', '4', '5']).withDescription('Ringtone of the alarm'),
            e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(180).withValueStep(1)
                .withUnit('s').withDescription('Alarm time'),
            e.binary('preheat', ea.STATE, true, false).withDescription('Indicates sensor preheat is active'),
        ],
        whiteLabel: [
            tuya.whitelabel('TuYa', 'DY-RQ500A', 'Gas sensor', ['_TZE204_zougpkpy']),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'gas_detection_state', tuya.valueConverterBasic.lookup({ 'alarm': tuya.enum(0), 'normal': tuya.enum(1) })],
                [2, 'gas_value', tuya.valueConverter.divideBy10],
                [6, 'alarm_ringtone', tuya.valueConverterBasic.lookup({
                    'melody1': tuya.enum(0), 'melody2': tuya.enum(1), 'melody3': tuya.enum(2), 'melody4': tuya.enum(3),
                    'melody5': tuya.enum(4),
                })],
                [7, 'alarm_time', tuya.valueConverter.raw],
                [8, 'self_test', tuya.valueConverter.raw],
                [9, 'self_test_result', tuya.valueConverter.selfTestResult],
                [10, 'preheat', tuya.valueConverter.raw],
                [13, null, null], // alarm_switch; ignore for now since it is unclear what it does 
                [16, 'silence', tuya.valueConverter.raw],
            ],
        },
    },
```

And then Commit changes?

[image]

Koenkk, 2023-10-29:
> And then Commit changes?

Yes!

Waterbrain, 2023-10-29:
! Done ¡

Waterbrain, 2023-10-29:
Thank you very much, and you know, for whatever you need me.

Waterbrain, 2023-10-29:
I have only been able to change the seller successfully, now that I think I've got the hang of it, I try to change the rest, but when I hit the pencil it comes out the same as at the beginning. I don't know how it looks like.
Tomorrow more.