Koenkk/zigbee2mqtt #19259: [Wrong device]: MINI-ZSB smart button is being recognized as TS0041

bedgerotto, 2023-10-11:
### Link

https://pt.aliexpress.com/item/1005005978561013.html?spm=a2g0o.order_list.order_list_main.22.5ff1caa4W0Q01n&gatewayAdapt=glo2bra

### Model

MINI-ZSB

### Description

Smart Button

### Vendor

Tuya

### Picture (link)

https://iili.io/J3lRpSf.png

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c13874b2dfc565","nwkAddr":39730,"manufId":4417,"manufName":"_TZ3000_qgwcxxws","powerSource":"Battery","modelId":"TS0041","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1697045257537,"defaultSendRequestWhen":"immediate"}

### Notes

The smart button is recognized by zigbee2mqtt as https://www.zigbee2mqtt.io/devices/TS0041.html#tuya-ts0041 and it works partially

The button has 3 actions:
- Single action ✅  `Received Zigbee message from '0xa4c13874b2dfc565', type 'raw', cluster 'genOnOff', data '{"data":[1,96,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0`
- Double action ✅ `Received Zigbee message from '0xa4c13874b2dfc565', type 'raw', cluster 'genOnOff', data '{"data":[1,97,253,1],"type":"Buffer"}' from endpoint 1 with groupID 0`
- Long press 🚫 `Received Zigbee message from '0xa4c13874b2dfc565', type 'raw', cluster 'genOnOff', data '{"data":[1,98,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0`

Koenkk, 2023-10-12:
- Fixed the device detection
- Long press seems to be the same as the single press, we cannot distinguish it (it's the last number in `data`)

bedgerotto, 2023-10-13:
> * Fixed the device detection
> * Long press seems to be the same as the single press, we cannot distinguish it (it's the last number in `data`)

Thank you very much @Koenkk 🙇🏼

closed 2023-10-13 by Koenkk

balboafer, 2024-07-24:
Hi! newbie here! how can i change the config to the correct one?

bedgerotto, 2024-07-24:
Upgrading Zigbeet2Mqtt to the latest version should make it.

admit888, 2025-03-26:
Zigbee2Mqtt Version: 2.1.3-1
I encountered the same issue. How can I enable the "Long press" function?

stripedew, 2025-04-15:
Seems like this is an issue again, any ideas on how to fix it?