Koenkk/zigbee2mqtt #19261: Stuck During `zigbee2mqtt` Installation on Raspberry Pi Zero

nbc8915, 2023-10-12:
### What happened?

Hello.

I am experiencing an issue while attempting to install `zigbee2mqtt` on my Raspberry Pi Zero. The installation process seems to get stuck at the following point:
```
reify:typescript: timing reifyNode:node_modules/zigbee-herdsman-converters Completed in 233749ms
```
I have followed the installation instructions carefully, but I'm unable to proceed beyond this point. I suspect there may be some underlying issues, possibly related to hardware constraints or dependencies.
Here are some details about my setup:
- Raspberry Pi Zero
- OS: Raspbian GNU/Linux 11 (bullseye)
- Node.js version: v16.19.1
I've tried the following troubleshooting steps:
1. Checked for available system resources and memory.
2. Enabled a swap file to increase available memory.
3. Verified that Node.js is installed correctly and up-to-date.
Despite these efforts, I'm still unable to complete the installation process. I'm hoping someone in the community can provide guidance on how to overcome this issue. If anyone has faced a similar problem or has suggestions for resolving it, I'd greatly appreciate your help.
Thank you in advance for your assistance.
Best regards,

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

20221226

### Adapter

SonoFF CC2652P

lux73, 2023-10-12:
Pls check this Thread:

https://github.com/Koenkk/zigbee2mqtt/issues/12034

If not working please search around Issues "closed" - there a several more Thread about install issues on Pi Zero 😉

nbc8915, 2023-10-27:
In the end, I tried referencing the closed issues, but nothing worked out as expected. Therefore, I decided to perform a fresh installation of the OS and start from scratch.

closed 2023-10-27 by nbc8915