Koenkk/zigbee2mqtt #19262: [Wrong device]:

djrm05, 2023-10-12:
### Link

https://es.aliexpress.com/item/1005005665327210.html?spm=a2g0o.productlist.main.51.7773bnS2bnS21d&algo_pvid=e09ea618-0523-4a2b-ab1d-4f463ac23ae8&algo_exp_id=e09ea618-0523-4a2b-ab1d-4f463ac23ae8-25&pdp_npi=4%40dis%21EUR%2112.75%215.73%21%21%2113.23%21%21%40211b615316970956200652857ebbb5%2112000033943581758%21sea%21ES%211774855181%21&curPageLogUid=sJSXvi9D5QHs

### Model

LSPA9

### Description

Smart plug 20A with energy monitoring

### Vendor

ELIVCO

### Picture (link)

https://m.media-amazon.com/images/I/41xfxoGeUFL._AC_UF1000,1000_QL80_.jpg

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x70b3d52b600316d8","nwkAddr":51104,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_okaz9tjs","zclVersion":2,"appVersion":100,"modelId":"TS011F","powerSource":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":237,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,1]}},"genOnOff":{"attributes":{"onOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1697092406962,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-10-12:
According to https://github.com/Koenkk/zigbee2mqtt/issues/17537 `_TZ3000_okaz9tjs` is a BSD29, can you please ask the author in that issue?

djrm05, 2023-10-13:
Thanks @Koenkk 
Device seems to be working except OTA updates. Device doesn't seems to be the same as the described in this issue and the printed model is also different, see the picture.
[image]

n1tehawk, 2023-10-17:
I also got the very same smart plug from an online auction platform, and can confirm the database entry.
```json
{"id":7,"type":"Router","ieeeAddr":"0x70b3d52b60031b0a","nwkAddr":6609,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":2,"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102"}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":231,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":0,"tuyaBacklightMode":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZSKT31BS102","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2123427102},"lastSeen":1697545028173,"defaultSendRequestWhen":"immediate"}
```
The package states _SKU 720055213835_ and _Shenzen Long Sipu Technology Limited_ as the manufacturer. Seems to be another "white label" Tuya product.

My zigbee2mqtt (1.33.1 commit 7e630391) detected and joined the device just fine via Sonoff ZBDongle-E. It gets displayed as **TuYa BSD29** and seems to be fully supported / functions just fine. It just show the wrong picture (British plug instead of EU), the one at <https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html> would be suitable.

So it's a simple question of "BSD29" vs. "TS011F_plug_3"?
Regards, NiteHawk

VerodinDev, 2023-11-28:
Confirming this device is wrongly identified. Checked database which matches previous comments. As mentioned it seems to work, but OTA errors.

rroblik, 2023-12-04:
@VerodinDev / @n1tehawk I have same devis (x3) ; not able to see them under z2m docker latest release.
How did you succeed ?

Thanks !

phixenl, 2023-12-22:
It have almost the same device as shown in the picture above, however I have the 16A version with SKU 720416123547.

It is auto detected as  a TS011F_plug_1 model (via Sonoff ZBDongle-E), it fails OTA support and exposes 'wrong' values. Don't get 'Energy'  data and 'Current'  is reported as 244A (which is impossible ;) ).

`{"id":6,"type":"Router","ieeeAddr":"0x67ca04602bd5b370","nwkAddr":62956,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,4096,10,57345,57344,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":288,"activePower":61}},"genOnOff":{"attributes":{"32768":0,"moesStartUpOnOff":2,"onTime":0,"onOff":1,"tuyaBacklightMode":0,"offWaitTime":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,33]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null}],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1703236113117,"defaultSendRequestWhen":"immediate"}`

I also question if it shcould be BSD29 or TS011F_plug_3.  I doubt if it should be TS011F_plug_3 as i do receive auto updates (i.e. i don't have to poll).

Any hints to fix?

phixenl, 2024-01-19:
> It have almost the same device as shown in the picture above, however I have the 16A version with SKU 720416123547.
> 
> It is auto detected as a TS011F_plug_1 model (via Sonoff ZBDongle-E), it fails OTA support and exposes 'wrong' values. Don't get 'Energy' data and 'Current' is reported as 244A (which is impossible ;) ).
> 
> `{"id":6,"type":"Router","ieeeAddr":"0x67ca04602bd5b370","nwkAddr":62956,"manufId":4660,"manufName":"_TZ3000_okaz9tjs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,4096,10,57345,57344,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_okaz9tjs","powerSource":1,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":230,"rmsCurrent":288,"activePower":61}},"genOnOff":{"attributes":{"32768":0,"moesStartUpOnOff":2,"onTime":0,"onOff":1,"tuyaBacklightMode":0,"offWaitTime":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,33]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x4c5bb3fffe93089b","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null}],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1703236113117,"defaultSendRequestWhen":"immediate"}`
> 
> I also question if it should be BSD29 or TS011F_plug_3. I doubt if it should be TS011F_plug_3 as i do receive auto updates (i.e. i don't have to poll).
> 
> Any hints to fix?

The latest docker build (downloaded today) seems to fix the report issue (now i have valid Current & Energy values). 
Device is still detected as TS011F_plug_1 (_TZ3000_okaz9tjs), OTA doesn't work but that's the lease of my concerns ;)

n1tehawk, 2024-01-23:
@rroblik 
> I have same devis (x3) ; not able to see them under z2m docker latest release. How did you succeed ?
> Thanks !

There was nothing special or complicated about this. As with any other device, I simply activated joining mode, paired the device to the network, and zigbee2mqtt took care of the rest. As stated in various links above, the device should be supported "out of the box" (assuming that your zigbee2mqtt is recent enough).

I have no experience with dockerized zigbee2mqtt. Mine is running 'natively' in a Debian 12 (nodejs + npm) LXC container. I would suggest that you double check your device data (database entry) against what's listed further up, and make to sure your zigbee2mqtt is up-to-date (going by the release notes I'd expect it should be 1.32+).

Regards, NiteHawk

closed 2024-08-22 by github-actions[bot]