Koenkk/zigbee2mqtt #19264: [Wrong device]: Please add https://zigbee.blakadder.com/Tuya_TM-YKQ004.html as Z2M compatible

cloudbr34k84, 2023-10-12:
### Link

https://zigbee.blakadder.com/Tuya_TM-YKQ004.html

### Model

TM-YKQ004

### Description

same

### Vendor

Tuya

### Picture (link)

https://zigbee.blakadder.com/assets/images/devices/Tuya_TM-YKQ004.webp

### Database entry

{"id":102,"type":"EndDevice","ieeeAddr":"0x70ac08fffe697b7f","nwkAddr":27374,"manufId":4098,"manufName":"_TZ3000_u3nv1jwk","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��g+\u0013��g+\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":31,"65508":1,"65534":0,"modelId":"TS0044","manufacturerName":"_TZ3000_u3nv1jwk","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1440156621},"lastSeen":1697093079053,"defaultSendRequestWhen":"immediate"}

### Notes

We just need to update the https://zigbee.blakadder.com/Tuya_TM-YKQ004.html to show its compatiable with Z2M

and we also need to update the picture 
[image]

Koenkk, 2023-10-12:
> We just need to update the https://zigbee.blakadder.com/Tuya_TM-YKQ004.html to show its compatiable with Z2M

I do not control the blakadder site, can you contact them instead?

cloudbr34k84, 2023-10-12:
> > We just need to update the https://zigbee.blakadder.com/Tuya_TM-YKQ004.html to show its compatiable with Z2M
> 
> I do not control the blakadder site, can you contact them instead?

I just checked and it's not on your site either.

Koenkk, 2023-10-13:
Fixed! Device will be added to the site with the next release.

closed 2023-10-13 by Koenkk

deathrattle666, 2024-10-10:
Does someone has an idea how to reset/reconnect the device? Removing the battery over several hours did not bring it back to pairing mode. I am a kind of lost, as I tried various button press combinations too.