Koenkk/zigbee2mqtt #19265: [Wrong device]: _TZE200_v1jqz5cy

cloudbr34k84, 2023-10-12:
### Link

N/a

### Model

N/a

### Description

N/a

### Vendor

N/a

### Picture (link)

https://www.smatrulsmart.com/cdn/shop/files/b8421501ceee8ed8defa49d79fc692df.jpg?v=1690182748&width=1024

### Database entry

N/a

### Notes

Just need to update the picture please

Koenkk, 2023-10-12:
It's correct already? https://www.zigbee2mqtt.io/devices/BLE-YL01.html

cloudbr34k84, 2023-10-12:
> It's correct already? https://www.zigbee2mqtt.io/devices/BLE-YL01.html

[image]

[image]

I'll try re adding

Koenkk, 2023-10-13:
you are not using the latest edge, also make sure to remove your ext converter

cloudbr34k84, 2023-10-13:
Ahhh the external converter! Thank you mate

closed 2023-10-14 by Koenkk