Koenkk/zigbee2mqtt #19268: [Feature request]: Add a mandatory device_type field

fede843, 2023-10-12:
### Is your feature request related to a problem? Please describe

There is no **reliable way** to determine if a device is capable of routing traffic or not.
The common accepted rule in the community is that battery devices are end devices, which probably it is 100% accurate.
Then all wired devices without neutral are considered end devices, and the wired devices with neutra are routers. And this is not accurate. It fits most of cases, but **not all**.
I would really like to be able to commit myself to buy a product knowing exactly before hand if it can act as router or not.

### Describe the solution you'd like

My proposal is to have a mandatory field **device_type** with 4 possible values: _end device_, _router_, _coordinator_ and _not available yet_.
When a contributor adds a new device it is simple enough to populate the field accordingly. I does not require any extra effort really.
For the already existing devices _not available yet_ should be used to kick start. Then the community naturally can help with it. The completeness won't be happening in one day, but eventually it will grow.
I think it worth the shot.

### Describe alternatives you've considered

Search in internet to see if someone claims the device is router or not.
There is no much else you can do really, besides the common accepted rule mentioned above.

### Additional context

The original discussion: https://github.com/Koenkk/zigbee2mqtt/discussions/19255
Another related request for blakadder database: https://github.com/blakadder/zigbee/issues/1051

blakadder, 2023-11-13:
There's a `supports:` field that can be filled with that information yet community isn't updating it

fede843, 2023-11-13:
Hi. What is the type field? What is it for?

closed 2024-06-11 by github-actions[bot]